thesofproject/sof #5494: [BUG][IPC4] FW error : Other failure of module instance initialization request on TGLU RVP and Upextreme when play with SSP0 and SSP2  simultaenously

XiaoyunWu6666, 2022-03-09:
**Describe the bug**
when play with SSP0 and SSP2  simultaneously
FW error of  **  FW reported error: 104 -Other failure of module instance initialization request**  happened first 
and **FW reported error: 9 - Specified resource not found** followed

**To Reproduce**
TPLG=/lib/firmware/intel/avs-tplg/cavs-tgl-nocodec.tplg MODEL=TGLU_RVP_NOCODEC_IPC4 ~/sof-test/test-case/multiple-pause-resume.sh -l 3 -r 5
OR
aplay -D hw:0,0 -r 48000 -c 2 -f S16_LE /dev/zero & aplay -D hw:0,2 -r 48000 -c 2 -f S16_LE /dev/zero

**Reproduction Rate**
almost 100% on TGLU_RVP_NOCODEC_IPC4
also reproducible on i11 UPextreme , Chromebook @ranj063 @bardliao 

**Environment**
Kernel Branch: integration/ipc4-wip
Kernel Commit: [fc948f43](https://github.com/thesofproject/linux/commit/fc948f43)
SOF Branch: main
SOF Commit: [c42ede6a97af](https://github.com/thesofproject/sof/commit/c42ede6a97af)
Zephyr Commit: [7e9e99379679](https://github.com/zephyrproject-rtos/zephyr/commit/7e9e99379679)
Topology:  /lib/firmware/intel/avs-tplg/sof-tgl-nocodec.tplg
Platform: TGLU

**Screenshots or console output**
[console]
```

```
[dmesg]
```

```

bardliao, 2022-03-09:
@XiaoyunWu6666 @ranj063 `multiple-pause-resume.sh -l 3 -r 5` and `run two aplay simultaneously` may not be the same issue. I can't see `FW reported error: ` when I run two aplay simultaneously.
[dmesg_ssp_simultaneously_playback.txt](https://github.com/thesofproject/sof/files/8211926/dmesg_ssp_simultaneously_playback.txt)

XiaoyunWu6666, 2022-03-09:
@bard  you were trying on your i11 extreme?

bardliao, 2022-03-09:
> @bard you were trying on your i11 extreme?

Yes.

> 
> I tried `multiple-pause-resume.sh -l 3 -r 5` and `run two aplay simultaneously` on TGLU_RVP_NOCODEC , both these two can cause the same error (I mean FW error Other failure of module instance initialization request first and Specified resource not found later)
> 
> Are we using the same recipe?

Kernel: https://github.com/thesofproject/linux/commits/integration/ipc4-wip + https://github.com/thesofproject/linux/pull/3493. 
Firmware: closed-source firmware
Topology: Should be the same as yours as there is only one topology for TGL nocodec now.

lgirdwood, 2022-03-09:
@RanderWang @marcinszkudlinski fyi.

XiaoyunWu6666, 2022-03-11:
Sorry for misleading information , this failure is found with cavs fw + integration/ipc4-wip kernel , not sof.