thesofproject/sof #5496: Confirmation of the limited memory space access via DMA in DSP

johnylin76, 2022-03-09:
Hi Intel,

We would like to make sure about the behavior of DMA transactions for DSP firmware running on Intel platforms.

Is it guaranteed that DSP strictly limits the explicit region of memory access via DMA? Is it possible for code running in DSP (or after random crash) to access host address space that is outside of the explicitly programmed DMA buffers?

paulstelian97, 2022-03-09:
Depends on platform. On i.MX platforms, for example, the DSP has complete access to the low 4GB of the host physical address space, and literally code sections _are_ inside a small 16MB-sized section of that address space rather than in its own memory. There are a few limitations so that really only a 128MB section is what is readily accessible (everything else has incompatible cache configuration on the DSP for any such access so other than by malice there isn't much access outside that region plus the audio components region).

I cannot talk about the Intel platforms which use DMA to transfer between host memory and DMA memory.

lgirdwood, 2022-03-09:
> Hi Intel,
>
> We would like to make sure about the behavior of DMA transactions for DSP firmware running on Intel platforms.
>
> Is it guaranteed that DSP strictly limits the explicit region of memory [...]

No, Intel HDA DMA can only access host regions programmed by host driver and DSP regions programmed via FW. This allows for security between host driver and DSP FW as host cannot program DSP regions and FW cannot program host regions.

johnylin76, 2022-03-10:
Thanks for the confirmation, Liam.

closed 2022-03-10 by johnylin76