thesofproject/sof #5510: warning: SMP_BOOT_DELAY (defined at kernel/Kconfig:851) was assigned the value 'y'

dbaluta, 2022-03-10:
Hello @lyakh,

Do you have any suggestion on fixing the following warning. 

```
-- Generated device_extern.h: /home/daniel/work/repos/z2/build-imx8m/zephyr/include/generated/device_extern.h
-- Including generated dts.cmake file: /home/daniel/work/repos/z2/build-imx8m/zephyr/dts.cmake
Parsing /home/daniel/work/repos/z2/zephyr/Kconfig
Loaded configuration '/home/daniel/work/repos/z2/zephyr/boards/xtensa/nxp_adsp_imx8m/nxp_adsp_imx8m_defconfig'
Merged configuration '/home/daniel/work/repos/z2/zephyr/samples/subsys/audio/sof/prj.conf'
Merged configuration '/home/daniel/work/repos/z2/zephyr/samples/subsys/audio/sof/boards/nxp_adsp_imx8m.conf'
Configuration saved to '/home/daniel/work/repos/z2/build-imx8m/zephyr/.config'
Kconfig header saved to '/home/daniel/work/repos/z2/build-imx8m/zephyr/include/generated/autoconf.h'

warning: SMP_BOOT_DELAY (defined at kernel/Kconfig:843) was assigned the value 'y' but got the value
'n'. Check these unsatisfied dependencies: SMP (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_SMP_BOOT_DELAY and/or look up
SMP_BOOT_DELAY in the menuconfig/guiconfig interface. The Application Development Primer, Setting
Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful
too.

```
i.MX8 processor are not  SMP capable but your commit:

commit 0a801aa1dd96a7288a70b4ae2f6e92ebbb815888
Author: Guennadi Liakhovetski <guennadi.liakhovetski@linux.intel.com>
Date:   Thu Apr 1 17:27:35 2021 +0200

    ADSP / SOF: don't force uniprocessor mode
    
    Allow SMP processing for SOF on ADSP. Note, that after this
    SOF will build with SMP support enabled on all cAVS versions,
    but secondary cores will remain disabled unless an SMP
    topology is loaded by the host.
    
    Signed-off-by: Guennadi Liakhovetski <guennadi.liakhovetski@linux.intel.com>
```

```
diff --git a/samples/subsys/audio/sof/boards/intel_adsp_cavs15.conf b/samples/subsys/audio/sof/boards/intel_adsp_cavs15.conf
index ecc40cf02a..bcaf871c13 100644
--- a/samples/subsys/audio/sof/boards/intel_adsp_cavs15.conf
+++ b/samples/subsys/audio/sof/boards/intel_adsp_cavs15.conf
@@ -1,7 +1,6 @@
 CONFIG_APOLLOLAKE=y
 CONFIG_INTEL_DMIC=y
 CONFIG_INTEL_SSP=y
-CONFIG_CORE_COUNT=1
 CONFIG_LP_MEMORY_BANKS=2
 CONFIG_HP_MEMORY_BANKS=8
 CONFIG_PERFORMANCE_COUNTERS=y
diff --git a/samples/subsys/audio/sof/prj.conf b/samples/subsys/audio/sof/prj.conf
index 2be9a027b5..3e7021795d 100644
--- a/samples/subsys/audio/sof/prj.conf
+++ b/samples/subsys/audio/sof/prj.conf
@@ -1,8 +1,10 @@
 CONFIG_SOF=y
-CONFIG_SMP=n
 CONFIG_LOG=y
-CONFIG_MP_NUM_CPUS=1
 CONFIG_BUILD_OUTPUT_BIN=n
 
 # Requires heap_info() be implemented, but no Zephyr wrapper
 CONFIG_DEBUG_MEMORY_USAGE_SCAN=n
+
+CONFIG_SCHED_DEADLINE=y
+CONFIG_SCHED_CPU_MASK=y
+CONFIG_SMP_BOOT_DELAY=y
```

looks like by selecting `CONFIG_SMP_BOOT_DELAY=y` forces CONFIG_SMP=y on all boards.

Should we select CONFIG_SMP_BOOT_DELAY in platform specific conf files?

Cc: @marc-hb

marc-hb, 2022-03-10:
BTW we also have this kconfig warning in XTOS, would be nice to kill two birds with one stone?
```
warning: the int symbol CORE_COUNT (defined at src/platform/Kconfig:286) has a non-int default MP_NUM_CPUS (undefined)
```

```
    286 config CORE_COUNT
    287         int "Number of cores"
    288         default MP_NUM_CPUS if KERNEL_BIN_NAME = "zephyr"
    289         default MAX_CORE_COUNT
    290         range 1 MAX_CORE_COUNT
    291         help
    292           Number of used cores
```
cc: @kv2019i

lyakh, 2022-03-11:
@dbaluta sure, let's move it to a different location from prj.conf

lyakh, 2022-03-11:
@dbaluta https://github.com/zephyrproject-rtos/zephyr/pull/43687

lgirdwood, 2022-03-29:
@lyakh ping.

lyakh, 2022-04-04:
> @lyakh ping.

@lgirdwood as you see above, I proposed a fix, but it isn't perfect. I asked for better ideas but haven't got any suggestions so far.