thesofproject/sof #5518: [FEATURE] Checkers with limits of DSP load tracker

johnylin76, 2022-03-11:
**Is your feature request related to a problem? Please describe.**
We need a fair metrics to evaluate the loading on DSP cores during audio playback/capture/hotwording. To prevent from overloading, checkers should be added with configurable threshold value in CI or maybe in release xtos builds.

**Describe the solution you'd like**
- DSP load tracker on upstream for xtos mode
- Have a configurable (e.g. Kconfig) threshold value to report error or forcedly drop the task with the lowest priority (and etc.) when the measured DSP load is excessive.

**Describe alternatives you've considered**
There is the performance counter which also measures cycle count. However it only reports peak which can be blustered by one burst  in some situations.

**Additional context**
DSP load tracker has been adopted in SOF Zephyr builds and proven to be very useful for debugging.

lgirdwood, 2022-03-11:
@jsarha can you link your PR here. Thanks

johnylin76, 2022-03-25:
#5407 is merged to upstream

lgirdwood, 2022-04-06:
Adding for v2.2 in case we want to improve.

lgirdwood, 2023-05-11:
@kv2019i I think we have load reporting now, and CPC checking is pending merge to main. So will move to v2.7 for full completion.

kv2019i, 2023-05-12:
I believe this can be closed. This was to backport (to XTOS SOF!) the load tracker we implemented for original Zephyr SOF code. This was done in #5407 , so I believe we are good to go. @johnylin76 @cujomalainey can you ack? We can continue perf observability in https://github.com/thesofproject/sof/issues/5521 (still open).

cujomalainey, 2023-05-15:
@kv2019i agreed, and it looks like [zephyr](https://elixir.bootlin.com/zephyr/v1.14.1-rc1/source/subsys/debug/tracing/cpu_stats.c) is much richer in terms of support than xtos for this

closed 2023-05-15 by cujomalainey