thesofproject/sof #5556: [BUG][Zephyr] multiple-pipeline-capture-200.sh fails on ADLP-NOCODEC

kv2019i, 2022-03-16:
**Describe the bug**

**To Reproduce**
TPLG=/lib/firmware/intel/sof-tplg/sof-adl-nocodec.tplg MODEL=ADLP_RVP_NOCODEC_ZEPHYR ~/sof-test/test-case/multiple-pipeline.sh -f c -l 200

**Reproduction Rate**
100% (with 200 iterations)

**Expected behavior**
Test passes

**Impact**
Recording fails.

**Environment**
1) sof-2.1 stable branch SOF+Zephyr
2) Name of the topology file
    * Topology: sof-cavs-nocodec
3) Name of the platform(s) on which the bug is observed.
    * Platform: ADLP, TGLP

**Screenshots or console output**
```
[       52109.633346] (          22.187500) c2 ssp-dai      1.0   /drivers/intel/ssp/ssp.c:962  INFO ssp_early_start(): SSE set for SSP0
[       52253.227090] (         143.593750) c2 ssp-dai      1.0   /drivers/intel/ssp/ssp.c:1079 INFO ssp_trigger() cmd 1
[       52273.331256] (          20.104166) c2 ssp-dai      1.0   /drivers/intel/ssp/ssp.c:977  INFO ssp_start()
[       80064.788485] (       27791.457031) c0 sa                          src/lib/agent.c:83   WARN validate(), ll drift detected, delta = 40517
[       86218.694491] (        6153.905762) c0 sa                          src/lib/agent.c:83   WARN validate(), ll drift detected, delta = 40763
[      136613.588321] (       50394.894531) c0 sa                          src/lib/agent.c:83   WARN validate(), ll drift detected, delta = 40503
[      148851.712835] (       12238.125000) c0 sa                          src/lib/agent.c:83   WARN validate(), ll drift detected, delta = 41096
[      195152.388079] (       46300.675781) c0 sa                          src/lib/agent.c:83   WARN validate(), ll drift detected, delta = 40399
[      197192.440081] (        2040.052002) c0 sa                          src/lib/agent.c:83   WARN validate(), ll drift detected, delta = 40367
[      213275.356109] (       16082.916016) c0 sa                          src/lib/agent.c:83   WARN validate(), ll drift detected, delta = 40700
[      221436.084951] (        8160.729004) c0 sa                          src/lib/agent.c:83   WARN validate(), ll drift detected, delta = 41717
[      229411.605467] (        7975.520508) c0 sa                          src/lib/agent.c:83   WARN validate(), ll drift detected, delta = 40619
```

lgirdwood, 2022-03-16:
@kv2019i I think we need to disable agent (so we only have 1 IRQ user for the timer - no context sharing) and add some checks like (untested).
```diff
diff --git a/src/schedule/zephyr_domain.c b/src/schedule/zephyr_domain.c
index 6b9960268..1c38db6d7 100644
--- a/src/schedule/zephyr_domain.c
+++ b/src/schedule/zephyr_domain.c
@@ -101,6 +101,7 @@ static void zephyr_domain_timer_fn(struct k_timer *timer)
 	struct zephyr_domain *zephyr_domain = k_timer_user_data_get(timer);
 	uint64_t now = k_uptime_ticks();
 	int core;
+	int missed_irq_warn = 0;
 
 	if (!zephyr_domain)
 		return;
@@ -112,8 +113,13 @@ static void zephyr_domain_timer_fn(struct k_timer *timer)
 	 * struct task::start for a strictly periodic Zephyr-based LL scheduler
 	 * implementation, they will be removed after a short grace period.
 	 */
-	while (zephyr_domain->ll_domain->next_tick <= now)
+	while (zephyr_domain->ll_domain->next_tick <= now) {
 		zephyr_domain->ll_domain->next_tick += LL_TIMER_PERIOD_TICKS;
+		if (!missed_irq_warn++)
+			tr_err(&ll_tr, "missed IRQ now %ld next %ld delta %ld", now,
+				zephyr_domain->ll_domain->next_tick,
+				now - zephyr_domain->ll_domain->next_tick);
+	}
 
 	for (core = 0; core < CONFIG_CORE_COUNT; core++) {
 		struct zephyr_domain_thread *dt = zephyr_domain->domain_thread + core;
diff --git a/src/schedule/zephyr_ll.c b/src/schedule/zephyr_ll.c
index 5af538767..1c19330da 100644
--- a/src/schedule/zephyr_ll.c
+++ b/src/schedule/zephyr_ll.c
@@ -21,12 +21,18 @@ DECLARE_SOF_UUID("zll-schedule", zll_sched_uuid, 0x1547fe68, 0xde0c, 0x11eb,
 
 DECLARE_TR_CTX(ll_tr, SOF_UUID(zll_sched_uuid), LOG_LEVEL_INFO);
 
+/* headroom for scheduling, warning if 10uS late. */
+/* TODO: use Zephyr timing macros instead of hard codec clocks */
+#define HEADROOM_CYCLES		(38400 / 100)
+
 /* per-scheduler data */
 struct zephyr_ll {
 	struct list_item tasks;			/* list of ll tasks */
 	unsigned int n_tasks;			/* task counter */
 	struct ll_schedule_domain *ll_domain;	/* scheduling domain */
 	unsigned int core;			/* core ID of this instance */
+	unsigned int period_cycles;		/* period of LL tick in cycles */
+	unsigned int last_tick_cycles;		/* last tick cycles */
 };
 
 /* per-task scheduler data */
@@ -174,6 +180,18 @@ static void zephyr_ll_run(void *data)
 	struct task *task;
 	struct list_item *list;
 	uint32_t flags;
+	unsigned int cycles_now = k_cycle_get_32();
+	unsigned int cycles_target = sch->last_tick_cycles + sch->period_cycles
+				   + HEADROOM_CYCLES;
+
+	/* are we scheduled on time ? if not lets report it as we could be
+	 * blocked by other higher priority threads or IRQs.
+	 */
+	if (cycles_now > cycles_target) {
+		tr_err(&ll_tr, "LL scheduling %d cycles late",
+			cycles_now - cycles_target);
+	}
+	sch->last_tick_cycles = cycles_now;
 
 	zephyr_ll_lock(sch, &flags);
 
@@ -343,6 +361,8 @@ static int zephyr_ll_task_schedule_common(struct zephyr_ll *sch, struct task *ta
 
 	zephyr_ll_unlock(sch, &flags);
 
+	//TODO: is this the best place to initial set last tick cycles.
+	sch->last_tick_cycles = k_cycle_get_32();
 	ret = domain_register(sch->ll_domain, task, &zephyr_ll_run, sch);
 	if (ret < 0)
 		tr_err(&ll_tr, "zephyr_ll_task_schedule: cannot register domain %d",
@@ -534,6 +554,9 @@ int zephyr_ll_scheduler_init(struct ll_schedule_domain *domain)
 	sch->ll_domain = domain;
 	sch->core = cpu_get_id();
 
+	// TODO: get this from topology and use Zephyr API to calc cycles.
+	sch->period_cycles = 38400;
+
 	scheduler_init(domain->type, &z
```

kv2019i, 2022-03-16:
#5557 did not help. Logs are less cluttered, but there's a still failure in configuring SSP1 DAI (on core3).

kv2019i, 2022-03-17:
@lgirdwood Some log data with your patches. First the IRQs. The status doesn't change during any tests when measured at agent. Probably if some irq is blocked, that would have to be observed outside the agent:

[    25633407.734375] (           9.739583) c0 sa                          src/lib/agent.c:87   WARN irq enabled 10040 status 222
[    25641487.421875] (          10.520833) c0 sa                          src/lib/agent.c:87   WARN irq enabled 10040 status 222
[    25645529.895833] (          11.119792) c0 sa                          src/lib/agent.c:87   WARN irq enabled 10040 status 222

kv2019i, 2022-03-17:
The missed IRQ trace didn't provide useful output. The next_tick is updated always once, so this floods the terminal. 

Instead I modified the "LL scheduler" trace a bit. With 10us headroam, this fires too often and affects test execution itself, but with 10% headroom and slight modification to print "x cycles late" without the headroom included, results are as follows:

```
[    26978454.166667] (       50684.347656) c1 zll-schedule       src/schedule/zephyr_ll.c:148  INFO ll task 0xbe083e80 pipe-task <f11818eb-e92e-4082-82a3-dc54c604ebb3> avg 10534, max 12566
[    26978466.510417] (          12.343750) c1 zll-schedule       ......../zephyr_domain.c:89   INFO ll timer avg 10938, max 12770, overruns 0
[    27019625.000000] (       41158.488281) c0 host         21.64         src/audio/host.c:481  INFO no bytes to copy, 0 avail in buffer, 256 free in DMA
[    27064062.968750] (       44437.968750) c1 zll-schedule       src/schedule/zephyr_ll.c:192  INFO LL scheduling 4038 cycles late
[    27166364.739583] (      102301.773438) c1 host         20.56         src/audio/host.c:481  INFO no bytes to copy, 0 avail in buffer, 768 free in DMA
[    27291377.708333] (      125012.968750) c1 zll-schedule       src/schedule/zephyr_ll.c:192  INFO LL scheduling 3947 cycles late
[    27383355.208333] (       91977.500000) c0 zll-schedule       src/schedule/zephyr_ll.c:148  INFO ll task 0xbe086700 pipe-task <f11818eb-e92e-4082-82a3-dc54c604ebb3> avg 3179, max 4953
[    27398462.942708] (       15107.734375) c0 zll-schedule       src/schedule/zephyr_ll.c:192  INFO LL scheduling 3845 cycles late
[    27473343.125000] (       74880.179688) c1 host         20.56         src/audio/host.c:481  INFO no bytes to copy, 0 avail in buffer, 768 free in DMA
[    27493476.015625] (       20132.890625) c0 zll-schedule       src/schedule/zephyr_ll.c:192  INFO LL scheduling 4437 cycles late
[    27604917.500000] (      111441.484375) c0 host         21.64         src/audio/host.c:481  INFO no bytes to copy, 0 avail in buffer, 256 free in DMA
[    27749323.098958] (      144405.593750) c0 zll-schedule       src/schedule/zephyr_ll.c:192  INFO LL scheduling 3950 cycles late
[    27795998.255208] (       46675.156250) c2 zll-schedule       src/schedule/zephyr_ll.c:148  INFO ll task 0xbe088d00 pipe-task <f11818eb-e92e-4082-82a3-dc54c604ebb3> avg 6923, max 8233
```

All cores are loaded, but the delay is not reported on all. Still, the latency spike happens at a very slow pace, here hundreds of milliseconds a part.

Here's another log excepr show warnings from all cores and includes avg/max data from the audio tasks (on different cores). So we get occasional variation in scheduling invocation, but audio processing load is not high and no peaks are seen:

```
[   407846942.916667] (       56481.980469) c0 zll-schedule       src/schedule/zephyr_ll.c:192  INFO LL scheduling 3980 cycles late
[   407850015.338542] (        3072.421875) c0 zll-schedule       src/schedule/zephyr_ll.c:192  INFO LL scheduling 3937 cycles late
[   407886468.411458] (       36453.074219) c1 zll-schedule       src/schedule/zephyr_ll.c:192  INFO LL scheduling 4267 cycles late
[   407930947.161458] (       44478.750000) c0 zll-schedule       src/schedule/zephyr_ll.c:192  INFO LL scheduling 4187 cycles late
[   407950400.104167] (       19452.943359) c1 host         20.56         src/audio/host.c:481  INFO no bytes to copy, 0 avail in buffer, 768 free in DMA
[   407969469.062500] (       19068.958984) c1 zll-schedule       src/schedule/zephyr_ll.c:192  INFO LL scheduling 3914 cycles late
[   408192947.447917] (      223478.390625) c0 zll-schedule       src/schedule/zephyr_ll.c:192  INFO LL scheduling 4205 cycles late
[   408202141.015625] (        9193.567383) c3 zll-schedule       src/schedule/zephyr_ll.c:148  INFO ll task 0xbe08c080 pipe-task <f11818eb-e92e-4082-82a3-dc54c604ebb3> avg 4214, max 5047
[   408202341.119792] (         200.104172) c3 zll-schedule       src/schedule/zephyr_ll.c:148  INFO ll task 0xbe08c1c0 pipe-task <f11818eb-e92e-4082-82a3-dc54c604ebb3> avg 7103, max 7653
[   408202353.593750] (          12.473958) c3 zll-schedule       ......../zephyr_domain.c:89   INFO ll timer avg 11600, max 12722, overruns 0
[   408203210.494792] (         856.901062) c2 zll-schedule       src/schedule/zephyr_ll.c:148  INFO ll task 0xbe088d00 pipe-task <f11818eb-e92e-4082-82a3-dc54c604ebb3> avg 7196, max 8373
[   408203222.942708] (          12.447917) c2 zll-schedule       ......../zephyr_domain.c:89   INFO ll timer avg 7561, max 8564, overruns 0
[   408227392.213542] (       24169.271484) c1 host         20.56         src/audio/host.c:481  INFO no bytes to copy, 0 avail in buffer, 768 free in DMA
[   408352454.843750] (      125062.632812) c0 zll-schedule       src/schedule/zephyr_ll.c:192  INFO LL scheduling 4501 cycles late
[   408401926.901042] (       49472.058594) c1 zll-schedule       src/schedule/zephyr_ll.c:192  INFO LL scheduling 4211 cycles late
[   408418354.244792] (       16427.343750) c1 zll-schedule       src/schedule/zephyr_ll.c:148  INFO ll task 0xbe083e80 pipe-task <f11818eb-e92e-4082-82a3-dc54c604ebb3> avg 10521, max 12456
[   408418366.328125] (          12.083333) c1 zll-schedule       ......../zephyr_domain.c:89   INFO ll timer avg 10791, max 12658, overruns 0
[   408560402.005208] (      142035.671875) c1 host         20.56         src/audio/host.c:481  INFO no bytes to copy, 0 avail in buffer, 768 free in DMA
[... cut ...]
[    41029484.114583] (          12.395833) c1 zll-schedule       ......../zephyr_domain.c:89   INFO ll timer avg 10705, max 12314, overruns 0
```

There is definitely some unexpected variation, but we stay within 15% budget all the time and no peaks in audio processing are reported that would exceed and cause problem even for a single scheduler run.

We can probably add similar lower cost tracking of the IRQ wakeups to get more data on this, without spamming the DMA-trace.  E.g. track avg+max of IRQ wakeup delay.

kv2019i, 2022-03-17:
Managed to reproduce on one of the CI machines now. There's a Zephyr OS panic. Debugging it now.

kv2019i, 2022-03-17:
Status update:
 - the Zephyr panic dumps are scrambled but @lyakh  managed to descramble one and it points to dai_report_xrun()
 - but in today's testing, failure has happened in multiple ways, there is not always a Zephyr panic
 - common thing is an xrun on SSP0 on core0 when three other streams are already active
 - other observed failure patterns
        - running out of memory allocating a new pipe
        - no panic, no out-of-mem, just xrun in the FW log

Bisect shows the spinlock #5286 makes this problem easier to reproduce.

kv2019i, 2022-03-17:
Another update:
 - issue hit without the new spinlock implementation (#5286) -- takes much longer, but can still be hit
 - issue hit on TGL -- harder to hit as well, but similar error footprint

kv2019i, 2022-03-21:
There's a draft workaround for the Zephyr timer reliability, but unfortunately the error is still happening with that workaround in place. There's still unexplained spikes in timer invocation, so there can still be a connection, but at least the error in timer average time does not explain the multiple-pipeline-capture fail.

kv2019i, 2022-03-23:
Status update:
 - one issue found in IRQ driver on Zephyr side -> https://github.com/zephyrproject-rtos/zephyr/pull/44104
 - one issue mutlicore coherency issue found with SOF scheduler code -> https://github.com/thesofproject/sof/pull/5588
 - discovery that on some systems the bug is almost impossible to hit -- no hypothesis what can explain this

kv2019i, 2022-03-23:
Status update:
- with the patches of previous comments and a workaround to https://github.com/zephyrproject-rtos/zephyr/issues/43964, I've today completed 400+ test rounds on three distinct systems without issues
- on one system, the error never appeared, which casts some doubt to the results

kv2019i, 2022-03-25:
Performed a bisect of sort on all current fixes and it turns out the k_sem coherency fix is the key change. Others are probably valid fixed, but not required for this bug. Test details at https://github.com/thesofproject/sof/pull/5588

kv2019i, 2022-03-29:
No errors in Intel daily plan 11433, so potentially fixed by the fix to #5556.

kv2019i, 2022-04-01:
Not seen in Intel daily testing for multiple days, closing.

closed 2022-04-01 by kv2019i