thesofproject/sof #5558: [BUG][ADL]: Glitches seen in HDMI/TBT audio playback with DMIC recording

jairaj-arava, 2022-03-16:
**Describe the bug**
Glitches/chopping are observed in HDMI/Thunderbolt (TBT) display playback with DMIC recording. The issue is not seen on speaker and headset playback with DMIC recording.
The firmware doesn't have WOV enabled but has RNTR noise suppression algo' running on C1. The issue is not seen with firmware with WOV & without RNTR.

**To Reproduce**
1. Connect HDMI/TBT display to the DUT.
2. Start DMIC  recording with **"arecord -Dhw:0,99 -f S32_LE -r 48000 -c 2 jai.wav"**
3. Start playback through HDMI/TBT display **"aplay -Dhw:0,2 -f dat Titanium.wav"**

**Reproduction Rate**
Always

**Expected behavior**
No glitches should be seen in HDMI/TBT display playback.

**Impact**
Showstopper

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: Chromeos-5.10 tree 
    * SOF: adl-004-drop-stable + some external patches to enable RTNR
2) Name of the topology file
    * Topology:  sof-adl-max98390-rt5682
3) Name of the platform(s) on which the bug is observed.
    * Platform: ADL

**Screenshots or console output**
    **12830398.135999] (        9512.343750) c0 sa                          src/lib/agent.c:79   WARN validate(), ll drift detected, delta = 74296
[    12839888.343955] (        9490.208008) c0 sa                          src/lib/agent.c:79   WARN validate(), ll drift detected, delta = 64511
[    12850897.249768] (       11008.906250) c0 sa                          src/lib/agent.c:79   WARN validate(), ll drift detected, delta = 84000
[    12860398.499390] (        9501.250000) c0 sa                          src/lib/agent.c:79   WARN validate(), ll drift detected, delta = 74312
[    12869879.540680] (        9481.041016) c0 sa                          src/lib/agent.c:79   WARN validate(), ll drift detected, delta = 64461
[    12880895.581909] (       11016.041016) c0 sa                          src/lib/agent.c:79   WARN validate(), ll drift detected, delta = 83852
[    12890401.987781] (        9506.406250) c0 sa                          src/lib/agent.c:79   WARN validate(), ll drift detected, delta = 74379
[    12899889.279071] (        9487.291016) c0 sa                          src/lib/agent.c:79   WARN validate(), ll drift detected, delta = 64537**
[    12910811.466137] (       10922.187500) c0 ipc                  src/ipc/ipc3/handler.c:1576 INFO ipc: new cmd 0x80010000
[    12910854.122385] (          42.656250) c0 sa                          src/lib/agent.c:79   WARN validate(), ll drift detected, delta = 83116
[    12911083.653626] (         229.531235) c0 ipc                  src/ipc/ipc3/handler.c:569  INFO ipc: dai 3.0 -> config
[    12911103.705709] (          20.052082) c0 dai                           src/lib/dai.c:164  INFO dai_get type 3 index 0 new sref 2
[    12911338.601533] (         234.895828) c0 hda-dai      3.0   /drivers/intel/hda/hda.c:36   INFO hda_set_config(): channels 2 rate 48000
[    12911357.664032] (          19.062500) c0 dai                           src/lib/dai.c:188  INFO dai_put type 3 index 0 new sref 1
[    12911376.309865] (          18.645832) c0 ipc                      src/ipc/ipc3/dai.c:187  INFO ipc_comp_dai_config() dai type = 3 index = 0
[    12911410.997363] (          34.687500) c0 dai          5.24        src/ipc/ipc3/dai.c:281  INFO dai_config() dai type = 3 index = 0 dd 0x9e1a0f00
[    12912027.820255] (         616.822876) c0 ipc                  src/ipc/ipc3/handler.c:1576 INFO ipc: new cmd 0x60010000
[    12912059.122338] (          31.302082) c0 pipe         5.25  ....../pipeline-params.c:222  INFO pipe params dir 0 frame_fmt 0 buffer_fmt 0 rate 48000
[    12912076.934837] (          17.812500) c0 pipe         5.25  ....../pipeline-params.c:226  INFO pipe params stream_tag 1 channels 2 sample_valid_bytes 2 sample_container_bytes 2
[    12912101.205669] (          24.270832) c0 hda-dai      3.0   /drivers/intel/hda/hda.c:53   INFO hda_get_hw_params(): channels 2 rate 48000
[    12912222.351498] (         121.145828) c0 dai          5.24        src/ipc/ipc3/dai.c:103  INFO dai_data_config() dai type = 3 index = 0 dd 0x9e1a0f00
[    12912243.184830] (          20.833332) c0 hda-dai      3.0   /drivers/intel/hda/hda.c:53   INFO hda_get_hw_params(): channels 2 rate 48000
[    12912304.018161] (          60.833332) c0 dai          5.24           src/audio/dai.c:347  INFO dai_playback_params() dest_dev = 0 stream_id = 0 src_width = 4 dest_width = 4
[    12912324.278577] (          20.260416) c0 dai          5.24           src/audio/dai.c:353  INFO dai_playback_params() fifo 0x0
[    12912351.778576] (          27.499998) c0 pipe         5.25  ....../pipeline-params.c:308  INFO pipe prepare
[    12912424.330656] (          72.552078) c0 dai          5.24           src/audio/dai.c:622  INFO dai_prepare()
[    12912445.111906] (          20.781250) c0 dai          5.24           src/audio/dai.c:587  INFO dai_config_prepare(), channel = 0
[    12912688.132729] (         243.020828) c0 dai          5.24           src/audio/dai.c:606  INFO dai_config_prepare(): new configured dma channel index 0
[    12920317.872009] (        7629.739258) c0 ipc                  src/ipc/ipc3/handler.c:1576 INFO ipc: new cmd 0x60040000
[    12920361.257424] (          43.385414) c0 sa                          src/lib/agent.c:79   WARN validate(), ll drift detected, delta = 73643
[    12920590.267832] (         229.010406) c0 pipe         5.25  ....../pipeline-stream.c:261  INFO pipe trigger cmd 9
[    12920832.038656] (         241.770828) c0 ll-schedule        ./schedule/ll_schedule.c:451  INFO task add 0xbe2301c0 pipe-task <f11818eb-e92e-4082-82a3-dc54c604ebb3>
[    12920850.371988] (          18.333332) c0 ll-schedule        ./schedule/ll_schedule.c:455  INFO task params pri 0 flags 0 start 0 period 1000
[    12920876.778237] (          26.406248) c0 ll-schedule        ./schedule/ll_schedule.c:357  INFO new added task->start 253426462 at 253396058
[    12920894.382403] (          17.604166) c0 ll-schedule        ./schedule/ll_schedule.c:360  INFO num_tasks 4 total_num_tasks 4
**[    12929985.319542] (        9090.937500) c0 hda-dma            ..../intel/hda/hda-dma.c:953  ERROR hda_dma_link_check_xrun(): underrun detected
[    12930172.923701] (         187.604156) c0 sa                          src/lib/agent.c:79   WARN validate(), ll drift detected, delta = 69939
[    12930575.736185] (         402.812469) c0 hda-dma            ..../intel/hda/hda-dma.c:953  ERROR hda_dma_link_check_xrun(): underrun detected
[    12931134.538246] (         558.802063) c0 dai          5.24           src/audio/dai.c:943  WARN dai_copy(): Copy_bytes 336 + avail bytes 0 < period bytes 384, possible glitch
[    12932887.923593] (        1753.385376) c0 dai          5.24           src/audio/dai.c:943  WARN dai_copy(): Copy_bytes 336 + avail bytes 0 < period bytes 384, possible glitch**

jairaj-arava, 2022-03-16:
@sathya-nujella @sathyap-chrome  @Vamshigopal @abonislawski @kv2019i @cujomalainey

plbossart, 2022-03-16:
> The firmware doesn't have WOV enabled but has RNTR noise suppression algo' running on C1. The issue is not seen with firmware with WOV & without RNTR.

@jairaj-arava is the RNTR on for the DMIC path, and if yes do you have an ALSA kcontrol to disable RNTR at run-time? just checking if we can reduce the load on the DSP by putting this processing in bypass while keeping the data path identical.

ranj063, 2022-03-16:
> @plbossart yes, the controls are in the UCM and can be modified accordingly. Although the entire DMIC + Speaker path is on c1 and HDMI is on C0. I wonder if there is a scheduler issue here.

@cujomalainey the interesting thing is DMIC recording + speaker/headset playback seems to be OK. It is only the link DMA that runs into an xrun. So not sure if it is the scheduler acting up

cujomalainey, 2022-03-16:
> @cujomalainey the interesting thing is DMIC recording + speaker/headset playback seems to be OK. It is only the link DMA that runs into an xrun. So not sure if it is the scheduler acting up

isn't these lines signs that c0 is lagging?
```
[ 12890401.987781] ( 9506.406250) c0 sa src/lib/agent.c:79 WARN validate(), ll drift detected, delta = 74379
[ 12899889.279071] ( 9487.291016) c0 sa src/lib/agent.c:79 WARN validate(), ll drift detected, delta = 64537
```
Even though c0 is just doing a straight through copy?

cujomalainey, 2022-03-17:
Looks like I may have made an error due to the fact that intel-generic-kwd and intel-generic use similar but not the exact same names, and undefined params just silently default. So RTNR is running on c0 it seems

cujomalainey, 2022-03-17:
Filed https://github.com/thesofproject/sof/issues/5560

jairaj-arava, 2022-03-17:
@cujomalainey
As discussed, I moved the core ID to 1 in tools/topology/topology1/platform/intel/intel-generic-dmic.m4 and could see the RTNR is running on CORE1 and issue is not seen any more.
Will submit the changes soon.

jairaj-arava, 2022-03-18:
Hi, The fix https://github.com/thesofproject/sof/pull/5565 for this issue is under review.

closed 2022-06-13 by abonislawski