thesofproject/sof #5560: [FEATURE] align intel-generic-dmic*

cujomalainey, 2022-03-17:
**Is your feature request related to a problem? Please describe.**
The following two files use roughly the same params but have slightly different names, this makes switching between them dangerous and prone to error.

tools/topology/topology1/platform/intel/intel-generic-dmic-kwd.m4                                                                                                                                                                                                                         
tools/topology/topology1/platform/intel/intel-generic-dmic.m4

e.g.
`DMIC_PIPELINE_48k_CORE_ID` and `DMIC_48k_CORE_ID`

**Describe the solution you'd like**
Align the params to the same names

**Describe alternatives you've considered**
N/A

lgirdwood, 2022-03-18:
@singalsu @ranj063 fyi - need to align these in topology 2 as well.

lgirdwood, 2023-05-11:
@jsarha can you confirm we have this solution in topology2.

jsarha, 2023-05-26:
> @jsarha can you confirm we have this solution in topology2.

My understanding of topology1 is quite limited so I do not fully understand the details of the problem here. However, since all the DMIC objects in topology2 files are instances of the same DMIC class [1] I can not see how we could have the problem in topology2. What do you say @ranj063 ?

[1] https://github.com/thesofproject/sof/blob/main/tools/topology/topology2/include/dais/dmic.conf

ranj063, 2023-05-31:
@jsarha @lgirdwood the DMIC pipeline definitions for the 2 DMICs are split between 2 separate files today and the pipeline ID for the 48K DMIC is set in dmic-default.conf. Since we already have a clean split for the 2 DMICs, hopefully we won't run into the same problem with topology2.

lgirdwood, 2023-05-31:
> @jsarha @lgirdwood the DMIC pipeline definitions for the 2 DMICs are split between 2 separate files today and the pipeline ID for the 48K DMIC is set in dmic-default.conf. Since we already have a clean split for the 2 DMICs, hopefully we won't run into the same problem with topology2.

@ranj063 are you saying this is not an issue in topology2 ?

lgirdwood, 2023-06-01:
@ranj063 thanks, closing then since it's been resolved in tplg2 arch.

closed 2023-06-01 by lgirdwood