thesofproject/sof #5562: [BUG][Zephyr] ipc tx error for COMP_FREE when testing multiple pipeline all on ADLP Zephyr

miRoox, 2022-03-17:
**Describe the bug**
An IPC tx error for COMP_FREE when testing multiple pipeline all on ADLP Zephyr has been observed by CI daily test 11126:

```
[ 1332.672288] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx error for 0x30020000 (msg/reply size: 12/12): -19
[ 1332.672381] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: failed to free widget DMIC1.IN
```

**To Reproduce**
```
TPLG=/lib/firmware/intel/sof-tplg/sof-adl-nocodec.tplg MODEL=ADLP_RVP_NOCODEC_ZEPHYR ~/sof-test/test-case/multiple-pipeline.sh -f p -c 20 -l 50
```

**Reproduction Rate**
Only happened once in the CI daily test, I cannot reproduce it manually.

**Expected behavior**
No IPC error.

**Environment**
Kernel: sof-dev/https://github.com/thesofproject/linux/commit/c26e1b03638d
SOF : main/https://github.com/thesofproject/sof/commit/219903903dc3
Zephyr Commit: https://github.com/zephyrproject-rtos/zephyr/commit/22b06c238856

**Screenshots or console output**

**dmesg**
```
[ 1332.671959] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30020000: GLB_TPLG_MSG: COMP_FREE
[ 1332.671979] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[ 1332.671994] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[ 1332.672288] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx error for 0x30020000 (msg/reply size: 12/12): -19
[ 1332.672381] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: failed to free widget DMIC1.IN
[ 1332.672386] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30210000: GLB_TPLG_MSG: BUFFER_FREE
[ 1332.672473] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0xe0]=0x20440000 successful
[ 1332.672586] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x30210000: GLB_TPLG_MSG: BUFFER_FREE
[ 1332.672591] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: widget BUF21.2 freed
[ 1332.672595] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[ 1332.672990] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[ 1332.673004] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[ 1332.673088] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[ 1332.673136] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[ 1332.673423] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[ 1332.673546] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[ 1332.673719] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x1c0]=0x20440000 successful
[ 1332.673999] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[ 1332.674007] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[ 1332.674018] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[ 1332.674032] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30020000: GLB_TPLG_MSG: COMP_FREE
[ 1332.674178] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x30020000: GLB_TPLG_MSG: COMP_FREE
[ 1332.674181] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: widget EQIIR21.0 freed
[ 1332.674183] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30210000: GLB_TPLG_MSG: BUFFER_FREE
[ 1332.674333] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x30210000: GLB_TPLG_MSG: BUFFER_FREE
[ 1332.674335] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: widget BUF21.1 freed
[ 1332.674385] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[ 1332.674664] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[ 1332.674777] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30020000: GLB_TPLG_MSG: COMP_FREE
[ 1332.675013] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[ 1332.675026] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[ 1332.675158] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x30020000: GLB_TPLG_MSG: COMP_FREE
[ 1332.675161] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: widget Dmic1 freed
[ 1332.675164] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30210000: GLB_TPLG_MSG: BUFFER_FREE
[ 1332.675363] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x30210000: GLB_TPLG_MSG: BUFFER_FREE
[ 1332.675365] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: widget BUF21.0 freed
[ 1332.675368] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30020000: GLB_TPLG_MSG: COMP_FREE
[ 1332.675526] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x30020000: GLB_TPLG_MSG: COMP_FREE
[ 1332.675528] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30110000: GLB_TPLG_MSG: PIPE_FREE
[ 1332.675676] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x30110000: GLB_TPLG_MSG: PIPE_FREE
[ 1332.675677] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: widget PIPELINE.21.DMIC1.IN freed
[ 1332.675678] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: widget PCM11C freed
[ 1332.675679] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Failed to free connected widgets
[ 1332.675683] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_free on 0000:00:1f.3: -19
[ 1332.675698] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: close stream 11 dir 1
[ 1332.675791] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: free stream 1 dir 1
```

**slogger**
```
[    81183517.346974] (         550.937500) c0 ipc                  src/ipc/ipc3/handler.c:1584 INFO ipc: new cmd 0x30210000
[    81183538.336556] (          20.989582) c0 ipc                  src/ipc/ipc3/handler.c:1366 INFO ipc: comp 69 -> free
[    81183936.253207] (         397.916656) c0 ipc                  src/ipc/ipc3/handler.c:1584 INFO ipc: new cmd 0x60050000
[    81184049.430286] (         113.177078) c2 pipe         11.55 ....../pipeline-stream.c:261  INFO pipe trigger cmd 0
[    81184682.555261] (         633.125000) c2 zll-schedule       src/schedule/zephyr_ll.c:67   INFO task complete 0xbe08aa40 pipe-task 
[    81182779.274086] (          44.218750) c0 ipc                    src/ipc/ipc-helper.c:272  ERROR ipc_comp_free(): comp id: 70 is not found
[    81182966.409495] (         187.135406) c0 ipc                  src/ipc/ipc3/handler.c:1373 ERROR ipc: comp 70 free failed -19
[    81184701.721926] (          19.166666) c2 zll-schedule       src/schedule/zephyr_ll.c:69   INFO num_tasks 3 total_num_tasks 9
[    81185022.451080] (         320.729156) c0 ipc                  src/ipc/ipc3/handler.c:1584 INFO ipc: new cmd 0x60030000
[    81185067.555245] (          45.104164) c0 pipe         6.35  ......./pipeline-graph.c:342  INFO pipe reset
[    81185193.336490] (         125.781242) c0 dai          6.34           src/audio/dai.c:669  INFO dai_reset()
[    81185230.367739] (          37.031250) c0 dw-dma                 src/drivers/dw/dma.c:258  INFO dw_dma_channel_put(): dma 1 channel 3 put
```

lgirdwood, 2022-03-18:
@kv2019i looks like we need a debug KCONFIG option that dumps Zephyr heap usage (free/used)  when we add and remove components.

keqiaozhang, 2022-03-21:
Let's close this one and track all types of IPC errors in https://github.com/thesofproject/sof/issues/5554

closed 2022-03-21 by keqiaozhang