thesofproject/sof #5564: [BUG] IPC4 timestamping - LLP and wall clock are not updated in the same sequence

plbossart, 2022-03-17:
The current implementation of the timestamping doesn't seem quite right.

```c
void dai_dma_position_update(struct comp_dev *dev)
{
	struct dai_data *dd = comp_get_drvdata(dev);
	struct ipc4_llp_reading_slot slot;
	struct dma_chan_status status;
	uint32_t llp_reg_offset;
	uint32_t node_id;
	uint32_t llp_data[2];

	get_llp_reg_info(dd, &node_id, &llp_reg_offset);
	if (!node_id)
		return;

	status.ipc_posn_data = llp_data;
	dma_status(dd->chan, &status, dev->direction);

	slot.node_id = node_id;
	slot.reading.llp_l = llp_data[0];
	slot.reading.llp_u = llp_data[1];
	slot.reading.wclk_l = (uint32_t)dd->wallclock;
	slot.reading.wclk_u = (uint32_t)(dd->wallclock >> 32);

	mailbox_sw_regs_write(llp_reg_offset, &slot, sizeof(slot));
}
````

The wallclock value should be updated in the same sequence where the LLP values are read, not read from a cached value updated heavens knows when. 

@RanderWang @lgirdwood @mmaka1 would you concur with my analysis?

RanderWang, 2022-03-18:
@plbossart  thanks! good catch! It a typical copy-paste-error. please check https://github.com/thesofproject/sof/pull/5568.

RanderWang, 2022-03-18:
I search it in audio driver and find that it is not used now

closed 2022-03-22 by kfrydryx