thesofproject/sof #5573: [BUG][Zephyr] IPC tx timed out(COMP_CONNECT/COMP_NEW/PIPE_FREE) on ADLP Zephyr platform

keqiaozhang, 2022-03-21:
**Describe the bug**
CI detected some IPC tx timed out errors on ADLP_NOCODEC_CI zephyr platform when test multiple pipeline related cases. 
The time out IDs are varied:
1. ipc tx: 0x30010000: GLB_TPLG_MSG: COMP_NEW
2. ipc tx: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
3. ipc tx: 0x30110000: GLB_TPLG_MSG: PIPE_FREE

**To Reproduce**
$ ~/sof-test/test-case/multiple-pipeline.sh -f c -l 50
$ ~/sof-test/test-case/multiple-pipeline.sh -f p -c 20 -l 50

**Reproduction Rate**
~50%

**Expected behavior**
No IPC related errors

**Impact**
IPC timed out

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: sof-dev/https://github.com/thesofproject/linux/commit/d781aecaed77
    * SOF: main/https://github.com/thesofproject/sof/commit/2c4e29187d18
2) Name of the topology file
    * Topology: sof-adl-nocodec.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: ADLP_RVP_NOCODEC_ZEPHYR

**Screenshots or console output**
```
2022-03-20 23:00:00 UTC [REMOTE_COMMAND] arecord   -D hw:0,2 -c 2 -r 48000 -f S16_LE /dev/null -q
2022-03-20 23:00:00 UTC [REMOTE_INFO] wait 0.5s for aplay_opts()
2022-03-20 23:00:00 UTC [REMOTE_INFO] checking pipeline status
error: in logger_read(), fread(..., /sys/kernel/debug/sof/trace) failed: Operation not permitted(-1)
2022-03-20 23:00:01 UTC [REMOTE_INFO] Letting playback/capture run for 5s
arecord: set_params:1432: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 32
CHANNELS: 2
RATE: 48000
PERIOD_TIME: (85333 85334)
PERIOD_SIZE: 4096
PERIOD_BYTES: 16384
PERIODS: 4
BUFFER_TIME: (341333 341334)
BUFFER_SIZE: 16384
BUFFER_BYTES: 65536
TICK_TIME: 0
aplay: pcm_write:2086: write error: Input/output error
arecord: pcm_read:2178: read error: Connection timed out
arecord: set_params:1432: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 32
CHANNELS: 2
RATE: 48000
PERIOD_TIME: (85333 85334)
PERIOD_SIZE: 4096
PERIOD_BYTES: 16384
PERIODS: 4
BUFFER_TIME: (341333 341334)
BUFFER_SIZE: 16384
BUFFER_BYTES: 65536
TICK_TIME: 0
[... 61 lines cut ...]
[ 1358.296736] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30110000: GLB_TPLG_MSG: PIPE_FREE
[ 1358.296825] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[ 1358.296833] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[ 1358.297792] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[ 1358.297801] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[ 1358.735845] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[ 1358.735897] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[ 1358.762825] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[ 1358.762859] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[ 1358.801589] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx timed out for 0x30110000 (msg/reply size: 12/12)
[ 1358.801838] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: preventing DSP entering D3 state to preserve context
[ 1358.801847] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump start ]------------
[ 1358.801869] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: hda irq intsts 0x00000000 intlctl 0xc0000081 rirb 00
[ 1358.801883] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: dsp irq ppsts 0x00000000 adspis 0x00000000
[ 1358.801900] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[ 1358.801911] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump end ]------------
[ 1358.801920] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 1358.801929] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware exception
[ 1358.801938] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (6)
[ 1358.801955] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: status: fw entered - code 00000005
[ 1358.802094] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: unexpected fault 0x00000000 trace 0x00004000
[ 1358.802108] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
```
[dmesg.txt](https://github.com/thesofproject/sof/files/8313116/dmesg.txt)
[logger.txt](https://github.com/thesofproject/sof/files/8313117/logger.txt)

kv2019i, 2022-03-21:
I suspect this is related to #5556 . Assigning to myself as I'm working on that.

kv2019i, 2022-03-25:
Potential fix to #5556 found. May fix this scenario as well as even in 5556 the bug had many different failure modes (not always the same IPC, sometimes xrun but not always, sometimes an OS panic, etc).

kv2019i, 2022-03-29:
No errors in Intel daily plan 11433, so potentially fixed by the fix to #5556.

kv2019i, 2022-04-01:
@keqiaozhang can we close? Not seen in Intel daily testing for multiple days.

keqiaozhang, 2022-04-01:
> @keqiaozhang can we close? Not seen in Intel daily testing for multiple days.

Yes, not seen this issue in CI this week. Closing this bug.

closed 2022-04-01 by keqiaozhang