thesofproject/sof #5594: [BUG][Zephyr] IPC tx timed out for TRIG_STOP when multiple-pipeline-capture on CML_RVP_NOCODEC

miRoox, 2022-03-24:
**Describe the bug**
CI (ID: 11355) detected an ipc tx timed out for TRIG_STOP when running multiple-pipeline-capture-50 on CML_RVP_NOCODEC.

**To Reproduce**
```
TPLG=/lib/firmware/intel/sof-tplg/sof-cnl-nocodec.tplg MODEL=CML_RVP_NOCODEC_ZEPHYR ~/sof-test/test-case/multiple-pipeline.sh -f c -l 50
```

**Reproduction Rate**
Only once in CI, didn't reproduce it manually.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: sof-dev/https://github.com/thesofproject/linux/commit/2e48d724eef4
    * SOF: main/https://github.com/thesofproject/sof/commit/f8934a9e2264
2) Name of the topology file
    * Topology: /lib/firmware/intel/sof-tplg/sof-cnl-nocodec.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: CML

**Screenshots or console output**

**dmesg**
```
[  700.899951] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: pcm: trigger stream 0 dir 1 cmd 0
[  700.899951] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: pcm: trigger stream 10 dir 1 cmd 0
[  700.899962] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  700.907768] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: pcm: trigger stream 1 dir 1 cmd 0
[  700.907771] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: pcm: trigger stream 11 dir 1 cmd 0
[  701.403943] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx timed out for 0x60050000 (msg/reply size: 12/12)
[  701.403964] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: preventing DSP entering D3 state to preserve context
[  701.403969] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ------------[ IPC dump start ]------------
[  701.403992] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: hda irq intsts 0x00000000 intlctl 0xc000001f rirb 00
[  701.404001] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: dsp irq ppsts 0x00000000 adspis 0x00000000
[  701.404021] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[  701.404029] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ------------[ IPC dump end ]------------
[  701.404036] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ------------[ DSP dump start ]------------
[  701.404042] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: Firmware exception
[  701.404048] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (6)
[  701.404067] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: status: fw entered - code 00000005
[  701.404161] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: unexpected fault 0x00000000 trace 0x00004000
[  701.404169] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ------------[ DSP dump end ]------------
[  701.404207] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0xa0]=0x20240000 successful
[  701.404222] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -110
[  701.404233] kernel:  DMIC: ASoC: trigger FE cmd: 0 failed: -110
[  701.404250] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: error: trace IO error
[  701.404272] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  701.404295] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: pcm: free stream 10 dir 1
[  701.908005] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx timed out for 0x60050000 (msg/reply size: 12/12)
[  701.908026] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: Firmware exception
[  701.908064] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0xe0]=0x20440000 successful
[  701.908067] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  701.908084] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -110
[  701.908096] kernel:  Port0: ASoC: trigger FE cmd: 0 failed: -110
[  702.412029] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx timed out for 0x60050000 (msg/reply size: 12/12)
[  702.412050] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: Firmware exception
[  702.412084] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0xc0]=0x20340000 successful
[  702.412096] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  702.412104] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -110
[  702.412119] kernel:  DMIC16kHz: ASoC: trigger FE cmd: 0 failed: -110
[  702.915994] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx timed out for 0x60050000 (msg/reply size: 12/12)
[  702.916003] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: Firmware exception
[  702.916024] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0x100]=0x20540000 successful
[  702.916030] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -110
[  702.916034] kernel:  Port1: ASoC: trigger FE cmd: 0 failed: -110
[  702.916056] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  703.419950] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx timed out for 0x60030000 (msg/reply size: 12/12)
[  703.419960] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: Firmware exception
[  703.419997] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x30020000: GLB_TPLG_MSG: COMP_FREE
[  703.923903] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx timed out for 0x30020000 (msg/reply size: 12/12)
[  703.923925] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: Firmware exception
[  703.923935] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: failed to free widget DMIC0.IN
[  703.923955] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x30210000: GLB_TPLG_MSG: BUFFER_FREE
[  704.427743] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx timed out for 0x30210000 (msg/reply size: 12/12)
[  704.427764] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: Firmware exception
[  704.427775] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: failed to free widget BUF20.2
[  704.427795] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x30020000: GLB_TPLG_MSG: COMP_FREE
[  704.931948] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx timed out for 0x30020000 (msg/reply size: 12/12)
[  704.931968] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: Firmware exception
[  704.931978] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: failed to free widget EQIIR20.0
[  704.931997] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x30210000: GLB_TPLG_MSG: BUFFER_FREE
[  705.435882] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx timed out for 0x30210000 (msg/reply size: 12/12)
[  705.435903] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: Firmware exception
[  705.435914] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: failed to free widget BUF20.1
```

lgirdwood, 2022-03-30:
@miRoox @ujfalusi not sure why we dont get a FW register dump here if we know there is a FW exception ?
```
> [  701.403943] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx timed out for 0x60050000 (msg/reply size: 12/12)
> [  701.403964] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: preventing DSP entering D3 state to preserve context
> [  701.403969] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ------------[ IPC dump start ]------------
> [  701.403992] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: hda irq intsts 0x00000000 intlctl 0xc000001f rirb 00
> [  701.404001] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: dsp irq ppsts 0x00000000 adspis 0x00000000
> [  701.404021] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
> [  701.404029] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ------------[ IPC dump end ]------------
> [  701.404036] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ------------[ DSP dump start ]------------
> [  701.404042] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: Firmware exception
```

ujfalusi, 2022-03-30:
My guess ia that Zephyr might not providing such? Timeout is 'normal' behaviour it might not be seen on the firmware side as an error at all.
I'll take a look, but if the firmware has not given us anything, we can not print anyrhing

miRoox, 2022-03-31:
What a similar conversation as before https://github.com/thesofproject/sof/issues/5103#issuecomment-999010301. Maybe the log message should be corrected?

miRoox, 2022-05-09:
This error wasn't observed again for a long time.
Reopen it if we find it again.

closed 2022-05-09 by miRoox