thesofproject/sof #5595: [BUG] glitches in the whole file during stress test with Dmic on JSL

kkarask, 2022-03-24:
**Describe the bug**
The whole file with glitches was found in test 05_03_TestDmicFifoBSimpleCapture48000hz32b32b4ch

Capture on DMIC interface FIFO B with signal verification.

**Topology**
```
  +----------+      +------------------+
    |    OS    |----->|   AUDIO DEVICE   |-----------------+
    +----------+      +------------------+                 |
                                                           |
                  +---------------------------+            |
    +----------+  | +------+   +------------+ |  +------+  |
    | Host cap |<---|Buffer|<--| Dai Fifo_B |<---| DMIC |<-+
    +----------+  | +------+   +------------+ |  +------+
                  +---------------------------+
```

**To Reproduce**
Run test 05_03_TestDmicFifoBSimpleCapture48000hz32b32b4ch with Diagnostic driver

**Reproduction Rate**
1/75

**Environment**
1) Firmware branch name and commit
    * SOF: Branch: v2.1-rc1, Hash: d93efb7a68676e23306e1468ff345318b0eade06

2) Name of the platform(s) on which the bug is observed.
    * Platform: JSL

**Screenshots or console output**
[image]
[JSL_05_03.zip](https://github.com/thesofproject/sof/files/8340900/JSL_05_03.zip)

singalsu, 2022-03-29:
Thanks, can you please also upload the wav file?

kkarask, 2022-03-31:
Wav file:
[TestDmicFifoBSimpleCapture48000hz32b32b4ch.test_2022_03_24_02_58_52_fifo_b_tplg0_iter18.wav.zip](https://github.com/thesofproject/sof/files/8389393/TestDmicFifoBSimpleCapture48000hz32b32b4ch.test_2022_03_24_02_58_52_fifo_b_tplg0_iter18.wav.zip)

wszypelt, 2022-12-30:
unsupported platform
on TGL issue does not occur
Tested FW hash: https://github.com/thesofproject/sof/commit/4943aaa8907149cf6c02e9046e2ef5d40ba72d74

closed 2022-12-30 by wszypelt