thesofproject/sof #5597: [FEATURE] Disable trace suppression for select messages

cujomalainey, 2022-03-24:
**Is your feature request related to a problem? Please describe.**
Suppression makes sense for most duplicate messages, but there are some cases where it is not helpful, e.g. a heap trace

**Describe the solution you'd like**
Special traces that cannot be suppressed that are not errors

**Describe alternatives you've considered**
Increasing the log level of the messages

**Additional context**
```
[      390619.671978] (         142.500000) c0 memory                      src/lib/alloc.c:602  INFO heap: runtime shared status
[      390636.963644] (          17.291666) c0 memory                      src/lib/alloc.c:567  INFO  heap: 0x9e19f000 size 229376 blocks 5 caps 0x45
[      390654.567810] (          17.604166) c0 memory                      src/lib/alloc.c:569  INFO   used 31296 free 198080
[      390807.536554] (         152.968750) c0 memory                      src/lib/alloc.c:604  INFO heap: system shared status
[      390820.921970] (          13.385416) c0 dma-trace                 src/trace/trace.c:137  INFO Suppressed 19 similar messages:   block %d base 0x%x size %d
[      390836.442803] (          15.520833) c0 memory                      src/lib/alloc.c:567  INFO  heap: 0x9e1d7000 size 5376 blocks 0 caps 0x45
[      390852.119886] (          15.677083) c0 dma-trace                 src/trace/trace.c:137  INFO Suppressed 22 similar messages:    count %d free %d
[      390865.713635] (          13.593749) c0 memory                      src/lib/alloc.c:569  INFO   used 3840 free 1536

```

ranj063, 2022-03-24:
@cujomalainey does increasing CONFIG_TRACE_BURST_COUNT=4 help at all?

cujomalainey, 2022-03-24:
@ranj063 the count in the example there is 22 and 19.

ranj063, 2022-03-24:
I see how turning off adaptive filtering. TRACE_FILTERING_ADAPTIVE? Not sure if this is even allowed

cujomalainey, 2022-03-24:
Yea, just seems counter when trying to test a production build

lgirdwood, 2022-03-25:
@cujomalainey can we work around this with a tmp fix as the Zephyr logging has different infrastructure ?

cujomalainey, 2022-03-25:
This is low priority so I'm fine waiting until we switch to zephyr to minimize effort

marc-hb, 2023-05-29:
New issue about Zephyr dictionary:
- #7703

lgirdwood, 2023-07-04:
This will come from Zephyr. @kv2019i not sure of Zephyr feature availability.

marc-hb, 2023-07-05:
> @cujomalainey does increasing CONFIG_TRACE_BURST_COUNT=4 help at all?

4 is the default. Did you try 30?

You can also lower the TRACE_FILTERING_ADAPTIVE expiration to smaller number of microseconds.

PS: as just discussed in #4672, Zephyr does not seem to have any throttling feature.

cujomalainey, 2023-07-11:
@eddy1021 can you test the memory dump on a ADL board with that config?