thesofproject/sof #5602: [FEATURE][Zephyr] Move all hw init code from platform_init() to Zephyr

kv2019i, 2022-03-25:
**Is your feature request related to a problem? Please describe.**
Much of the HW init code is already ignored in Zephyr builds in the Zephyr main() -> sof_main() -> primary_core_init() -> platform_init() code path, but not all. This work should be completed.

Examples like platform_init_hw() must be moved to Zephyr side and not the SOF definition should not be used when building for Zephyr.

This has created multiple problems for unit tests on Zephyr side (like for cAVS SSP driver, FYI @juimonen  ).

lenghonglin, 2022-03-30:
see this https://github.com/thesofproject/sof/issues/5068#issuecomment-1011514361

lgirdwood, 2022-04-06:
Making a v2.2 for Intel platforms. @dbaluta NXP likewise for v2.2 ?

kv2019i, 2023-03-10:
@dbaluta I don't see any clear overlap in platform/imx8/platform.c where SOF side platform code would be initializing hw that is already handle in Zephyr. IMX8 interrupt handling  is one possible item, but I guess this is not overlapping with Zephyr yet (it's really handled on SOF side). Feel free to yet in the task tracking list.

lgirdwood, 2023-05-10:
@juimonen @dbaluta on track for v2.6 ?

kv2019i, 2023-08-17:
Moving this meta item to v2.8. For v2.7, we completed https://github.com/thesofproject/sof/pull/7896 and start to be completed for Intel targets.