thesofproject/sof #5619: [BUG]The first frame of recording is dirty data

lenghonglin, 2022-03-30:
**Describe the bug**
The first frame of recording is dirty data. host copy use `dummy-copy`

**To Reproduce**
capture

**Reproduction Rate**
100%

**Impact**
will produce dirty data.

**Environment**
    * SOF: 1.8

[image]

the question is host->dma_buffer is updated after copy data to ringbuffer, for first frame, the data is dirty data.

Is this a normal phenomenon? Why is it designed this way?

dbaluta, 2022-03-30:
@paulstelian97 have a look at this once you have some spare cycles.

lenghonglin, 2022-03-31:
@paulstelian97  hello, could you have a look

paulstelian97, 2022-03-31:
I will look at this when I have some spare time on my hand, but I make no promise of how soon it will happen.

dbaluta, 2022-03-31:
@lenghonglin since we are very busy can you help us further investigate this issue? 

Is the first frame of recording dirty data or just zeroes? 

Are you using latest SOF code? 

is this commit relevant?

```
commit 87e4f5aad80c22a1e82fa2bf78133640e39b8741
Author: Daniel Baluta <daniel.baluta@nxp.com>
Date:   Mon Aug 17 17:09:29 2020 +0300

    host: Fix first period not being copied to/from host

```

lenghonglin, 2022-04-01:
> 
dirty data not zero data.

the sof version is v1.8

and my point is we should update `host->dma_buffer` then copy `host->dma_buffer` to `ringbuffer`.

lgirdwood, 2022-04-05:
Does dummy_copy need to bzero() buffers prior to starting ?

lenghonglin, 2022-04-06:
> 

https://github.com/thesofproject/sof/blob/7b5aa7281370c287a16600c7b010279c3fd81def/src/audio/host.c#L926 look here

lgirdwood, 2022-04-06:
> > 
> 
> https://github.com/thesofproject/sof/blob/7b5aa7281370c287a16600c7b010279c3fd81def/src/audio/host.c#L926
> 
> look here

@lenghonglin does your platform need to writeback or invalidate any cache here ?

lenghonglin, 2022-04-06:
> > > 
> > 
> > 
> > https://github.com/thesofproject/sof/blob/7b5aa7281370c287a16600c7b010279c3fd81def/src/audio/host.c#L926
> > 
> > look here
> 
> @lenghonglin does your platform need to writeback or invalidate any cache here ?

i wanna know why update copy `host->dma_buffer` to ringbuffer and then update `host->dma_buffer`

lgirdwood, 2022-04-06:
> > > > 
> > > 
> > > 
> > > https://github.com/thesofproject/sof/blob/7b5aa7281370c287a16600c7b010279c3fd81def/src/audio/host.c#L926
> > > 
> > > look here
> > 
> > 
> > @lenghonglin does your platform need to writeback or invalidate any cache here ?
> 
> i wanna know why update copy `host->dma_buffer` to ringbuffer and then update `host->dma_buffer`

Sorry, not following. Do you mean the host ALSA ring buffer to DSP DMA ring buffer ?

lenghonglin, 2022-04-06:
> Sorry, not following. Do you mean the host ALSA ring buffer to DSP DMA ring buffer ?

Yes

closed 2025-07-25 by github-actions[bot]