thesofproject/sof #5626: [BUG] xrun on TGLU_RVP_NOCODEC when testing multiple-pipeline-all

keqiaozhang, 2022-03-31:
**Describe the bug**
CI detected the xrun issue on TGLU_RVP_NOCODEC without IPC times out or error. only xrun error reported in logger trace.
there're also 

**To Reproduce**
TPLG=/lib/firmware/intel/sof-tplg/sof-tgl-nocodec.tplg MODEL=TGLU_RVP_NOCODEC ~/sof-test/test-case/multiple-pipeline.sh -f p -c 20 -l 50

**Reproduction Rate**
Not sure , haven't reproduced yet 

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: [029b208dc0ce](https://github.com/thesofproject/linux/commit/029b208dc0ce)
SOF Branch: main
SOF Commit:[a5fd52ec43f3](https://github.com/thesofproject/sof/commit/a5fd52ec43f3)

**Screenshots or console output**
```
2022-03-30 23:21:41 UTC [REMOTE_INFO] /home/ubuntu/sof-test/test-case/multiple-pipeline.sh will use topology /usr/lib/firmware/intel/sof-tplg/sof-tgl-nocodec.tplg to run the test case
2022-03-30 23:21:41 UTC [REMOTE_INFO] Pipeline list to ignore is specified, will ignore 'pcm=HDA Digital' in test case
2022-03-30 23:21:41 UTC [REMOTE_INFO] Run command to get pipeline parameters
2022-03-30 23:21:41 UTC [REMOTE_COMMAND] sof-tplgreader.py /usr/lib/firmware/intel/sof-tplg/sof-tgl-nocodec.tplg -f 'type:capture & ~pcm:Amplifier Reference' -b ' pcm:HDA Digital' -s 0 -e
2022-03-30 23:21:41 UTC [REMOTE_INFO] Testing: DMIC [hw:0,10]
2022-03-30 23:21:41 UTC [REMOTE_COMMAND] arecord   -D hw:0,10 -c 4 -r 48000 -f S16_LE /dev/null -q
2022-03-30 23:21:41 UTC [REMOTE_INFO] Testing: DMIC16kHz [hw:0,11]
2022-03-30 23:21:41 UTC [REMOTE_COMMAND] arecord   -D hw:0,11 -c 4 -r 16000 -f S16_LE /dev/null -q
2022-03-30 23:21:41 UTC [REMOTE_INFO] Testing: Port0 [hw:0,0]
2022-03-30 23:21:41 UTC [REMOTE_COMMAND] arecord   -D hw:0,0 -c 2 -r 48000 -f S16_LE /dev/null -q
2022-03-30 23:21:41 UTC [REMOTE_INFO] Testing: Port1 [hw:0,1]
2022-03-30 23:21:41 UTC [REMOTE_COMMAND] arecord   -D hw:0,1 -c 2 -r 48000 -f S16_LE /dev/null -q
2022-03-30 23:21:41 UTC [REMOTE_INFO] Testing: Port2 [hw:0,2]
2022-03-30 23:21:41 UTC [REMOTE_COMMAND] arecord   -D hw:0,2 -c 2 -r 48000 -f S16_LE /dev/null -q
2022-03-30 23:21:41 UTC [REMOTE_INFO] wait 0.5s for aplay_opts()
2022-03-30 23:21:42 UTC [REMOTE_INFO] checking pipeline status
2022-03-30 23:21:42 UTC [REMOTE_INFO] Letting playback/capture run for 5s
aplay: pcm_write:2086: write error: Input/output error
aplay: pcm_write:2086: write error: Input/output error
2022-03-30 23:21:47 UTC [REMOTE_INFO] checking pipeline status again
2022-03-30 23:21:47 UTC [REMOTE_ERROR] Running process count is 7, but 9 is expected
68776 aplay -D hw:0,1 -c 2 -r 48000 -f S16_LE /dev/zero -q
68785 aplay -D hw:0,2 -c 2 -r 48000 -f S16_LE /dev/zero -q
68813 arecord -D hw:0,10 -c 4 -r 48000 -f S16_LE /dev/null -q
68822 arecord -D hw:0,11 -c 4 -r 16000 -f S16_LE /dev/null -q
68832 arecord -D hw:0,0 -c 2 -r 48000 -f S16_LE /dev/null -q
68841 arecord -D hw:0,1 -c 2 -r 48000 -f S16_LE /dev/null -q
68849 arecord -D hw:0,2 -c 2 -r 48000 -f S16_LE /dev/null -q
/home/ubuntu/sof-test/test-case/../case-lib/lib.sh: line 397: 68776 Killed                  aplay $SOF_ALSA_OPTS $SOF_APLAY_OPTS "$@"
/home/ubuntu/sof-test/test-case/../case-lib/lib.sh: line 397: 68785 Killed                  aplay $SOF_ALSA_OPTS $SOF_APLAY_OPTS "$@"
2022-03-30 23:21:47 UTC [REMOTE_ERROR] Starting func_exit_handler(), exit status=1, FUNCNAME stack:
2022-03-30 23:21:47 UTC [REMOTE_ERROR]  func_error_exit()  @  /home/ubuntu/sof-test/test-case/multiple-pipeline.sh
2022-03-30 23:21:47 UTC [REMOTE_ERROR]  ps_checks()  @  /home/ubuntu/sof-test/test-case/multiple-pipeline.sh:144
2022-03-30 23:21:47 UTC [REMOTE_ERROR]  main()  @  /home/ubuntu/sof-test/test-case/multiple-pipeline.sh:198
/home/ubuntu/sof-test/test-case/../case-lib/lib.sh: line 403: 68832 Killed                  arecord $SOF_ALSA_OPTS $SOF_ARECORD_OPTS "$@"
/home/ubuntu/sof-test/test-case/../case-lib/lib.sh: line 403: 68813 Killed                  arecord $SOF_ALSA_OPTS $SOF_ARECORD_OPTS "$@"
/home/ubuntu/sof-test/test-case/../case-lib/lib.sh: line 403: 68822 Killed                  arecord $SOF_ALSA_OPTS $SOF_ARECORD_OPTS "$@"
/home/ubuntu/sof-test/test-case/../case-lib/lib.sh: line 403: 68849 Killed                  arecord $SOF_ALSA_OPTS $SOF_ARECORD_OPTS "$@"
/home/ubuntu/sof-test/test-case/../case-lib/lib.sh: line 403: 68841 Killed                  arecord $SOF_ALSA_OPTS $SOF_ARECORD_OPTS "$@"
2022-03-30 23:21:47 UTC [REMOTE_INFO] Starting /usr/local/bin/sof-logger -l /etc/sof/sof-tgl.ldc -o /home/ubuntu/sof-test/logs/multiple-pipeline/2022-03-30-23:18:30-29511/etrace.txt
2022-03-30 23:21:50 UTC [REMOTE_INFO] pkill -TERM sof-logger
Terminated
2022-03-30 23:21:51 UTC [REMOTE_INFO] nlines=95 /home/ubuntu/sof-test/logs/multiple-pipeline/2022-03-30-23:18:30-29511/etrace.txt
2022-03-30 23:21:51 UTC [REMOTE_INFO] Test Result: FAIL!
```
** trace **
```
[   377765794.468103] (       34020.257812) c0 dma-trace                    src/lib/wait.c:51   ERROR poll timeout reg 488520 mask 63 val 0 us 937
[   380456182.694530] (     2690388.250000) c2 dma-trace              src/drivers/dw/dma.c:1126 ERROR dw_dma_get_data_size(): xrun detected
[   380456206.496612] (          23.802082) c2 dma-trace                   src/audio/dai.c:879  ERROR dai_report_xrun(): underrun due to no data available
[   380456228.267444] (          21.770832) c2 dma-trace                   src/audio/dai.c:748  ERROR comp_underrun(): dev->comp.id = 2, source->avail = 192, copy_bytes = 0
[   380456296.132025] (          67.864578) c2 dma-trace          ....../pipeline-stream.c:134  ERROR pipeline_copy(): ret = -61, start->comp.id = 2, dir = 1
[   380456319.100774] (          22.968750) c2 dma-trace          ..../pipeline-schedule.c:183  ERROR pipeline_task(): xrun recovery failed! pipeline is stopped.
[   380927917.988285] (      471598.875000) c2 dma-trace             src/trace/dma-trace.c:649  ERROR dtrace_add_event(): number of dropped logs = 7
[   382347901.890193] (     1419983.875000) c2 dma-trace             src/trace/dma-trace.c:649  ERROR dtrace_add_event(): number of dropped logs = 209
[   383864284.590354] (     1516382.750000) c0 dma-trace             src/trace/dma-trace.c:649  ERROR dtrace_add_event(): number of dropped logs = 158
[   391755829.589272] (     7891545.000000) c0 dma-trace                    src/lib/wait.c:51   ERROR poll timeout reg 488520 mask 63 val 0 us 937
[   391759793.859948] (        3964.270752) c3 dma-trace                    src/lib/wait.c:51   ERROR poll timeout reg 488008 mask 63 val 0 us 937
Skipped 4960 bytes after the last statement.
```
** logger **
```
[   380451708.007207] (          25.156248) c2 idc                ......./intel/cavs/idc.c:186  INFO idc_do_cmd()
[   380451740.403039] (          32.395832) c2 pipe         11.55 ....../pipeline-stream.c:261  INFO pipe trigger cmd 7
[   380451982.538446] (         242.135406) c2 dai          1.2            src/audio/dai.c:949  WARN dai_copy(): nothing to copy
[   380452040.611361] (          58.072914) c2 ll-schedule        ./schedule/ll_schedule.c:500  INFO task add 0xbe1e0a00 pipe-task 
[   380452059.309277] (          18.697916) c2 ll-schedule        ./schedule/ll_schedule.c:504  INFO task params pri 0 flags 0 start 0 period 1000
[   380452087.382192] (          28.072916) c2 ll-schedule        ./schedule/ll_schedule.c:405  INFO new added task->start 3010379207 at 3010346844
[   380452108.736358] (          21.354166) c2 ll-schedule        ./schedule/ll_schedule.c:408  INFO num_tasks 3 total_num_tasks 9
[   380453980.350867] (        1871.614502) c2 mixer        1.0      src/audio/component.c:64   INFO comp_set_state(), state already set to 5

[   380456243.684111] (          17.447916) c2 pipe         1.3   ....../pipeline-stream.c:261  INFO pipe trigger cmd 6

[   380456336.236190] (          18.958332) c2 ll-schedule        ./schedule/ll_schedule.c:147  INFO task complete 0xbe1e0280 pipe-task [   380456182.277863] (        2201.927002) c2 dw-dma                 src/drivers/dw/dma.c:1126 ERROR dw_dma_get_data_size(): xrun detected[   380456204.048695] (          21.770832) c2 dai          1.2            src/audio/dai.c:879  ERROR dai_report_xrun(): underrun due to no data available[   380456226.236195] (          22.187500) c2 dai          1.2            src/audio/dai.c:748  ERROR comp_underrun(): dev->comp.id = 2, source->avail = 192, copy_bytes = 0[   380456295.298692] (          51.614582) c2 pipe         1.3   ....../pipeline-stream.c:134  ERROR pipeline_copy(): ret = -61, start->comp.id = 2, dir = 1[   380456317.277858] (          21.979166) c2 pipe         1.3   ..../pipeline-schedule.c:183  ERROR pipeline_task(): xrun recovery failed! pipeline is stopped.
```

[dmesg.txt](https://github.com/thesofproject/sof/files/8385777/dmesg.txt)
[logger.txt](https://github.com/thesofproject/sof/files/8385778/logger.txt)

kv2019i, 2022-04-01:
I believe this has same rootcause as #5469 . I have a set of PRs in review that will address this issue.

keqiaozhang, 2022-04-18:
Haven't seen this issue in CI for 2 weeks, all related fixes have been merged. Closing this bug.

closed 2022-04-18 by keqiaozhang