thesofproject/sof #5650: [BUG] IPC timed out for 0x60040000(TRIG_START) when testing multiple-pause-resume on BYT_MB_NOCODEC

keqiaozhang, 2022-04-06:
**Describe the bug**
CI detected this IPC timed out in our daily test. Test ID:11602
But this issue cannot be reproduced manually. 
error:
```
[ 1541.721947] kernel: sof-audio-acpi-intel-byt 80860F28:00: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[ 1541.722602] kernel: sof-audio-acpi-intel-byt 80860F28:00: ipc tx succeeded: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[ 1541.722619] kernel: sof-audio-acpi-intel-byt 80860F28:00: pcm: stream dir 0, posn mailbox offset is 1329664
[ 1541.725025] kernel: sof-audio-acpi-intel-byt 80860F28:00: pcm: trigger stream 1 dir 0 cmd 1
[ 1541.725052] kernel: sof-audio-acpi-intel-byt 80860F28:00: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[ 1542.199826] kernel: sof-audio-acpi-intel-byt 80860F28:00: pcm: trigger stream 0 dir 1 cmd 0
[ 1542.231179] kernel: sof-audio-acpi-intel-byt 80860F28:00: ipc tx timed out for 0x60040000 (msg/reply size: 12/12)
[ 1542.231483] kernel: sof-audio-acpi-intel-byt 80860F28:00: preventing DSP entering D3 state to preserve context
[ 1542.231508] kernel: sof-audio-acpi-intel-byt 80860F28:00: ------------[ DSP dump start ]------------
[ 1542.231540] kernel: sof-audio-acpi-intel-byt 80860F28:00: Firmware exception
[ 1542.231570] kernel: sof-audio-acpi-intel-byt 80860F28:00: fw_state: SOF_FW_BOOT_COMPLETE (6)
[ 1542.231664] kernel: sof-audio-acpi-intel-byt 80860F28:00: unexpected fault 0x90020000 trace 0x00000000
[ 1542.231701] kernel: sof-audio-acpi-intel-byt 80860F28:00: error: ipc host -> DSP: pending no complete no raw 0x8000000000000000
[ 1542.231735] kernel: sof-audio-acpi-intel-byt 80860F28:00: error: mask host: pending no complete no raw 0x0
[ 1542.231901] kernel: sof-audio-acpi-intel-byt 80860F28:00: error: ipc DSP -> host: pending yes complete no raw 0x90020000
[ 1542.231937] kernel: sof-audio-acpi-intel-byt 80860F28:00: error: mask DSP: pending no complete no raw 0x0
[ 1542.231969] kernel: sof-audio-acpi-intel-byt 80860F28:00: ------------[ DSP dump end ]------------
[ 1542.232063] kernel: sof-audio-acpi-intel-byt 80860F28:00: trace IO error
[ 1542.241557] kernel: sof-audio-acpi-intel-byt 80860F28:00: ASoC: error at soc_component_trigger on 80860F28:00: -110
[ 1542.241586] kernel:  PCM Deep Buffer: ASoC: trigger FE cmd: 1 failed: -110
```

**To Reproduce**
$ /sof-test/test-case/multiple-pause-resume.sh -r 50

**Reproduction Rate**
Hard to tell, only observed this issue once in CI.

**Expected behavior**
No IPC errors

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: [842216535791](https://github.com/thesofproject/linux/commit/842216535791)

SOF Branch: main
SOF Commit: [7b5aa7281370](https://github.com/thesofproject/sof/commit/7b5aa7281370)

[dmesg.txt](https://github.com/thesofproject/sof/files/8424530/dmesg.txt)

keqiaozhang, 2022-04-07:
Observed the similar issue on BDW_WSB_RT286 when testing check-playback-100times. IPC timed out for 0x60050000(TRIG_STOP).  Daily test ID:11656
Dmesg:
```
[  616.808356] kernel: sof-audio-acpi-intel-bdw INT3438:00: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[  616.808495] kernel: sof-audio-acpi-intel-bdw INT3438:00: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
[  627.275981] kernel: sof-audio-acpi-intel-bdw INT3438:00: pcm: trigger stream 1 dir 0 cmd 0
[  627.276004] kernel: sof-audio-acpi-intel-bdw INT3438:00: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  627.779975] kernel: sof-audio-acpi-intel-bdw INT3438:00: ipc tx timed out for 0x60050000 (msg/reply size: 12/12)
[  627.780041] kernel: sof-audio-acpi-intel-bdw INT3438:00: preventing DSP entering D3 state to preserve context
[  627.780049] kernel: sof-audio-acpi-intel-bdw INT3438:00: ------------[ DSP dump start ]------------
[  627.780092] kernel: sof-audio-acpi-intel-bdw INT3438:00: Firmware exception
[  627.780126] kernel: sof-audio-acpi-intel-bdw INT3438:00: fw_state: SOF_FW_BOOT_COMPLETE (6)
[  627.780242] kernel: sof-audio-acpi-intel-bdw INT3438:00: unexpected fault 0x00000000 trace 0x80000000
[  627.780289] kernel: sof-audio-acpi-intel-bdw INT3438:00: error: ipc host -> DSP: pending yes complete no raw 0x80000000
[  627.780340] kernel: sof-audio-acpi-intel-bdw INT3438:00: error: mask host: pending no complete no raw 0x7fff0000
[  627.780388] kernel: sof-audio-acpi-intel-bdw INT3438:00: error: ipc DSP -> host: pending no complete no raw 0x00000000
[  627.780438] kernel: sof-audio-acpi-intel-bdw INT3438:00: error: mask DSP: pending no complete no raw 0x7f9e0000
[  627.780486] kernel: sof-audio-acpi-intel-bdw INT3438:00: ------------[ DSP dump end ]------------
```

mengdonglin, 2022-09-06:
Closing because the extremely low reproduce rate.

closed 2022-09-06 by mengdonglin