thesofproject/sof #5657: [BUG][zephyr][CML] IPC tx timed out for 0x60010000(PCM_PARAMS) when testing multiple-pipeline-capture

keqiaozhang, 2022-04-07:
**Describe the bug**
Observed this issue in CI daily test 11656 and this issue only happened once and can not be reproduced manually.
Error:
```
[ 5016.197789] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx succeeded: 0x30130000: GLB_TPLG_MSG: PIPE_COMPLETE
[ 5016.197797] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: stream_tag 5
[ 5016.197807] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[ 5016.198075] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[ 5016.198091] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[ 5016.690980] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: pcm: trigger stream 10 dir 1 cmd 0
[ 5016.690980] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: pcm: trigger stream 11 dir 1 cmd 0
[ 5016.698924] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx timed out for 0x60010000 (msg/reply size: 108/20)
[ 5016.698972] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: pcm: trigger stream 0 dir 1 cmd 0
[ 5016.698995] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: preventing DSP entering D3 state to preserve context
[ 5016.698998] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ------------[ IPC dump start ]------------
[ 5016.699010] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: hda irq intsts 0x00000000 intlctl 0xc000000f rirb 00
[ 5016.699014] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: dsp irq ppsts 0x00000000 adspis 0x00000000
[ 5016.699025] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[ 5016.699029] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ------------[ IPC dump end ]------------
[ 5016.699031] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 5016.699034] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: Firmware exception
[ 5016.699037] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (6)
[ 5016.699044] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: status: fw entered - code 00000005
[ 5016.699124] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: unexpected fault 0x00000000 trace 0x00004000
[ 5016.699127] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ------------[ DSP dump end ]------------
```

**To Reproduce**
~/sof-test/test-case/multiple-pipeline.sh -f c -l 50

**Reproduction Rate**
hard to tell, only observed it once in CI.

**Expected behavior**
No ipc error

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: [8e9f9fcfcb85](https://github.com/thesofproject/linux/commit/8e9f9fcfcb85)
SOF Branch: main
SOF Commit: [7b5aa7281370](https://github.com/thesofproject/sof/commit/7b5aa7281370)
Zephyr Commit: [a52aa7f2ab98](https://github.com/zephyrproject-rtos/zephyr/commit/a52aa7f2ab98)
Platform:CML_RVP_NOCODEC_ZEPHYR

[logger.txt](https://github.com/thesofproject/sof/files/8439332/logger.txt)
[dmesg.txt](https://github.com/thesofproject/sof/files/8439333/dmesg.txt)

lgirdwood, 2022-04-07:
@kv2019i duplicate ?

kv2019i, 2022-04-08:
This may be something different. Let me take a look.

keqiaozhang, 2022-05-23:
We haven't seen this issue in CI in the past month and a half. Let's close this one and reopen if it happens again.

closed 2022-05-23 by keqiaozhang