thesofproject/sof #5674: [BUG] Apparently buggy code in codec_adapter.c

lyakh, 2022-04-11:
**Describe the bug**
Loop

https://github.com/thesofproject/sof/blob/5eb94a8a801ffa4922f1f173180751b70528b776/src/audio/codec_adapter/codec_adapter.c#L512-L528

introduced by 4f8a70dd23 appears to contain a bug: the `i` index is never incremented in it.

**Expected behavior**
Presumably `i` should be incremented after line 527.

dbaluta, 2022-04-11:
@ranj063  @lyakh probably this works because we only have 1 input buffer and 1 output buffer?

lgirdwood, 2022-04-13:
> @ranj063 @lyakh probably this works because we only have 1 input buffer and 1 output buffer?

Ack - I think @ranj063 had a plan to do all changes with 1:1 buffers and then add the m:n support.

lyakh, 2023-09-25:
fixed in the current version of `module_adapter_process_output()`

closed 2023-09-25 by lyakh