thesofproject/sof #5678: [BUG] DMIC 4ch capture is NOT working

fredoh9, 2022-04-12:
**Describe the bug**
This is iPC4 + CAVS FW. I tested with ADL-P RVP + Realtek AIOC + Transducer board. But I don't think this is platform specific.
DMIC 4 ch capture is NOT working.
DMIC 2 ch capture is working fine.

**To Reproduce**
arecord -Dhw:0,6 -r 48000 -c 4 -f S32_LE -d 10 test_dmic4ch.wav -vv

**Reproduction Rate**
100%

**Expected behavior**
DMIC 4ch shall create a recorded file without any problem

**Impact**
No 4ch DMIC recording.

**Environment**
Kernel : topic/sof-dev, 9ca6d6e0dfc6
CAVS FW is used, avs/tgl/dsp_basefw.bin
SOF : main, 5243f0c4ee92 for topology
Tplg: cavs-mixin-mixout-hda.tplg

**Screenshots or console output**
```
[    6.017511] sof-audio-pci-intel-tgl 0000:00:1f.3: intel_nhlt_get_dmic_geo: found 2 format definitions
[    6.017512] sof-audio-pci-intel-tgl 0000:00:1f.3: intel_nhlt_get_dmic_geo: max channels found 4
[    6.017513] sof-audio-pci-intel-tgl 0000:00:1f.3: intel_nhlt_get_dmic_geo: Array with 4 dmics
[    6.017514] sof-audio-pci-intel-tgl 0000:00:1f.3: intel_nhlt_get_dmic_geo: dmic number 4 max_ch 4
[    6.017515] sof-audio-pci-intel-tgl 0000:00:1f.3: overriding DMICs detected in NHLT tables 4 by kernel param 2
[    6.017517] sof-audio-pci-intel-tgl 0000:00:1f.3: DMICs detected in NHLT tables: 2
```
For above log, BIOS setting has still 4 ch DMIC enabled, I tried to override it with 2ch by the module param. I had the same error. Sorry for the confusion if you have.

```
[  103.127177] sof-audio-pci-intel-tgl 0000:00:1f.3: Endpoint format: ch=4 fmt=16/16 rate=48000
[  103.127181] sof-audio-pci-intel-tgl 0000:00:1f.3: Endpoint format: ch=4 fmt=32/32 rate=48000
[  103.127185] sof-audio-pci-intel-tgl 0000:00:1f.3: Endpoint: vbus_id=2 link_type=3 dir=0 dev_type = 0
[  103.127189] sof-audio-pci-intel-tgl 0000:00:1f.3: Endpoint: vbus_id=2 link_type=3 dir=1 dev_type = 0
[  103.127193] sof-audio-pci-intel-tgl 0000:00:1f.3: no matching blob for sample rate: 48000 sample width: 32 channels: 2
[  103.127402] sof-audio-pci-intel-tgl 0000:00:1f.3: failed to prepare widget copier.DMIC.12.1
[  103.127409] sof-audio-pci-intel-tgl 0000:00:1f.3: Failed to prepare connected widgets
[  103.127415] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed widget list set up for pcm 10 dir 1
[  103.127421] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -22
[  103.127464]  DMIC: ASoC: __soc_pcm_hw_params() failed (-22)
[  103.127481]  DMIC: ASoC: dpcm_fe_dai_hw_params failed (-22)
```

Full dmesg log : (https://github.com/thesofproject/sof/files/8468442/0408_dmesg02_hw_params_4ch_error.log)

fredoh9, 2022-04-12:
@aiChaoSONG yes, IPC3 works on same device. I updated topology I used above again.
cavs-mixin-mixout-hda-4ch-tgl and cavs-mixin-mixout-hda-2ch-tgl are used.

I will capture log today again.

fredoh9, 2022-04-12:
I re-captred the log, BIOS setting has DMIC 4ch 38.4MHz for NHLT, no module param set.

[0412_dmesg02_bios4ch+arecord_dmic_fail.log](https://github.com/thesofproject/sof/files/8475330/0412_dmesg02_bios4ch%2Barecord_dmic_fail.log)

```
[    5.518777] sof-audio-pci-intel-tgl 0000:00:1f.3: intel_nhlt_get_dmic_geo: found 2 format definitions
[    5.518778] sof-audio-pci-intel-tgl 0000:00:1f.3: intel_nhlt_get_dmic_geo: max channels found 4
[    5.518778] sof-audio-pci-intel-tgl 0000:00:1f.3: intel_nhlt_get_dmic_geo: Array with 4 dmics
[    5.518780] sof-audio-pci-intel-tgl 0000:00:1f.3: intel_nhlt_get_dmic_geo: dmic number 4 max_ch 4
[    5.518782] sof-audio-pci-intel-tgl 0000:00:1f.3: DMICs detected in NHLT tables: 4
[    5.518803] sof-audio-pci-intel-tgl 0000:00:1f.3: loading firmware
[    5.520863] sof-audio-pci-intel-tgl 0000:00:1f.3: request_firmware intel/avs/adl/dsp_basefw.bin successful
...
[    5.626944] sof-audio-pci-intel-tgl 0000:00:1f.3: loading topology:intel/avs-tplg/sof-hda-generic-4ch.tplg
...
[   91.505586] sof-audio-pci-intel-tgl 0000:00:1f.3: sof_ipc4_prepare_copier_module: copier copier.DMIC.12.1, type 28
[   91.505621] sof-audio-pci-intel-tgl 0000:00:1f.3: snd_sof_get_nhlt_endpoint_data: dai index 0 nhlt type 2 direction 1
[   91.505626] sof-audio-pci-intel-tgl 0000:00:1f.3: Looking for configuration:
[   91.505629] sof-audio-pci-intel-tgl 0000:00:1f.3:   vbus_id=0 link_type=2 dir=1, dev_type=0
[   91.505633] sof-audio-pci-intel-tgl 0000:00:1f.3:   ch=2 fmt=32/32 rate=48000
[   91.505639] sof-audio-pci-intel-tgl 0000:00:1f.3: Endpoint count=3
[   91.505642] sof-audio-pci-intel-tgl 0000:00:1f.3: Endpoint: vbus_id=0 link_type=2 dir=1 dev_type = 0
[   91.505646] sof-audio-pci-intel-tgl 0000:00:1f.3: Endpoint format count=2
[   91.505649] sof-audio-pci-intel-tgl 0000:00:1f.3: Endpoint format: ch=4 fmt=16/16 rate=48000
[   91.505653] sof-audio-pci-intel-tgl 0000:00:1f.3: Endpoint format: ch=4 fmt=32/32 rate=48000
[   91.505657] sof-audio-pci-intel-tgl 0000:00:1f.3: Endpoint: vbus_id=2 link_type=3 dir=0 dev_type = 0
[   91.505661] sof-audio-pci-intel-tgl 0000:00:1f.3: Endpoint: vbus_id=2 link_type=3 dir=1 dev_type = 0
[   91.505681] sof-audio-pci-intel-tgl 0000:00:1f.3: no matching blob for sample rate: 48000 sample width: 32 channels: 2
[   91.505780] sof-audio-pci-intel-tgl 0000:00:1f.3: failed to prepare widget copier.DMIC.12.1
[   91.505788] sof-audio-pci-intel-tgl 0000:00:1f.3: Failed to prepare connected widgets
[   91.505796] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed widget list set up for pcm 10 dir 1
[   91.505803] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -22
[   91.505850]  DMIC: ASoC: __soc_pcm_hw_params() failed (-22)
[   91.505868]  DMIC: ASoC: dpcm_fe_dai_hw_params failed (-22)
```

juimonen, 2022-04-12:
@fredoh9 so we are trying to look for 2 channel blob, which doesn't exist. The channel count is coming from snd_pcm_hw_params... so are you specifying the 4 channel with arecord? if that doesn't work, we probably have somewhere in the pcm topology definition a 2 channel limitation.

juimonen, 2022-04-12:
@fredoh9 you could try to set channels_max to 4 in include/pipelines/cavs/passthrough-capture.conf and compile topologies and see what happens... not sure if that would work

ranj063, 2022-04-12:
@fredoh9 @juimonen i'm working on the tplg fix for this. Will post a PR soon

closed 2022-04-19 by lgirdwood