thesofproject/sof #5690: [BUG]S24_LE does not work on TGLU_UP_HDA_IPC4 and ADLP_RVP_NOCODEC_IPC4

XiaoyunWu6666, 2022-04-15:
**Describe the bug**
aplay with format S24_LE cause IO error 
ipc time out when GLB_SET_PIPELINE_STATE ,  failed to set state for pipeline

**To Reproduce**
aplay   -Dhw:0,3 -r 48000 -c 2 -f S24_LE -d 3 /dev/zero -v -q on TGLU_UP_HDA_IPC4
aplay   -Dhw:0,0 -r 48000 -c 2 -f S24_LE -d 3 /dev/zero -v -q on  ADLP_RVP_NOCODEC_IPC4

**Reproduction Rate**
100%

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: [4542828f16f1](https://github.com/thesofproject/linux/commit/4542828f16f1)
SOF Branch: main
SOF Commit: [87caf973ebac](https://github.com/thesofproject/sof/commit/87caf973ebac)
Zephyr Commit: [03231fba951b](https://github.com/zephyrproject-rtos/zephyr/commit/03231fba951b)
Topology:cavs-mixin-mixout-hda.tplg for TGLU_UP_HDA_IPC4 and cavs-tgl-nocodec.tplg for ADLP_RVP_NOCODEC_IPC4

**Screenshots or console output**
take TGLU_UP_HDA_IPC4 as example
[image]

[console]
```
aplay: pcm_write:2086: write error: Input/output error
```
[dmesg]
```
[  122.994094] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc4 set pipeline 6 state 3
[  122.994099] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13060003|0x0: GLB_SET_PIPELINE_STATE
[  122.994169] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx reply: 0x33000000|0x0: GLB_SET_PIPELINE_STATE
[  122.994181] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx done : 0x13060003|0x0: GLB_SET_PIPELINE_STATE
[  122.994186] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc4 set pipeline 6 state 4
[  122.994192] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13060004|0x0: GLB_SET_PIPELINE_STATE
[  123.497682] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc timed out for 0x13060004|0x0
[  123.497699] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: failed to set state 4 for pipeline 6
[  123.497705] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -110
[  123.497714] kernel:  HDMI1: ASoC: trigger FE cmd: 1 failed: -110
[  123.497726] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: trigger stream 3 dir 0 cmd 0
[  123.497731] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc4 set pipeline 5 state 3
[  123.497737] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13050003|0x0: GLB_SET_PIPELINE_STATE
[  124.001623] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc timed out for 0x13050003|0x0
[  124.001639] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: failed to pause pipeline 5
[  124.001655] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x160]=0x20140000 successful
[  124.001662] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -110
[  124.001671] kernel:  HDMI1: ASoC: trigger FE cmd: 0 failed: -110
[  124.505930] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: free stream 3 dir 0
[  124.505938] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc4 set pipeline 5 state 3
[  124.505944] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13050003|0x0: GLB_SET_PIPELINE_STATE
[  125.009502] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc timed out for 0x13050003|0x0
[  125.009516] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: failed to pause pipeline 5
[  125.009556] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: sof_ipc4_route_free: unbind modules copier.host.5.1 -> copier.HDA.6.1
[  125.009564] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x46000004|0x10004: MOD_UNBIND
[  125.513440] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc timed out for 0x46000004|0x10004
[  125.513450] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: failed to unbind modules copier.host.5.1 -> copier.HDA.6.1
[  125.513460] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x12050000|0x0: GLB_DELETE_PIPELINE
[  126.017383] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc timed out for 0x12050000|0x0
[  126.017393] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: failed to free pipeline widget pipeline.5
[  126.017404] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x12060000|0x0: GLB_DELETE_PIPELINE
[  126.521335] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc timed out for 0x12060000|0x0
[  126.521344] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: failed to free pipeline widget pipeline.6
[  126.521350] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Failed to free connected widgets
[  126.521362] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_free on 0000:00:1f.3: -110
[  126.521473] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: close stream 3 dir 0

```

XiaoyunWu6666, 2022-04-15:
this is not a new issue actually but was hided by  https://github.com/thesofproject/sof/issues/5661 which was newly addressed by sof PR https://github.com/thesofproject/sof/pull/5671

lgirdwood, 2022-04-20:
@XiaoyunWu6666 can we close ?

XiaoyunWu6666, 2022-04-21:
@lgirdwood I'm afraid that not yet ?
This error can be seen in today's daily result still , also the theoretically  potential fix hasn't been merged #5696

keqiaozhang, 2022-04-25:
@XiaoyunWu6666 , please help to check this issue with https://github.com/thesofproject/sof/pull/5696. Thanks!

XiaoyunWu6666, 2022-04-25:
> @XiaoyunWu6666 , please help to check this issue with #5696. Thanks!

Sure , I tried with #5696 , it can solve this issue (though [FW#5701](https://github.com/thesofproject/sof/issues/5701)
 [BUG]IPC4_BUSY when MOD_SET_DX on TGLU_UP_HDA_IPC4 will affect the testcase also

closed 2022-04-25 by lgirdwood