thesofproject/sof #5693: [FEATURE] CMake can pass all SOF_MAJOR, SOF_MINOR and SOF_MICRO to rimage

mengdonglin, 2022-04-17:
**Purpose**
Lower the risk of releasing FW binaries which are built and signed before SOF release tag is applied.
This feature can help us check FW version of a FW binary *.ri using https://github.com/thesofproject/sof/blob/main/tools/sof_ri_info/sof_ri_info.py
Thus we needn't check FW version of each *.ri by testing in Linux and checking version in dmesg of FW ready IPC on each supported platform.

**Describe the solution you'd like**
Please find the proposal and additional context in in https://github.com/thesofproject/rimage/issues/90

lgirdwood, 2022-04-20:
@marc-hb any comments ?

mengdonglin, 2022-04-30:
@aiChaoSONG Thanks, you will find my PR is almost same as your comments :)

closed 2022-05-09 by lgirdwood