thesofproject/sof #5701: [BUG]IPC4_BUSY when ctx_save during suspend on TGLU_UP_HDA_IPC4

XiaoyunWu6666, 2022-04-20:
**Describe the bug**
on IPC4 platform TGLU_UP_HDA_IPC4 which run sof zephyr IPC4 firmware
FW reported error: 4 - The system or resource is busy when MOD_SET_DX

This happened after https://github.com/thesofproject/linux/pull/3579 merged yesterday

**To Reproduce**
After loading the firmware, check the dmesg log or run below command to catch kernel log error directly
```
 TPLG=/lib/firmware/intel/avs-tplg/cavs-mixin-mixout-hda.tplg MODEL=TGLU_UP_HDA_IPC4 ~/sof-test/test-case/verify-kernel-boot-log.sh
```

**Reproduction Rate**
100%

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: [ba893537c5e0](https://github.com/thesofproject/linux/commit/ba893537c5e0)
SOF Branch: main
SOF Commit: [39a913d7daa8](https://github.com/thesofproject/sof/commit/39a913d7daa8)
Zephyr Commit: [c6b05afc9255](https://github.com/zephyrproject-rtos/zephyr/commit/c6b05afc9255)
Topology:cavs-mixin-mixout-hda.tplg
 Platform: TGLU_UP_HDA_IPC4

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

Please also include the relevant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). Kernel log is taken from *dmesg* and firmware log from *sof-logger*. See https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

ujfalusi, 2022-04-21:
@RanderWang, can this be because we are missing some cleanup in kernel or firmware side? It looks to me that the SET_DX fails fro primary core on suspend (ctx_save) in this sequence:
```
[  113.190397] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13040003|0x0: GLB_SET_PIPELINE_STATE
[  113.190506] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx reply: 0x33000007|0x0: GLB_SET_PIPELINE_STATE
[  113.190509] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW reported error: 7 - Unsupported operation requested
[  113.190531] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc error for msg 0x13040003|0x0
[  113.190535] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: failed to pause pipeline 4
[  113.190538] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -22
[  113.190542] kernel:  HDA Analog: ASoC: trigger FE cmd: 1 failed: -22
[  113.190548] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: trigger stream 0 dir 1 cmd 0
...
[  113.192655] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: widget pipeline.3 freed
[  113.192657] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: widget copier.host.3.1 freed
[  113.192751] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: close stream 0 dir 1
[  115.682855] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x47000000|0x0: MOD_SET_DX [data size: 8]
[  115.682965] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx reply: 0x67000004|0x0: MOD_SET_DX
[  115.682969] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW reported error: 4 - The system or resource is busy
[  115.683016] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc error for msg 0x47000000|0x0
[  115.683020] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ctx_save IPC error during suspend: -16
[  119.713853] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x47000000|0x0: MOD_SET_DX [data size: 8]
[  119.714032] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx reply: 0x67000004|0x0: MOD_SET_DX
[  119.714036] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW reported error: 4 - The system or resource is busy
[  119.715108] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc error for msg 0x47000000|0x0
[  119.715115] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ctx_save IPC error during suspend: -16
[  119.893332] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x47000000|0x0: MOD_SET_DX [data size: 8]
[  119.893457] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx reply: 0x67000004|0x0: MOD_SET_DX
[  119.893461] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW reported error: 4 - The system or resource is busy
[  119.893493] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc error for msg 0x47000000|0x0
[  119.893497] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ctx_save IPC error during suspend: -16
```
We have a fail with pipeline 4 during preparing for playback, this aborts the setup and we go to stop and do some cleanup. Then runtime suspend, but SET_DX fail due to either secondary core is still enabled or we still have a pipeline which is active. If we miss something in cleanup, then we might be leaving a pipeline enabled?

@ranj063 FYI.

ujfalusi, 2022-04-22:
After more investigation it turned out that we might have another explanation:
Zephyr does not support turning cores off and the SET_DX for core0 fails because other core than core0 is enabled:
```
secondary cores 0x9 still active
```
This is on tgl where core3 is needed for booting up the DSP.
With a PR from @kv2019i this error is worked around for the time being: https://github.com/thesofproject/sof/pull/5708

XiaoyunWu6666, 2022-04-27:
Thanks for the fix!

This wasn't observed in daily IPC4 test result (id:12080) after #5708  merged .
May we close it ? :)

lgirdwood, 2022-04-27:
@kv2019i can we close ?

kv2019i, 2022-04-27:
As @XiaoyunWu6666  and @ujfalusi both confirmed, then yes.

closed 2022-04-27 by kv2019i