thesofproject/sof #5732: [BUG] IPC tx error for TRIG_START on APL_UP2_PCM512X when testing multiple pause-resume

miRoox, 2022-04-26:
**Describe the bug**
CI (ID: 12065) detected an IPC tx error for TRIG_START when running multiple-pause-resume-50 on APL_UP2_PCM512X.

**To Reproduce**
```
TPLG=/lib/firmware/intel/sof-tplg/sof-apl-pcm512x.tplg ~/sof-test/test-case/multiple-pause-resume.sh -r 50
```

**Reproduction Rate**
It is only observed once in CI and cannot be reproduced manually.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: https://github.com/thesofproject/linux/commit/95ec8d32e6b6
    * SOF: https://github.com/thesofproject/sof/commit/68e8c9066f52
2) Name of the topology file
    * Topology: sof-apl-pcm512x.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: APL

**Screenshots or console output**

**dmesg**
```
[ 3590.828791] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: pcm: trigger stream 0 dir 0 cmd 1
[ 3590.829008] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: FW Poll Status: reg[0x180]=0x24001e successful
[ 3590.829019] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[ 3590.829297] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx error for 0x60040000 (msg/reply size: 12/12): -22
[ 3590.829559] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ASoC: error at soc_component_trigger on 0000:00:0e.0: -22
[ 3590.829598] kernel:  Port5: ASoC: trigger FE cmd: 1 failed: -22
[ 3590.829624] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: pcm: trigger stream 0 dir 0 cmd 0
[ 3590.829632] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[ 3590.852427] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: pcm: trigger stream 4 dir 0 cmd 4
[ 3590.852444] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: FW Poll Status: reg[0x160]=0x2014001e successful
[ 3591.271191] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc rx: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[ 3591.271234] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc rx done: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[ 3591.336177] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx timed out for 0x60050000 (msg/reply size: 12/12)
[ 3591.336285] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: preventing DSP entering D3 state to preserve context
[ 3591.336294] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ------------[ IPC dump start ]------------
[ 3591.336369] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: hda irq intsts 0x00000000 intlctl 0xc0000181 rirb 00
[ 3591.336447] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: dsp irq ppsts 0x00000000 adspis 0x00000000
[ 3591.336520] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[ 3591.336600] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ------------[ IPC dump end ]------------
[ 3591.336669] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ------------[ DSP dump start ]------------
[ 3591.336738] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: IPC timeout
[ 3591.336789] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: fw_state: SOF_FW_BOOT_COMPLETE (6)
[ 3591.336918] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: status: fw entered - code 00000005
[ 3591.337030] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: unexpected fault 0x00000000 trace 0x00004000
[ 3591.337104] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ------------[ DSP dump end ]------------
[ 3591.337224] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: FW Poll Status: reg[0x180]=0x240000 successful
[ 3591.337238] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ASoC: error at soc_component_trigger on 0000:00:0e.0: -110
[ 3591.337315] kernel:  Port5: ASoC: trigger FE cmd: 0 failed: -110
[ 3591.338462] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: trace IO error
```

miRoox, 2022-04-29:
Happened again in today's daily test (ID: 12177).

mengdonglin, 2022-09-06:
Close as this issue cannot be reproduced for a long time.