thesofproject/sof #5733: [BUG] [ADLP] KMOD test failed on ADLP_GMB_I2S_ZEPHYR

keqiaozhang, 2022-04-26:


marc-hb, 2022-04-28:
More severe kmod crashes just spotted in
- #5747

keqiaozhang, 2022-05-16:
@lyakh any updates? It always fails in CI.

lyakh, 2022-05-16:
> @lyakh any updates? It always fails in CI.

@keqiaozhang I'm still debugging this. I think I have to discuss with Zephyr developers more actively

lgirdwood, 2022-05-25:
@lyakh ping any updates ?

lyakh, 2022-05-27:
> @lyakh ping any updates ?

@lgirdwood a few updates, yes. Mostly I now have two lines of working with this:
1. #5852 eliminates a "scheduling in atomic context" bug, that has been identified in the DMIC driver as a part of this work, but which isn't really the cause of these failues
2. DSP core power down with Zephyr on TGL. This isn't supported so far, and when this problem occurs I see core 1 booting after kernel module reloading and getting confused because a suitable context hasn't been prepared for it. From that I *assumed* that the reason for core 1 booting automagically is that it was used before module unloading and wasn't powered down correctly. I'm working on patches to verify and fix this, but they aren't quite working yet.

lyakh, 2022-06-09:
I was able to reproduce the sporadic core boot problem with a modified native Zephyr test, that bug is now tracked in https://github.com/zephyrproject-rtos/zephyr/issues/46372

lyakh, 2022-06-21:
An update: a brute-force fix is available: https://github.com/zephyrproject-rtos/zephyr/issues/46372#issuecomment-1161393193 but a clean fix still has to be developed

keqiaozhang, 2022-07-04:
Conformed that this issue is fixed byhttps://github.com/zephyrproject-rtos/zephyr/pull/46587 and it's not reproducible on GMB, closing this bug.

closed 2022-07-04 by keqiaozhang