thesofproject/sof #5747: [BUG] check-kmod-load-unload.sh  causes hard kernel lockups

marc-hb, 2022-04-28:
**Describe the bug**

`~/sof-test/test-case/check-kmod-load-unload.sh -l 25` causes hard kernel lockups

This is a very recent regression in daily tests.

**To Reproduce**

```
TPLG=/lib/firmware/intel/sof-tplg/sof-tgl-nocodec-ci.tplg MODEL=ADLP_RVP_NOCODEC_CI ~/sof-test/test-case/check-kmod-load-unload.sh -l 25
```

or

```
TPLG=/lib/firmware/intel/sof-tplg/sof-tgl-rt715-rt711-rt1308-mono.tplg MODEL=TGLU_SKU0A3E_SDW ~/sof-test/test-case/check-kmod-load-unload-after-playback.sh -l 5
```

**Reproduction Rate**

Too early to tell.

**Environment**

First bad daily test: 12140

Start Time: 2022-04-27 22:29:55 UTC
End Time: 2022-04-28 02:49:19 UTC
< >>
Kernel Branch: topic/sof-dev
Kernel Commit: 95ec8d32e6b6
SOF Branch: main
SOF Commit: 94e6c9969730

Last good daily test: 12106

Start Time: 2022-04-26 22:29:40 UTC
End Time: 2022-04-27 05:52:13 UTC
< > >>
Kernel Branch: topic/sof-dev
Kernel Commit: 95ec8d32e6b6
SOF Branch: main
SOF Commit: 3334bede2a79
 

Only these commits are different between last good and first bad daily tests:

94e6c9969730 eddyhsu@chromium.org // (origin/main, origin/HEAD) kpb: Clean up params on component reset
99eac8de991d rander.wang@intel.com // ipc4: add format conversion support of s32le to s24le
f4f9e6541ee7 chao.song@linux.intel.com // sof: replace log calls with zephyr logging api
d67e2ea8a4bc eddyhsu@chromium.org // dcblock: Clean up processing function on reset.
426542d17f18 daniel.baluta@nxp.com // codec_adapter: cadence: Resolve api_id at runtime
02dd7a6dfb74 daniel.baluta@nxp.com // codec_adapter: cadence: Remove api_id parameter
05969a0e34da marc.herbert@intel.com // .github: IPC4: build supported TGL platforms first
dccf9c0798d5 marc.herbert@intel.com // .github/zephyr: download docker image from ghcr.io instead of docker.io
75dea11da97d chunxu.li@mediatek.com // topology1: add mt8186 related pipelines and components

Last good commit   3334bede2a79 seppo.ingalsuo@linux.intel.com // Tools: Testbench: Add components to testbench load

marc-hb, 2022-04-28:
ADLP_RVP_NOCODEC_CI sh-adlp-rvp-nocodec-02 

```
2022-04-28 00:17:08 UTC [REMOTE_INFO] ==== firmware boot complete: 16 of 25 ====
2022-04-28 00:17:11 UTC [REMOTE_INFO] ===== Starting iteration 17 of 25 =====
2022-04-28 00:17:11 UTC [REMOTE_INFO] wait dsp power status to become suspended

   sudden death, no clean shutdown, no logs
```

TGLU_SKU0A3E_SDW  jf-tglu-sku0a3e-sdw-01 
```
2022-04-28 00:08:53 UTC Sub-Test: [REMOTE_INFO] ===== Testing: (Round: 1/1) (PCM: HDMI 3 [hw:0,7]) (Loop: 1/1) =====
2022-04-28 00:08:53 UTC Sub-Test: [REMOTE_COMMAND] aplay   -Dhw:0,7 -r 48000 -c 2 -f S16_LE -d 3 /dev/zero -v -q

 sudden death, no clean shutdown, no logs
```

aiChaoSONG, 2022-04-28:
It doesn't make sense to me that the SOF code change causes kernel issue, and in the meanwhile, there is no error in sof logger.

plbossart, 2022-04-28:
Agree with @aiChaoSONG, we should try to see if we can identify regressions due to recent upstream merges. 

95ec8d32e6b6355514bbe267f5bef4abb6c8b4b3 (04/21)
8f77d721dd559bcf67368c606dd5d3f5f776c929 (04/18)

marc-hb, 2022-04-28:
A short reminder that we have been regularly having lockups in suspend/resume tests since forever; we just got used to them and many are likely NOT audio related (but we can't tell of course).

While they could be more than one day old, hard lockups in kmod tests seem very new.

plbossart, 2022-04-29:
@marc-hb It's hard to conclude for TGLU_SKU0A3E_SDW, there's only one failed test but it comes without any dmesg info.

````
12177 pass
12140 fail
12106 pass
12065 pass
````

marc-hb, 2022-04-29:
Yes the reproduction rate has been very low so it's very hard to draw any conclusion. Apologies for sharing with this SOF git log too early.

I spent time gathering data and filing this bug because:
- I don't remember us ever experiencing any hard crash on kmod testing. Suspend/resume crashes: yes, kmod: no.
- We had 3 systems experiencing hard lockups in the same kmod test in the same daily run. That was too much coincidence. The 3rd system was sh-tglh-0a70-hda-01 . I did not share its logs because it experienced worse NTP issues which made the logs very hard to make sense of - but there was a hard lockup on sh-tglh-0a70-hda-01 too.

> it comes without any dmesg info.

Automation going back and collecting the right journalctl entry after a crash and reboot is theoretically possible (because anything is theoretically possible) but we're very far from having that capability now. To file this bug I logged into the systems and gathered info **manually** which... told me very little besides the fact that the systems actually crashed. So post-crash collection of journalctl may not be worth a big development effort. Something like  this would be better instead but it's unlikely to come "for free" either: https://chromium.googlesource.com/chromiumos/platform2/+/HEAD/crash-reporter/docs/collectors.md#kernel_collector

plbossart, 2022-04-29:
I will agree with you @marc-hb that in general kmod load/unload issues are directly related to audio issues, since we don't touch the other drivers. That makes the occurrences of such bugs less likely than the generic case where all possible drivers could impact suspend-resume.

I also agree that there is no such thing as coincidence, we should absolutely keep track of this issue. I wasn't trying to close it or dismiss your opinion, I commented to acknowledge that I am aware of the issue but could not suggest any useful explanation or lead to look further into the problem.

ujfalusi, 2022-05-03:
It is hard to reproduce for sure, last week I took the tglh machine and I saw once a lockup:
```
[19659.122778] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x40010000: GLB_PM_MSG: CTX_SAVE
[19659.123126] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x40010000: GLB_PM_MSG: CTX_SAVE
[19659.124310] sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x4]=0x3010303 successful
[19659.124315] sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x4]=0x303 successful
[19659.124317] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[19661.564260] INFO: task journal-offline:163719 blocked for more than 122 seconds.
[19661.564277]       Not tainted 5.18.0-rc2-daily-default-20220427-0-g95ec8d32e6b6 #dev
[19661.564283] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[19661.564288] task:journal-offline state:D stack:13760 pid:163719 ppid:     1 flags:0x00004000
[19661.564300] Call Trace:
[19661.564304]  <TASK>
[19661.564315]  __schedule+0x2b4/0x8c0
[19661.564326]  ? sched_clock_cpu+0xd/0xd0
[19661.564343]  schedule+0x4e/0xb0
[19661.564349]  ? folio_wait_bit+0xdd/0x1d0
[19661.564356]  io_schedule+0x12/0x40
[19661.564362]  folio_wait_bit+0xdd/0x1d0
[19661.564372]  ? filemap_invalidate_unlock_two+0x40/0x40
[19661.564386]  folio_wait_writeback+0x56/0xb0
[19661.564392]  __filemap_fdatawait_range+0x87/0x120
[19661.564420]  file_write_and_wait_range+0xc6/0xe0
[19661.564436]  ext4_sync_file+0x1a5/0x3e0
[19661.564449]  do_fsync+0x38/0x70
[19661.564460]  __x64_sys_fsync+0x10/0x20
[19661.564465]  do_syscall_64+0x38/0x90
[19661.564474]  entry_SYSCALL_64_after_hwframe+0x44/0xae
[19661.564482] RIP: 0033:0x7ff09e5491fb
[19661.564489] RSP: 002b:00007ff090a10c70 EFLAGS: 00000293 ORIG_RAX: 000000000000004a
[19661.564497] RAX: ffffffffffffffda RBX: 0000557f19479f60 RCX: 00007ff09e5491fb
[19661.564501] RDX: 00007ff09e3989de RSI: 00007ff09e3989de RDI: 000000000000002b
[19661.564505] RBP: 00007ff09e39a1a0 R08: 0000000000000000 R09: 00007ff090a11700
[19661.564508] R10: 000000000000001e R11: 0000000000000293 R12: 0000000000000002
[19661.564512] R13: 00007ffc6eacda8f R14: 00007ffc6eacda90 R15: 00007ff090a10d80
[19661.564536]  </TASK>
[19661.564580] INFO: task kworker/u32:1:63196 blocked for more than 122 seconds.
[19661.564588]       Not tainted 5.18.0-rc2-daily-default-20220427-0-g95ec8d32e6b6 #dev
[19661.564593] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[19661.564598] task:kworker/u32:1   state:D stack:12824 pid:63196 ppid:     2 flags:0x00004000
[19661.564610] Workqueue: writeback wb_workfn (flush-259:0)
[19661.564622] Call Trace:
[19661.564626]  <TASK>
[19661.564637]  __schedule+0x2b4/0x8c0
[19661.564645]  ? __blk_flush_plug+0xdf/0x120
[19661.564660]  schedule+0x4e/0xb0
[19661.564667]  io_schedule+0x12/0x40
[19661.564673]  blk_mq_get_tag+0x11c/0x280
[19661.564681]  ? sched_clock_cpu+0xd/0xd0
[19661.564688]  ? complete_all+0x60/0x60
[19661.564702]  __blk_mq_alloc_requests+0x188/0x2e0
[19661.564709]  ? blk_attempt_plug_merge+0x76/0x90
[19661.564721]  blk_mq_submit_bio+0x421/0x700
[19661.564744]  submit_bio_noacct_nocheck+0x259/0x2b0
[19661.564759]  ext4_io_submit+0x20/0x40
[19661.564766]  ext4_bio_write_page+0x16e/0x520
[19661.564778]  mpage_submit_page+0x4e/0x70
[19661.564786]  mpage_process_page_bufs+0xf8/0x110
[19661.564794]  mpage_prepare_extent_to_map+0x1f9/0x450
[19661.564826]  ext4_writepages+0x311/0xeb0
[19661.564843]  ? __lock_acquire.isra.0+0x125/0x550
[19661.564873]  do_writepages+0xbf/0x180
[19661.564879]  ? find_held_lock+0x32/0x90
[19661.564886]  ? sched_clock_cpu+0xd/0xd0
[19661.564897]  __writeback_single_inode+0x6f/0x470
[19661.564910]  writeback_sb_inodes+0x214/0x4d0
[19661.564936]  __writeback_inodes_wb+0x4c/0xe0
[19661.564948]  wb_writeback+0x2df/0x360
[19661.564965]  wb_workfn+0x304/0x500
[19661.564991]  process_one_work+0x29c/0x530
[... 64 lines cut ...]
[19661.930819]  </TASK>
[19661.930862] 
               Showing all locks held in the system:
[19661.930868] 3 locks held by kworker/0:1/16:
[19661.930877] 1 lock held by khungtaskd/96:
[19661.930880]  #0: ffffffff9f765cc0 (rcu_read_lock){....}-{1:2}, at: debug_show_all_locks+0x15/0xf7
[19661.930906] 1 lock held by journal-offline/33295:
[19661.930909]  #0: ffff9e0ac19c4c80 (&sbi->s_writepages_rwsem){....}-{0:0}, at: do_writepages+0xbf/0x180
[19661.930927] 1 lock held by in:imklog/577:
[19661.930930]  #0: ffff9e0aced8c2f8 (&f->f_pos_lock){....}-{3:3}, at: __fdget_pos+0x4a/0x60
[19661.930959] 1 lock held by dmesg/34165:
[19661.930963]  #0: ffff9e0ad5d400e0 (&user->lock){....}-{3:3}, at: devkmsg_read+0x4b/0x230
```
This looks awfully similar and if I recall I got the first lockup also not long after I first logged in to the tglh machine (`sh-tglh-0a70-hda-01`).
With FS error the logs are not going to be written out for sure, so chances to see this one post mortal is unlikely.

I don't know, but during boot we have this in the kernel log:
```
[    0.804691] software IO TLB: mapped [mem 0x0000000057cd6000-0x000000005bcd6000] (64MB)
[    0.804749] DMAR: DRHD: handling fault status reg 3
[    0.804758] DMAR: [DMA Read NO_PASID] Request device [00:02.0] fault addr 0x70e68000 [fault reason 0x07] Next page table ptr is invalid
[    0.804863] resource sanity check: requesting [mem 0xfedc0000-0xfedcdfff], which spans more than pnp 00:05 [mem 0xfedc0000-0xfedc7fff]
[    0.804865] caller __uncore_imc_init_box+0xc5/0x100 mapping multiple BARs
[    0.805264] clocksource: tsc: mask: 0xffffffffffffffff max_cycles: 0x2cd5a8bb032, max_idle_ns: 440795365915 ns
```

ujfalusi, 2022-05-03:
Run about 1700 iteration of `check-kmod-load-unload.sh` after rebooting `sh-tglh-0a70-hda-01` and not a single error.

ujfalusi, 2022-05-10:
Managed to reproduce it again after reserving and logging in to the machine:
```

[ 2704.414619] INFO: task jbd2/nvme0n1p2-:289 blocked for more than 122 seconds.
[ 2704.414633]       Not tainted 5.18.0-rc2-daily-default-20220509-1-g4c78c8bef64e #ci
[ 2704.414638] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[ 2704.414641] task:jbd2/nvme0n1p2- state:D stack:13824 pid:  289 ppid:     2 flags:0x00004000
[ 2704.414651] Call Trace:
[ 2704.414653]  <TASK>
[ 2704.414662]  __schedule+0x2b4/0x8c0
[ 2704.414673]  ? lock_release+0x148/0x3f0
[ 2704.414684]  schedule+0x4e/0xb0
[ 2704.414689]  jbd2_journal_wait_updates+0x79/0xe0
[ 2704.414696]  ? complete_all+0x60/0x60
[ 2704.414704]  jbd2_journal_commit_transaction+0x2b4/0x1cb0
[ 2704.414708]  ? sched_clock_cpu+0xd/0xd0
[ 2704.414713]  ? __lock_acquire.isra.0+0x125/0x550
[ 2704.414719]  ? find_held_lock+0x32/0x90
[ 2704.414726]  ? lock_timer_base+0x61/0x80
[ 2704.414730]  ? lock_acquired+0xe2/0x3b0
[ 2704.414739]  ? lock_release+0x148/0x3f0
[ 2704.414748]  ? _raw_spin_unlock_irqrestore+0x33/0x50
[ 2704.414755]  ? try_to_del_timer_sync+0x4f/0x80
[ 2704.414763]  kjournald2+0xbf/0x2c0
[ 2704.414771]  ? complete_all+0x60/0x60
[ 2704.414778]  ? commit_timeout+0x10/0x10
[ 2704.414783]  kthread+0xf0/0x120
[ 2704.414788]  ? kthread_complete_and_exit+0x20/0x20
[ 2704.414795]  ret_from_fork+0x1f/0x30
[ 2704.414814]  </TASK>
[ 2704.414818] INFO: task systemd-journal:336 blocked for more than 122 seconds.
[ 2704.414822]       Not tainted 5.18.0-rc2-daily-default-20220509-1-g4c78c8bef64e #ci
[ 2704.414826] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[ 2704.414828] task:systemd-journal state:D stack:12864 pid:  336 ppid:     1 flags:0x00000000
[ 2704.414836] Call Trace:
[ 2704.414838]  <TASK>
[ 2704.414843]  __schedule+0x2b4/0x8c0
[ 2704.414848]  ? sched_clock_cpu+0xd/0xd0
[ 2704.414856]  schedule+0x4e/0xb0
[ 2704.414860]  ? __folio_lock+0x11e/0x1f0
[ 2704.414866]  io_schedule+0x12/0x40
[ 2704.414871]  __folio_lock+0x11e/0x1f0
[ 2704.414878]  ? filemap_invalidate_unlock_two+0x40/0x40
[ 2704.414886]  ? ext4_da_release_space+0x150/0x150
[ 2704.414892]  block_page_mkwrite+0x14a/0x170
[ 2704.414901]  ext4_page_mkwrite+0x54d/0x700
[ 2704.414907]  ? sched_clock_cpu+0xd/0xd0
[ 2704.414916]  do_page_mkwrite+0x52/0xf0
[ 2704.414923]  do_wp_page+0x3de/0x6b0
[ 2704.414929]  __handle_mm_fault+0x6a5/0xcd0
[ 2704.414933]  ? sched_clock_cpu+0xd/0xd0
[ 2704.414935]  ? sched_clock_cpu+0xd/0xd0
[ 2704.414950]  handle_mm_fault+0x7a/0x1c0
[ 2704.414957]  do_user_addr_fault+0x1d7/0x660
[ 2704.414968]  exc_page_fault+0x64/0x1b0
[ 2704.414972]  ? asm_exc_page_fault+0x8/0x30
[ 2704.414976]  asm_exc_page_fault+0x1e/0x30
[ 2704.414980] RIP: 0033:0x7f00faa88160
[ 2704.414985] RSP: 002b:00007ffd7970b830 EFLAGS: 00010297
[ 2704.414990] RAX: 00000000000000ea RBX: 0000000000000040 RCX: 0000000000000006
[ 2704.414992] RDX: 0000000000000006 RSI: 00007f00fabe01bc RDI: 00007f00f0f46070
[ 2704.414995] RBP: 000055df073a8550 R08: 0000000000000768 R09: 000055df073a90b0
[ 2704.414997] R10: 00007f00f52a7b08 R11: 00007f00ef3d3fc0 R12: 0000000002440070
[ 2704.414999] R13: 00007ffd7970b858 R14: 0000000003893f80 R15: 00007f00f0f46070
[ 2704.415015]  </TASK>
[ 2704.415032] INFO: task BESClient:614 blocked for more than 122 seconds.
[ 2704.415037]       Not tainted 5.18.0-rc2-daily-default-20220509-1-g4c78c8bef64e #ci
[ 2704.415041] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[ 2704.415044] task:BESClient       state:D stack:12872 pid:  614 ppid:     1 flags:0x00000000
[ 2704.415051] Call Trace:
[... 149 lines cut ...]
RMMOD   soundwire_cadence
RMMOD   soundwire_generic_allocation
RMMOD   regmap_sdw
RMMOD   regmap_sdw_mbq
RMMOD   soundwire_bus
RMMOD   snd_intel_sdw_acpi
RMMOD   snd_soc_core
RMMOD   snd_hda_codec
RMMOD   snd_hda_core
RMMOD   snd_hwdep
RMMOD   snd_compress
SKIP    snd_pcm_dmaengine       not loaded
RMMOD   snd_pcm
2022-05-10 08:58:46 UTC [INFO] checking for general errors after kmod unload with sof-kernel-log-check tool
declare -- cmd="journalctl_cmd --since=@1652173121"
```
Plenty of memory was available before starting the test run:
```
ubuntu@sh-tglh-0a70-hda-01:~$ uptime 
 08:18:28 up 2 min,  3 users,  load average: 0.22, 0.12, 0.05
ubuntu@sh-tglh-0a70-hda-01:~$ free
              total        used        free      shared  buff/cache   available
Mem:       16108436      704812    14393420      151024     1010204    15068128
Swap:       2097148           0     2097148
```

marc-hb, 2022-05-10:
There's a hung journald also in
- https://github.com/thesofproject/linux/issues/3387

lgirdwood, 2022-05-11:
Could this be aplay/soundserver hanging on to an audio file ? I dont think its an audio device file (since ext4 is mentioned in the logs).

plbossart, 2022-05-11:
I saw a similar issue today on TGLU_SKU0A32_SDCA while running basic sof-test stuff manually, where we don't use files really - except for left/right tests.

ujfalusi, 2022-05-12:
It is more like journald causes filesystem lockup (or vice versa)
module unload load generates fair amount of logs due to tplg build up on some platforms, but the system must be able to handle this. It is nowhere near to an fs stress test

marc-hb, 2022-05-12:
> but the system must be able to handle this. It is nowhere near to an fs stress test

If we can avoid a filesystem and/or journald bug by reducing log spam then we just should. We have neither time nor resources nor expertise to debug stuff like this.

It's not like log spam is useful anyway, in fact it's unreadable and cumbersome and slowing us down even when it does not crash.

lgirdwood, 2022-05-13:
Can we try a ```sync``` between each ```rmmod``` and see if this has any impact ? This would flush FS cache.