thesofproject/sof #5751: [BUG]

lenghonglin, 2022-04-29:
**Describe the bug**
[image]

when enable `CONFIG_ASSERT` . It show `ASSERTION FAIL` .

**To Reproduce**
when triger start, maybe it will cause this assert.

**Reproduction Rate**
maybe 1/10

**Impact**
It will cause zephyr fatal

**Environment**
    * Zephyr: v2.7.0-rc2-335-g817dbe8afbb4
    * SOF: v1.8-rc1-592-ga7cbfc324cc8-dirty
    * Platform: CSK6001

plbossart, 2022-04-29:
asserts are evil. I don't know how many times I'll have to ask that they be removed with proper error handling used instead. @lgirdwood FYI

kv2019i, 2022-05-03:
@lenghonglin Can you retest with newer SOF and Zephyr? You are using very old versions of both (SOF1.8 versus latest SOF2.1 and Zephyr2.7 versus Zephyr3.0 released).

I don't see how this can happen with current code. The domain thread is specifically run in a Zephyr thread, not ISR:
```
       thread = k_thread_create(&dt->ll_thread,
»       »       »       »        ll_sched_stack[core],
»       »       »       »        ZEPHYR_LL_STACK_SIZE,
»       »       »       »        zephyr_domain_thread_fn, zephyr_domain, NULL, NULL,
```

lenghonglin, 2022-05-04:
> @lenghonglin Can you retest with newer SOF and Zephyr? You are using very old versions of both (SOF1.8 versus latest SOF2.1 and Zephyr2.7 versus Zephyr3.0 released).
>
> I don't see how this can happen [...]

https://github.com/zephyrproject-rtos/zephyr/pull/42806  this pr slove my problem

kv2019i, 2022-05-04:
Solved in latest release, so closing the issue.

closed 2022-05-04 by kv2019i