thesofproject/sof #5777: [BUG][IPC4 zph]set state for pipeline failure on ADLP_RVP_NOCODEC

XiaoyunWu6666, 2022-05-05:
**Describe the bug**
in inner daily test result 12303 
when arecord with -Dhw:0,0
Pcm read connection timed out  on model ADLP_RVP_NOCODEC when check-capture-all-formats with set pipeline state failure 

**To Reproduce**
 TPLG=/lib/firmware/intel/avs-tplg/cavs-tgl-nocodec.tplg MODEL=ADLP_RVP_NOCODEC_IPC4 ~/sof-test/test-case/check-capture.sh -d 3 -l 1 -r 1 -F
OR 
arecord   -Dhw:0,0 -r 48000 -c 2 -f S16_LE -d 3 /dev/null -v -q

**Reproduction Rate**
100%

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: [a80343f4274d](https://github.com/thesofproject/linux/commit/a80343f4274d)
SOF Branch: main
SOF Commit: [b4591ca8f464](https://github.com/thesofproject/sof/commit/b4591ca8f464)
Zephyr Commit: [aceefa90ec68](https://github.com/zephyrproject-rtos/zephyr/commit/aceefa90ec68)
 Topology: cavs-tgl-nocodec.tplg
[image]

 Platform: ADLP_RVP_NOCODEC 

**Screenshots or console output**
[dmesg]
```
[  153.140856] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc4 set pipeline 7 state 3
[  153.140860] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13070003|0x0: GLB_SET_PIPELINE_STATE
[  153.140965] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx reply: 0x33000000|0x0: GLB_SET_PIPELINE_STATE
[  153.140973] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx done : 0x13070003|0x0: GLB_SET_PIPELINE_STATE
[  153.140976] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc4 set pipeline 7 state 4
[  153.140987] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13070004|0x0: GLB_SET_PIPELINE_STATE
[  153.647147] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc timed out for 0x13070004|0x0
[  153.647167] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: failed to set state 4 for pipeline 7
[  153.647176] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -110
[  153.647186] kernel:  Port0: ASoC: trigger FE cmd: 1 failed: -110
[  153.647199] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: trigger stream 0 dir 1 cmd 0
[  153.647219] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x80]=0x20140000 successful
[  153.647239] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc4 set pipeline 7 state 2
[  153.647246] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13070002|0x0: GLB_SET_PIPELINE_STATE
[  154.151206] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc timed out for 0x13070002|0x0
[  154.151227] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: failed to set state 2 for pipeline 7
[  154.151236] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -110
[  154.151245] kernel:  Port0: ASoC: trigger FE cmd: 0 failed: -110
[  154.151981] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: free stream 0 dir 1
[  154.151990] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc4 set pipeline 7 state 2
[  154.151998] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13070002|0x0: GLB_SET_PIPELINE_STATE
[  154.655270] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc timed out for 0x13070002|0x0
[  154.655291] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: failed to set state 2 for pipeline 7
[  154.655337] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: sof_ipc4_route_free: unbind modules copier.SSP.8.1 -> copier.host.7.1
[  154.655347] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x46000004|0x10004: MOD_UNBIND
[  155.159319] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc timed out for 0x46000004|0x10004
[  155.159337] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: failed to unbind modules copier.SSP.8.1 -> copier.host.7.1
[  155.159348] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x12080000|0x0: GLB_DELETE_PIPELINE
[  155.663369] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc timed out for 0x12080000|0x0
[  155.663387] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: failed to free pipeline widget pipeline.8
[  155.663399] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x12070000|0x0: GLB_DELETE_PIPELINE
[  156.167433] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc timed out for 0x12070000|0x0
[  156.167454] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: failed to free pipeline widget pipeline.7
[  156.167464] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Failed to free connected widgets
[  156.167482] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_free on 0000:00:1f.3: -110
[  156.167631] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: close stream 0 dir 1

```

XiaoyunWu6666, 2022-05-05:
set state for pipeline failure also happen in inner daily result 12303 on model TGLU_UP_HDA when multiple-pipeline-playback-50  with sof xtos IPC4 fw

XiaoyunWu6666, 2022-05-06:
set state for pipeline failure also happend in inner daily result 12339  , with unknown fw error
on model TGLU_UP_HDA_IPC4ZPH when check-playback-100times at the 95/100 rounds
[dmesg]
```
[  889.391196] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc4 set pipeline 7 state 3
[  889.391199] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13070003|0x0: GLB_SET_PIPELINE_STATE
[  889.391270] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx reply: 0x33000000|0x0: GLB_SET_PIPELINE_STATE
[  889.391280] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx done : 0x13070003|0x0: GLB_SET_PIPELINE_STATE
[  889.391284] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc4 set pipeline 7 state 4
[  889.391287] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13070004|0x0: GLB_SET_PIPELINE_STATE
[  889.399683] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx reply: 0x33000006|0x0: GLB_SET_PIPELINE_STATE
[  889.399685] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW reported error: 6 - Unknown error while processing the request
[  889.399695] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc error for msg 0x13070004|0x0
[  889.399699] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: failed to set state 4 for pipeline 7
[  889.399701] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -22
[  889.399704] kernel:  HDMI2: ASoC: trigger FE cmd: 1 failed: -22
[  889.399708] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: trigger stream 4 dir 0 cmd 0
[  889.400195] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x160]=0x20140000 successful
[  889.400201] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc4 set pipeline 7 state 2
[  889.400204] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13070002|0x0: GLB_SET_PIPELINE_STATE
[  889.400551] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx reply: 0x33000000|0x0: GLB_SET_PIPELINE_STATE
[  889.400558] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx done : 0x13070002|0x0: GLB_SET_PIPELINE_STATE
[  889.907646] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: free stream 4 dir 0
```

kv2019i, 2022-05-12:
I debugged this on Zephyr build and got a OS panic:
[00000003] <err> os:  ** CPU 0 EXCCAUSE 13 (load/store PIF data error)
[...}
Backtrace:0xBE02198D:0xBE076400 0xBE021850:0xBE076440 0xBE021A1B:0xBE076480 0xBE021A96:0xBE0764C0 0xBE024E15:0xBE076530 0xBE02515F:0xBE0765E0 0xBE025402:0xBE076630 0xBE0255AD:0xBE076670 0xBE025C9C:0xBE0766A0 0xBE016A7E:0xBE0766D0 0xBE01D5E4:0xBE0766F0 0xBE024BB8:0xBE076710 0xBE03AB83:0xBE076730 0xBE0131CC:0xBE076770

That backtrace points to comp_get_endpoint_type() . I tried @RanderWang 's https://github.com/thesofproject/sof/pull/5790 and that solves the issue also in Zephyr build.

closed 2022-05-12 by lgirdwood