thesofproject/sof #5789: [BUG] xrun after TRIG_PAUSE on ADLP_GMB_I2S_ZEPHYR when testing multiple pause resume

miRoox, 2022-05-09:
**Describe the bug**
CI detected the xrun after receiving TRIG_PAUSE on ADLP_GMB_I2S_ZEPHYR when testing multiple pause resume.

**To Reproduce**
```
TPLG=/lib/firmware/intel/sof-tplg/sof-adl-max98390-rt5682.tplg MODEL=ADLP_GMB_I2S_ZEPHYR ~/sof-test/test-case/multiple-pause-resume.sh -r 50
```

**Reproduction Rate**
Not sure, need more investigation.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: topic/sof-dev|https://github.com/thesofproject/linux/commit/68b76766e8ba
    * SOF: main|https://github.com/thesofproject/sof/commit/b36db45a7da7
2) Name of the topology file
    * Topology: sof-adl-max98390-rt5682.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: ADLP_GMB_I2S_ZEPHYR

**Screenshots or console output**

**slogger**:
```
[    30674272.218614] (       61003.851562) c0 ipc                  src/ipc/ipc3/handler.c:1604 INFO ipc: new cmd 0x60060000
[    30674368.208193] (          95.989578) c1 pipe         10.51 ....../pipeline-stream.c:267  INFO pipe trigger cmd 2
[    30675165.656078] (         797.447876) c1 dw-dma                 src/drivers/dw/dma.c:376  INFO dw_dma_pause(): dma 1 channel 0 pause
[    30675188.624828] (          22.968750) c1 dmic-dai     2.0   ....../intel/dmic/dmic.c:418  INFO dmic_stop(), dmic_active_fifos_mask = 0x1
[    30675225.864409] (          37.239582) c1 zll-schedule       src/schedule/zephyr_ll.c:69   INFO task complete 0xbe092a00 pipe-task
[    30603190.815188] (        2373.697754) c0 hda-dma            ..../intel/hda/hda-dma.c:948  ERROR hda_dma_link_check_xrun(): underrun detected
[    30611276.439867] (        8085.624512) c0 hda-dma            ..../intel/hda/hda-dma.c:948  ERROR hda_dma_link_check_xrun(): underrun detected
[    30613268.366871] (        1991.927002) c0 hda-dma            ..../intel/hda/hda-dma.c:948  ERROR hda_dma_link_check_xrun(): underrun detected
[    30675245.916492] (          20.052082) c1 zll-schedule       src/schedule/zephyr_ll.c:71   INFO num_tasks 2 total_num_tasks 4
[    30675265.968574] (          20.052082) c1 zll-schedule       src/schedule/zephyr_ll.c:69   INFO task complete 0x9e086808 dmic-work <59c87728-d8f9-42f6-b89d-5870a87b0e1e>
[    30675282.947740] (          16.979166) c1 zll-schedule       src/schedule/zephyr_ll.c:71   INFO num_tasks 1 total_num_tasks 3
[    30675302.999823] (          20.052082) c1 zll-schedule       ......../zephyr_domain.c:210  INFO zephyr_domain_unregister domain->type 1 domain->clk 4
```

**dmesg**:
```
[ 3639.913213] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: trigger stream 99 dir 1 cmd 3
[ 3639.913228] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
[ 3640.416796] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx timed out for 0x60060000 (msg/reply size: 12/12)
[ 3640.416835] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: preventing DSP entering D3 state to preserve context
[ 3640.416845] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump start ]------------
[ 3640.416872] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: hda irq intsts 0x00000000 intlctl 0xc0000083 rirb 00
[ 3640.416886] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: dsp irq ppsts 0x00000000 adspis 0x00000000
[ 3640.416906] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[ 3640.416918] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump end ]------------
[ 3640.416928] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 3640.416937] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: IPC timeout
[ 3640.416948] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (6)
[ 3640.416969] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: status: fw entered - code 00000005
[ 3640.417110] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: unexpected fault 0x00000000 trace 0x00004000
[ 3640.417124] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
[ 3640.417164] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0xa0]=0x20240000 successful
[ 3640.417185] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -110
[ 3640.417202] kernel:  DMIC: ASoC: trigger FE cmd: 3 failed: -110
```

[full_slogger.txt](https://github.com/thesofproject/sof/files/8650061/full_slogger.txt)
[full_dmesg.txt](https://github.com/thesofproject/sof/files/8650062/full_dmesg.txt)

kv2019i, 2022-05-09:
Hmm, this is interesting.  The last statement of slogger log is PCM_PARAMS:
[     3206997.372565] (          20.052082) c1 ipc                      src/ipc/ipc3/dai.c:185  INFO ipc_comp_dai_config() dai type = 3 index = 3
[     3207454.924631] (         457.552063) c0 ipc                  src/ipc/ipc3/handler.c:1604 INFO ipc: new cmd 0x60010000
...
[     3207823.414199] (          34.687500) c0 pipe         8.43  ....../pipeline-params.c:340  INFO pipe prepare
[     3207868.205864] (          44.791664) c0 dai          8.42           src/audio/dai.c:622  INFO dai_prepare()
error: in logger_read(), fread(..., /sys/kernel/debug/sof/trace) failed: Operation not permitted(-1)

In the kernel log, there is still a lot of activity after the last PCM_PARAMS which occurs at:
[ 3639.521920] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: trigger stream 5 dir 0 cmd 1
[ 3639.522264] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS

So in other words, the slogger log is not so useful. The error has happened after the log.

It would be important to capture the output of  /sys/kernel/debug/sof/etrace when this happens. Did we @keqiaozhang have this in production yet (with cavstool.py)?

kv2019i, 2022-05-10:
I've been trying to reproduce this locally for a full day (not exactly same config but a cAVS2.5 I2S platform), but no luck so far. So would be really good to get the "etrace" output on CI when this is hit.

mengdonglin, 2022-09-06:
Cannot reproduce so closing it.

closed 2022-09-06 by mengdonglin