thesofproject/sof #5794: [FEATURE][meta] Use Zephyr native interfaces

kv2019i, 2022-05-10:
This is a meta issue to track work to move SOF code to use Zephyr native interfaces.

Change in brief: SOF application code will transition to Zephyr application interfaces. Change is done incrementally so that build with the "in-tree XTOS" continuous to work.

Guard rails:
 - code in sof/src/audio is the primary application code, should only use Zephyr interfaces
 - drivers in sof/src/drivers  should be moved to Zephyr upstream
 - arch code in sof/src/arch is not used for SOF Zephyr builds, this is only for XTOS builds

Transition tools:
 - sof/zephyr/wrapper.c  - xtos-to-zephyr wrapping that was used for initial introduction of Zephyr. this will grow smaller and smaller and eventually be removed  
 - src/xtensa-wrapper -  zephyr-to-xtos wrapper that can be used to build SOF app code on XTOS
 
Ongoing work:
- [ ] PM runtime move to Zephyr PM model
- [ ] Module init with native Zephyr interface #5406
- [x] Spinlocks #5286 
- [x] System clock for timing Wallclock Clocks #5393 
- [ ] Logging, partially done in #5707 
- [x] Wait #5770

lenghonglin, 2022-05-16:
> > is there any plan to move code in sof/src/platform/ to zephyr？ I want development drivers and platforms to be out of tree
> 
> Yes - we are just working out the stages for this. Topic at TSC tomorrow.

Is there something news to share after TSC？

lgirdwood, 2022-05-25:
@lenghonglin apologies, I and some other TSC member were travelling last week. 

Please see minutes here. https://github.com/orgs/thesofproject/teams/steering-committee/discussions/31

lenghonglin, 2022-05-26:
> @lenghonglin apologies, I and some other TSC member were travelling last week.
> 
> Please see minutes here. https://github.com/orgs/thesofproject/teams/steering-committee/discussions/31

but zephyr does not support DAI interface

lgirdwood, 2022-06-08:
@kv2019i we should add the DAI/DMA integration here too ?

kv2019i, 2023-01-11:
Marked multiple items as done. Added one new item for platform code init.

lgirdwood, 2023-01-11:
@abonislawski @mwasko @dbaluta any items you can add/update here ?

kv2019i, 2023-03-09:
A lot of work has been done to cleanup the XTOS/Zephyr split in SOF codebase for 2.5, and to continue the move to native calls in SOF application code. Some work is left though, so moving this meta item for v2.6 milestone. A few subitems are still queued for v2.5 and they will be included if they get merged to main before the cutoff.

lgirdwood, 2023-03-09:
@abonislawski @lyakh @juimonen anything else still todo ? cache API ?

lyakh, 2023-03-10:
> @abonislawski @lyakh @juimonen anything else still todo ? cache API ?

@lgirdwood yes, moving over to the Zephyr native `z_xtensa_cache_*()` API is one of the remaining items. `irq_local_*able()` is another XTOS API that's still widely used in SOF.

kv2019i, 2023-03-10:
Updated description of the transition tools and guarrails to include the new "{xtos/zephyr}/include/rtos" mechanism to abstract OS services. 

Also added @lyakh  the cache and local ireq control as tasks. Please do add items directly here as well. This will serve both as a tasklist, and will provide example how to do the transitions.

lgirdwood, 2023-03-13:
@dbaluta any Zephyr related interfaces pending for NXP ?

lgirdwood, 2023-05-10:
@juimonen @dbaluta are we on track for v2.6 ? Can we close ?

kv2019i, 2023-05-11:
@lgirdwood let's dis-engage this meta-item from the milestones and track the subitems with milestones instead.  FYI for @dbaluta . This can be then used for other vendors as well to help with the Zephyr migration.

dbaluta, 2023-05-11:
@kv2019i do you suggest on creating separate Issues for NXP for example, to track the switch to native interfaces?

Since Zephyr has a DAI interface, for us porting NXP DAIs should be low hanging fruits and doable for v2.6

We need to look into the DMA and also the IRQ controller as this could be trickier.

Cc: @iuliana-prodan

kv2019i, 2023-05-11:
@dbaluta I think we can keep NXP issues here as well. We can assign subitems (like the DAI port) to milestones and track them at that level.

iuliana-prodan, 2023-10-12:
@kv2019i @lgirdwood is the Zephyr native drivers for DAI, DMA, irq related with IPC4?
Meaning, can we move to zephyr native drivers for DAI, DMA and still using ipc3? or should we move to ipc4 also?

kv2019i, 2023-10-12:
@iuliana-prodan the DAI/DMA interfaces are IPC agnostic, so no  hard requirement to move to IPC4 to use native drivers. The main changes are in dai-zephyr.c/dai-legacy.c and host-zephyr.c/host-legacy.c.  There are a few IPC specific bits, but those are related to IPC functionality like bind/unbind (only in IPC4), so the main code is completely shared. Of course with Intel targets moved to IPC4, testing coverage is better with IPC4 now (would be nice to have some IPC3 devices in upstream SOF CI so we'd catch issues sooner).

dbaluta, 2023-10-12:
Cc: @LaurentiuM1234 who is working on this.

LaurentiuM1234, 2023-10-12:
Small update on NXP's side:

1. Driver status
    * irqsteer driver was tested (nothing fancy, just checked if aplay works, will most likely require a stress test to make sure everything's fine) and is done. Zephyr PR: https://github.com/zephyrproject-rtos/zephyr/pull/62776.
    * DAI/EDMA drivers are mostly done. Will upstream once I make sure everything's fine.
2. Switch to native drivers status
    * currently, I'm working on switching i.MX93 to using the Zephyr native drivers plus the dai-zephyr/host-zephyr combo.
    * so far no issue with the IPC version (as far as I can tell)
    * will probably have to twitch/rework the Zephyr DMA domain (unless we switch to the timer domain?) as things are a bit different now (e.g: IRQ management is done in the drivers, the state of the DMA channels is something internal to the driver and we can't query it ATM etc... + we have the option to use the DMA CB instead of enabling Zephyr shared interrupts)
3. To dos
    * as far as I can tell right now, before we can switch we need to sort out the domain issue and implement the rest of the drivers (SDMA, ESAI and MU). Hopefully I didn't miss anything?

lgirdwood, 2023-10-16:
@marcinszkudlinski @mwasko @abonislawski @softwarecki @kv2019i @lyakh any other nativization work that should be added to the main feature list as we are nearing completion of initial tasks.

kv2019i, 2024-04-09:
Add https://github.com/thesofproject/sof/issues/9015

LaurentiuM1234, 2024-04-25:
From NXP's side: `imx8`, `imx93`, and `imx8ulp` have all been ported to native Zephyr. All of these platforms have been switched to timer domain, which, so far, has been working well. For the switch, we've added drivers in Zephyr for the following IPs previously kept in SOF tree:

1. IRQ_STEER
2. SAI
3. ESAI
4. EDMA

lgirdwood, 2024-04-26:
@dbaluta @LaurentiuM1234 @iuliana-prodan - Great work everybody :)

I guess this now means you can remove NXP xtos code in main and do similar code clean ups like @kv2019i is doing atm for Intel to simplify your code base.

kv2019i, 2024-11-12:
Added a separate section for tasks related to "thinning out" the sof/src/platform layer for Zephyr builds. In short the the intention is to continue pruning this interface and move everything that is hardware specific, to Zephyr. This will reduce the work that needs to be done to add a new hardware target to SOF.

kv2019i, 2025-05-02:
See also enhancement ticket https://github.com/thesofproject/sof/issues/9983

kv2019i, 2025-07-03:
XTOS removal has started in main with https://github.com/thesofproject/sof/pull/10075 merged. We have now two items https://github.com/thesofproject/sof/issues/9983 and this to track removal of remaining abstractions (that are no longer needed). We can close one of these.