thesofproject/sof #5797: SOF 2.2 Release Checklist

mengdonglin, 2022-05-10:
Pre RC1 Steps:
- [ ] tag any of the issues and feature lists with milestone v2.2
- [ ] Choose a commit on main branch as candidate for branch out stable-v2.2
- [ ] Testing & sign off this commit
- [ ] branch out stable-v2.2

RC1 Steps: 
- [ ] Cherry pick critical bug fixings and features to stable-v2.2 for RC1
- [ ] Testing & sign off
- [ ] Tag v2.2-rc1
- [ ] Make rc1 release on https://github.com/thesofproject/sof/releases
- [ ] signing of the FW, upload FW and topology binaries to internal release folder
- [ ] Make rc1 release on https://github.com/thesofproject/sof-bin
- [ ] Send the mail to ALSA mailing list?

Final 2.2 steps:
- [ ] Cherry pick critical bug fixings to stable-v2.2
- [ ] Testing & sign off
- [ ] Tag v2.2
- [ ] Make release on https://github.com/thesofproject/sof/releases
- [ ] signing of the FW, upload FW and topology binaries to internal release folder
- [ ] Make release on https://github.com/thesofproject/sof-bin
- [ ] Send the mail to ALSA mailing list?
- [ ] Declare the release on https://www.sofproject.org/blog/

marc-hb, 2022-05-10:
Should this be part of https://github.com/thesofproject/sof-bin/blob/main/HOWTO-new-release.md ?

mengdonglin, 2022-05-12:
> Should this be part of https://github.com/thesofproject/sof-bin/blob/main/HOWTO-new-release.md ?

@marc-hb Do you mean we should list the required steps in https://github.com/thesofproject/sof-bin/blob/main/HOWTO-new-release.md?

marc-hb, 2022-05-12:
Yes or as another, template file close to it. Then for each release we can copy the template into a new issue like this one.

This would make the template reviewed, version controlled and available outside Github.

mengdonglin, 2022-05-13:
> Yes or as another, template file close to it. Then for each release we can copy the template into a new issue like this one.
> 
> This would make the template reviewed, version controlled and available outside Github.

@marc-hb Okay. I can work on PR to update in https://github.com/thesofproject/sof-bin/blob/main/HOWTO-new-release.md

lgirdwood, 2022-05-25:
Stable branch created.

kv2019i, 2022-06-09:
https://github.com/thesofproject/sof/releases/tag/v2.2-rc1 created

lgirdwood, 2022-06-17:
@mengdonglin any updates here ?

mengdonglin, 2022-08-30:
Closing this as v2.2 was release.

closed 2022-08-30 by mengdonglin