thesofproject/sof #5809: [BUG][IPC4]  "Failed to set up connected widgets" on second PCM open

kv2019i, 2022-05-12:
**Describe the bug**
A variant of https://github.com/thesofproject/sof/issues/5777 

FAILS:
aplay   -Dhw:0,1 -r 48000 -c 2 -f S16_LE -d 10 /dev/zero -v -q && \
aplay   -Dhw:0,1 -r 48000 -c 2 -f S16_LE -d 10 /dev/zero -v -q && \
aplay   -Dhw:0,1 -r 48000 -c 2 -f S16_LE -d 10 /dev/zero -v -q

WORKS:
aplay   -Dhw:0,1 -r 48000 -c 2 -f S16_LE -d 10 /dev/zero -v -q && \
sleep 3 && \
aplay   -Dhw:0,1 -r 48000 -c 2 -f S16_LE -d 10 /dev/zero -v -q && \
sleep 3 && \
aplay   -Dhw:0,1 -r 48000 -c 2 -f S16_LE -d 10 /dev/zero -v -q

With the sleeps, driver puts device to D3 between the aplays.

**Reproduction Rate**
100%

**Expected behavior**
Test should pass even without the pauses.

**Logs** 
When run without the sleeps (so driver doesn't go to D3 between the two aplay runs), following error is hit:

[  846.821629] sof-audio-pci-intel-tgl 0000:00:1f.3: pipeline: 3 memory pages: 1
[  846.821631] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x11030001|0x0: GLB_CREATE_PIPELINE
[  847.321948] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc timed out for 0x11030001|0x0
[  847.321964] sof-audio-pci-intel-tgl 0000:00:1f.3: failed to create module pipeline.3
[  847.321970] sof-audio-pci-intel-tgl 0000:00:1f.3: Failed to set up connected widgets
[  847.322541] sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: trigger stream 1 dir 0 cmd 1

**Impact**
sof-test check-playback.sh fails

**Environment**
Same as for https://github.com/thesofproject/sof/issues/5777

kv2019i, 2022-05-12:
FYI @RanderWang , @ranj063 and @miRoox

kv2019i, 2022-05-12:
If logging is enabled with CONFIG_ZEPHYR_LOG=y (all SOF logs go to SRAM window via Zephyr logging backend), the bug is not reproduced. This suggests timing plays a role in triggering the problem. Logging via SRAM changes DSP execution timing.

ranj063, 2022-05-12:
@kv2019i do you have a typo in your command? Ar eyou really trying to open the same PCM device three times?
``` 
aplay -Dhw:0,1 -r 48000 -c 2 -f S16_LE -d 10 /dev/zero -v -q &&
aplay -Dhw:0,1 -r 48000 -c 2 -f S16_LE -d 10 /dev/zero -v -q &&
aplay -Dhw:0,1 -r 48000 -c 2 -f S16_LE -d 10 /dev/zero -v -q
```

lgirdwood, 2022-05-13:
@mwasko @marcinszkudlinski fyi - not sure if you have seen this issue yet.

mwasko, 2022-05-16:
@kv2019i I will ask firmware validation team to verify if we can reproduce this issue in scenario with Diagnostic Driver.  

I would also like to ask you to do an experiment and add background capture stream (that will prevent D3 entry) and then run:

aplay -Dhw:0,1 -r 48000 -c 2 -f S16_LE -d 10 /dev/zero -v -q
sleep 3
aplay -Dhw:0,1 -r 48000 -c 2 -f S16_LE -d 10 /dev/zero -v -q
sleep 3
aplay -Dhw:0,1 -r 48000 -c 2 -f S16_LE -d 10 /dev/zero -v -q

If the issue will reproduce it will indicate that we have some pipeline clean up issue that require D3 transition as a w/a before next playback start. However if the issue will not reproduce with the background capture stream then I would put my bet on race condition in how driver interact with the firmware.

Does the issue reproduce only in IPC4 + Zephyr configuration? What about IPC3 + Zephyr, IPC4 + SOF XTOS, IPC3 + SOF XTOS configurations?

kv2019i, 2022-05-23:
Tested this with latest main as of today and issue no longer happens. Most probably #5817 is the fix. Closing.

closed 2022-05-23 by kv2019i