thesofproject/sof #5822: [BUG][MTL] 4xch DMIC capture returned

yongzhi1, 2022-05-14:
**Describe the bug**
With 4xch DMIC capture returns:
arecord: pcm_read:2178: read error: Input/output error

**Topology**
diff --git a/tools/topology/topology2/cavs/CMakeLists.txt b/tools/topology/topology2/cavs/CMakeLists.txt
index 9caab6880..33a396cde 100644
--- a/tools/topology/topology2/cavs/CMakeLists.txt
+++ b/tools/topology/topology2/cavs/CMakeLists.txt
@@ -29,7 +29,7 @@ PREPROCESS_PLUGINS=nhlt,NHLT_BIN=nhlt.bin"
 "cavs-nocodec\;cavs-tgl-nocodec\;NUM_DMICS=4,PDM1_MIC_A_ENABLE=1,PDM1_MIC_B_ENABLE=1,\
 PREPROCESS_PLUGINS=nhlt,NHLT_BIN=nhlt-cavs-tgl-nocodec.bin"
-"cavs-nocodec\;sof-mtl-nocodec\;PLATFORM=mtl,NUM_DMICS=2,\
+"cavs-nocodec\;sof-mtl-nocodec\;PLATFORM=mtl,NUM_DMICS=4,PDM1_MIC_A_ENABLE=1,PDM1_MIC_B_ENABLE=1,\
 PREPROCESS_PLUGINS=nhlt,NHLT_BIN=nhlt-ace-mtl-nocodec.bin"
 )
 
**To reproduce**
~$ arecord -Dhw:0,10 -c 4 -r 48000 -f S32_LE -vv /tmp/test.wav
Recording WAVE '/tmp/test.wav' : Signed 32 bit Little Endian, Rate 48000 Hz, Channels 4
Hardware PCM card 0 'sof-nocodec' device 10 subdevice 0
Its setup is:
stream : CAPTURE
access : RW_INTERLEAVED
format : S32_LE
subformat : STD
channels : 4
rate : 48000
exact rate : 48000 (48000/1)
msbits : 32
buffer_size : 4096
period_size : 1024
period_time : 21333
tstamp_mode : NONE
tstamp_type : MONOTONIC
period_step : 1
avail_min : 1024
period_event : 0
start_threshold : 1
stop_threshold : 4096
silence_threshold: 0
silence_size : 0
boundary : 4611686018427387904
appl_ptr : 0
hw_ptr : 0
arecord: pcm_read:2178: read error: Input/output error

Note:
DMIC has two entry for chan_ctrl_cfg[2], but only dmic_dai_index 0 has been updated here:
https://github.com/alsa-project/alsa-utils/blob/master/topology/nhlt/intel/dmic/dmic-process.c#L818

juimonen, 2022-05-16:
the dmic dai index 1 is for 16khz fifo. So we should be able to run 48 khz fifo also with 4 ch

juimonen, 2022-05-25:
fix in https://github.com/thesofproject/sof/pull/5837

closed 2022-05-28 by yongzhi1