thesofproject/sof #5840: [BUG] Captured signal is silent in TestMixerPlayback

wszypelt, 2022-05-20:
**Describe the bug**
Captured signal is silent in TestMixerPlayback
Issue was found in:
[ICL]
12_00_TestMixerPlayback16000Hz16b16b8ch Repro: 1/75

**Topology**
Description:
Simultaneous playback and capture on external loopback (same SSP) with signal verification.
Playback pipeline contains SRC component and plays audio formats with different frequencies.
```
Topology:
                plb_ppl
                +--------------------------------+
    +------+    | +---+  +- -+  +---+  +-------+ |
    | Host |----->|Buf|->|Src|->|Buf|->|SSP Dai|---+
    +------+    | +---+  +-- +  +---+  +-------+ | |
                +--------------------------------+ |
                                                   |
                cap_ppl                            |
                +------------------+               |
    +------+    | +---+  +-------+ |               |
    | Host |<-----|Buf|<-|SSP Dai|<----------------+
    +------+    | +---+  +-------+ |
                +------------------+

```

**To Reproduce**
Run tests with Diagnostic driver:
12_00_TestMixerPlayback16000Hz16b16b8ch Repro: 1/75
With options --playback-iterations=75

**Reproduction Rate**
1/75

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * SOF BRANCH: main HASH: 6c4f3ea42f0d04e9a421dd8325cdbc9be31056bf
2) Name of the platform(s) on which the bug is observed.
    * Platform: ICL RVP, APL RVP
3) Test environment:
BRANCH: master HASH: 4c2b56f9d4e4519a5fe456ea2ef33f166ea6a45d

**Screenshots or console output**
```
23:45:14.206       => Wait for Streamers to play out of data
23:45:17.355       => Stop playback and capture streams
23:45:17.495       => Free streams
23:45:17.685     => Free components
23:45:17.716   => Post-iteration cleanup
23:45:17.716   => Verification
23:45:17.716     => Topology Iteration 0, Playback Iteration 0
23:45:17.779       Captured signal is silent

23:45:17.779     => Topology Iteration 0, Playback Iteration 1
23:45:17.922     => Topology Iteration 0, Playback Iteration 2
23:45:18.078     => Topology Iteration 0, Playback Iteration 3
23:45:18.220     => Topology Iteration 0, Playback Iteration 4
23:45:18.361     => Topology Iteration 0, Playback Iteration 5
23:45:18.517     => Topology Iteration 0, Playback Iteration 6
23:45:18.658     => Topology Iteration 0, Playback Iteration 7
23:45:18.803     => Topology Iteration 0, Playback Iteration 8
23:45:18.959     => Topology Iteration 0, Playback Iteration 9
23:45:19.087     => Topology Iteration 0, Playback Iteration 10
23:45:19.227     => Topology Iteration 0, Playback Iteration 11
```
[image]

**Logs**
[12_00_TestMixerPlayback16000Hz16b16b8ch.zip](https://github.com/thesofproject/sof/files/8738731/12_00_TestMixerPlayback16000Hz16b16b8ch.zip)

wszypelt, 2022-05-25:
@lgirdwood My mistake, I have already corrected the description and topology

dkrzysiak, 2022-05-26:
Same mistake occurred in these 4 cases:

12_00_TestMixerPlayback8000Hz24b32b8ch
log:
[TMP_8000_24_32_8.zip](https://github.com/thesofproject/sof/files/8780983/TMP_8000_24_32_8.zip)

12_00_TestMixerPlayback16000Hz32b32b4ch
log:
[TMP_16000_32_32_4.zip](https://github.com/thesofproject/sof/files/8780989/TMP_16000_32_32_4.zip)

12_00_TestMixerPlayback16000Hz24b32b8ch
log:
[TMP_16000_24_32_8.zip](https://github.com/thesofproject/sof/files/8780991/TMP_16000_24_32_8.zip)

12_00_TestMixerPlayback24000Hz32b32b4ch
log:
[TMP_24000_32_32_4.zip](https://github.com/thesofproject/sof/files/8780994/TMP_24000_32_32_4.zip)

closed 2025-07-25 by github-actions[bot]