thesofproject/sof #5841: [BUG] THD value is larger than acceptable in TestEqFirSimple

dkrzysiak, 2022-05-20:
**Describe the bug**
THD value is larger than acceptable in TestEqFirSimple
Issue found in:
[ICL]
20_00_TestEqFirSimple8000Hz32b32b4ch

**Topology**
Description:
Test is performing simultaneous playback and capture on external SSP interface with recorded signal quality verification.
Playback pipeline contains FIR EQ component with basic configuration.
```

Topology:
+---------------------------------------------------+
| +------+   +-----+   +------+   +-----+   +-----+ |   +----+
| | Host |-->| Buf |-->|FIR EQ|-->| Buf |-->| Dai |---->|SSPx|
| +------+   +-----+   +------+   +-----+   +-----+ |   +----+
+---------------------------------------------------+    |  |
                                                         |  |
                                                         |  |
+-------------------------------+                        |  |
| +------+   +-----+    +-----+ |                       +----+
| | Host |<--| Buf |<---| Dai |<------------------------|SSPx|
| +------+   +-----+    +-----+ |                       +----+
+-------------------------------+

```

**To Reproduce**
Run tests with Diagnostic driver:
20_00_TestEqFirSimple8000Hz32b32b4ch Repro: 1/75
With options --playback-iterations=75

**Reproduction Rate**
1/75

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * SOF BRANCH: main HASH: https://github.com/thesofproject/sof/commit/6c4f3ea42f0d04e9a421dd8325cdbc9be31056bf
2) Name of the platform(s) on which the bug is observed.
    * Platform: ICL RVP, APL RVP
3) Test environment:
   BRANCH: master HASH: 4c2b56f9d4e4519a5fe456ea2ef33f166ea6a45d

**Screenshots or console output**
01:54:18.354     => Topology Iteration 0, Playback Iteration 0
01:54:18.434       audio_verification.py:74, in log_result()
                    THD value is larger than acceptable value (-80.00) on channel 0: -9.16

01:54:18.434       audio_verification.py:74, in log_result()
                    THD value is larger than acceptable value (-80.00) on channel 1: -11.47

01:54:18.434       audio_verification.py:74, in log_result()
                    THD value is larger than acceptable value (-80.00) on channel 2: -10.85

01:54:18.439       audio_verification.py:74, in log_result()
                    THD value is larger than acceptable value (-80.00) on channel 3: -10.84

[image]
[image]

[12_00_EqFir.zip](https://github.com/thesofproject/sof/files/8739923/12_00_EqFir.zip)

lgirdwood, 2022-05-25:
@dkrzysiak do we know if these are pipeline glitches or processing quality issues ?

dkrzysiak, 2022-05-25:
@lgirdwood I would bet on pipeline, repro was performed locally on pre-checked ICL. To my mind, machine error was excluded.

antonirx, 2023-01-05:
unsupported platform
on TGL issue does not occur
Tested FW hash: 4943aaa

closed 2023-01-05 by wszypelt