thesofproject/sof #5842: [BUG] Speech synthesis always skips randomly on speakers

Tatsh, 2022-05-20:
**Describe the bug**
When I watch a Twitch stream or use the speech synthesis API in Chrome, the start of the TTS always gets skipped, and from there parts are skipped randomly.

If you try to have Chrome say 'Hello World', it just says 'o' and cuts off everything else. Longer messages play a bit more but the beginning is still cut off. The monitor shows the same result so it is not the speakers.

Sample: [hello-world-tts.wav.zip](https://github.com/thesofproject/sof/files/8740860/hello-world-tts.wav.zip)

**To Reproduce**

In Chrome or similar, run the following in console:

```
var msg = new SpeechSynthesisUtterance('Hello World');
window.speechSynthesis.speak(msg);
```

**Reproduction Rate**
All the time.

**Expected behavior**
The speech should play normally.

**Impact**
Very annoying to watch certain Twitch streams.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 5.17.7
    * SOF: 2.1.1
2) Name of the topology file
    * Topology: `sof-hda-generic-2ch.tplg`
3) Name of the platform(s) on which the bug is observed.
    * Platform: TigerLake
   
```
snd_hda_intel 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
sof-audio-pci-intel-tgl 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
sof-audio-pci-intel-tgl 0000:00:1f.3: enabling device (0000 -> 0002)
sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
sof-audio-pci-intel-tgl 0000:00:1f.3: bound 0000:00:02.0 (ops 0xffffffffab8bacc0)
sof-audio-pci-intel-tgl 0000:00:1f.3: use msi interrupt mode
sof-audio-pci-intel-tgl 0000:00:1f.3: hda codecs found, mask 5
sof-audio-pci-intel-tgl 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
sof-audio-pci-intel-tgl 0000:00:1f.3: DMICs detected in NHLT tables: 2
Loading firmware: intel/sof/sof-tgl-h.ri
sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 2:1:1-3964a
sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:21:0 Kernel ABI 3:18:0
sof-audio-pci-intel-tgl 0000:00:1f.3: warn: FW ABI is more recent than kernel
sof-audio-pci-intel-tgl 0000:00:1f.3: unknown sof_ext_man header type 3 size 0x30
sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 2:1:1-3964a
sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:21:0 Kernel ABI 3:18:0
sof-audio-pci-intel-tgl 0000:00:1f.3: warn: FW ABI is more recent than kernel
Loading firmware: intel/sof-tplg/sof-hda-generic-2ch.tplg
sof-audio-pci-intel-tgl 0000:00:1f.3: Topology: ABI 3:21:0 Kernel ABI 3:18:0
sof-audio-pci-intel-tgl 0000:00:1f.3: warn: topology ABI is more recent than kernel
input: sof-hda-dsp Mic as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card1/input16
input: sof-hda-dsp Headphone as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card1/input17
input: sof-hda-dsp HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card1/input21
input: sof-hda-dsp HDMI/DP,pcm=4 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card1/input22
input: sof-hda-dsp HDMI/DP,pcm=5 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card1/input23
Loading firmware: intel/sof-tplg/sof-hda-generic-2ch.tplg
Loading firmware: intel/sof/sof-tgl-h.ri
``

Kernel config:

```
CONFIG_SND_SOC_SOF_TOPLEVEL=y
CONFIG_SND_SOC_SOF_PCI_DEV=y
CONFIG_SND_SOC_SOF_PCI=y
CONFIG_SND_SOC_SOF=y
CONFIG_SND_SOC_SOF_PROBE_WORK_QUEUE=y
CONFIG_SND_SOC_SOF_INTEL_TOPLEVEL=y
CONFIG_SND_SOC_SOF_INTEL_COMMON=y
CONFIG_SND_SOC_SOF_INTEL_TGL=y
CONFIG_SND_SOC_SOF_TIGERLAKE=y
CONFIG_SND_SOC_SOF_HDA_COMMON=y
CONFIG_SND_SOC_SOF_HDA_LINK=y
CONFIG_SND_SOC_SOF_HDA_AUDIO_CODEC=y
CONFIG_SND_SOC_SOF_HDA_LINK_BASELINE=y
CONFIG_SND_SOC_SOF_HDA=y
CONFIG_SND_SOC_SOF_INTEL_SOUNDWIRE_LINK_BASELINE=y
CONFIG_SND_SOC_SOF_XTENSA=y
```

kv2019i, 2022-05-24:
@Tatsh Which OS are you using? ChromeOS or some Linux distro?

singalsu, 2022-05-24:
@Tatsh The attached wav file does not represent what you hear when the bug happens. Please record the output that you get.

Tatsh, 2022-05-24:
> @Tatsh Which OS are you using? ChromeOS or some Linux distro?

Gentoo

plbossart, 2022-05-24:
@Tatsh if this was due to the SOF firmware/driver, you would see the same effect on headphones, can you please double-check? If you don't then it's an amplifier issue.

Tatsh, 2022-05-25:
> @Tatsh if this was due to the SOF firmware/driver, you would see the same effect on headphones, can you please double-check? If you don't then it's an amplifier issue.

You are correct. With headphones the issue never occurs.

lgirdwood, 2022-05-25:
@Tatsh any idea on which speaker amplifier is used on your device ? There are Realtek folks here so may be able to provide help if it's a Realtek amp.

Tatsh, 2022-05-25:
@lgirdwood it is Realtek. On the MSI website for Windows drivers:

[image]

lgirdwood, 2022-05-25:
@bardliao which Realtek folks would be able to give help here ?

Tatsh, 2022-05-25:
Here's alsa-info and I will report after rebooting on the option change.
[alsa-info.txt](https://github.com/thesofproject/sof/files/8774491/alsa-info.txt)

Tatsh, 2022-05-25:
Adding the option did not make a difference. I don't see proof that it used the option but I did try putting it on the command line with `snd-intel-dspcfg.dsp_driver=1` and this did not make a difference either.

```
sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
sof-audio-pci-intel-tgl 0000:00:1f.3: hda codecs found, mask 5
sof-audio-pci-intel-tgl 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
Loading firmware: intel/sof-tplg/sof-hda-generic-2ch.tplg
skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: Parent card not yet available, widget card binding deferred
snd_hda_codec_realtek ehdaudio0D0: autoconfig for ALC298: line_outs=1 (0x17/0x0/0x0/0x0/0x0) type:speaker
snd_hda_codec_realtek ehdaudio0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
snd_hda_codec_realtek ehdaudio0D0:    hp_outs=1 (0x21/0x0/0x0/0x0/0x0)
snd_hda_codec_realtek ehdaudio0D0:    mono: mono_out=0x0
snd_hda_codec_realtek ehdaudio0D0:    inputs:
snd_hda_codec_realtek ehdaudio0D0:      Mic=0x18
snd_hda_codec_realtek ehdaudio0D0: ASoC: sink widget AIF1TX overwritten
snd_hda_codec_realtek ehdaudio0D0: ASoC: source widget AIF1RX overwritten
skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: sink widget hifi3 overwritten
skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: sink widget hifi2 overwritten
skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: sink widget hifi1 overwritten
skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: source widget Codec Output Pin1 overwritten
skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: sink widget Codec Input Pin1 overwritten
skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: sink widget Analog Codec Playback overwritten
skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: sink widget Digital Codec Playback overwritten
skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: sink widget Alt Analog Codec Playback overwritten
skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: source widget Analog Codec Capture overwritten
skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: source widget Digital Codec Capture overwritten
skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: source widget Alt Analog Codec Capture overwritten
skl_hda_dsp_generic skl_hda_dsp_generic: hda_dsp_hdmi_build_controls: no PCM in topology for HDMI converter 3
input: sof-hda-dsp Mic as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input10
input: sof-hda-dsp Headphone as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input11
input: sof-hda-dsp HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input15
input: sof-hda-dsp HDMI/DP,pcm=4 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input16
input: sof-hda-dsp HDMI/DP,pcm=5 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input17
snd_hda_intel 0000:01:00.1: enabling device (0000 -> 0002)
snd_hda_intel 0000:01:00.1: Disabling MSI
input: HDA NVidia HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:01.0/0000:01:00.1/sound/card1/input21
input: HDA NVidia HDMI/DP,pcm=7 as /devices/pci0000:00/0000:00:01.0/0000:01:00.1/sound/card1/input22
input: HDA NVidia HDMI/DP,pcm=8 as /devices/pci0000:00/0000:00:01.0/0000:01:00.1/sound/card1/input23
input: HDA NVidia HDMI/DP,pcm=9 as /devices/pci0000:00/0000:00:01.0/0000:01:00.1/sound/card1/input24
input: HDA NVidia HDMI/DP,pcm=10 as /devices/pci0000:00/0000:00:01.0/0000:01:00.1/sound/card1/input25
input: HDA NVidia HDMI/DP,pcm=11 as /devices/pci0000:00/0000:00:01.0/0000:01:00.1/sound/card1/input26
```

plbossart, 2022-05-25:
> Here's alsa-info and I will report after rebooting on the option change. [alsa-info.txt](https://github.com/thesofproject/sof/files/8774491/alsa-info.txt)

ok, so bog standard ALC298 codec with 2 DMICs. That's been used by tons of people...

plbossart, 2022-05-25:
@Tatsh the option clearly didn't take effect, you should only see snd-hda-intel instead.

bardliao, 2022-05-26:
> @bardliao which Realtek folks would be able to give help here ?

I always ping @shumingfan :)

Tatsh, 2022-05-26:
I think I booted correctly now. There is no mention of dsp in the boot log. pavucontrol shows more than just the `Play HiFi` profile.

The issue persists in the no-SOF/DSP mode.

shumingfan, 2022-05-26:
@Tatsh I reported this issue to the owner of Realtek engineer and copied his comments below.

Then end users can try Windows to see if the situation is the same.
If it is, it is confirmed to be a hardware (hardware codec) problem, please end user to change the machine.
If not, please end user to give us more details.
"sudo su root" type password  to enter root
"echo 1 > /sys/module/snd_hda_codec/parameters/dump_coef "
"alsa-info --no-upload"
give us "/tmp/alsa-info.txt.xxxx"

Tatsh, 2022-05-26:
Updated alsa-info: [alsa-info.txt](https://github.com/thesofproject/sof/files/8782283/alsa-info.txt)

Windows does not have the problem. I ran the same videos, ran speech synthesis, and had no issues.

Tatsh, 2025-06-06:
Hi, can I get an update on this? The issue still exists.

bardliao, 2025-06-09:
@Tatsh if I understand correctly, you generated a speech, but the start of the speech is skipped when you play with speaker and monitor. And you can hear the whole speech if you play with a headphone. Am I correct? Do you know how long will the speech be skipped? What if you record a wav file form another device, like your cellphone and play the wav file on your laptop's speaker? Try to speak before start recording. We need to make sure there is audio data at the very beginning of the recorded file.
Also, please do record what you hear as [singalsu](https://github.com/singalsu) commented https://github.com/thesofproject/sof/issues/5842#issuecomment-1135538249.
Last, can you try with the latest kernel? It might fix the issue.
@KailangYang Could you take a look? Please find the alsa-info https://github.com/thesofproject/sof/issues/5842#issuecomment-1139083270

Tatsh, 2025-06-09:
The issue I am having is not limited to speech synthesis. It is just easily reproducible that way. In general the speakers are just having issues where audio skips randomly compared to headphones. It's not a clip that one might think is caused by a buffering issue, but it is definitely something else.

bardliao, 2025-06-09:
> The issue I am having is not limited to speech synthesis. It is just easily reproducible that way. In general the speakers are just having issues where audio skips randomly compared to headphones. [...]

Sorry, but what do you mean by "audio skips randomly"? Does the audio be skipped in the middle of the audio stream? Or the skipped part is always at the beginning? If the audio is "abcdefg", will it be "abcefg" I.e. the "d" is skipped or only could be "bcdefg", "cdefg", etc? I.e. The audio at the beginning could be skipped.

Tatsh, 2026-09-09:
I sent a patch: https://lore.kernel.org/all/20260909193104.26508-1-audvare@gmail.com