thesofproject/sof #5867: [BUG]

jimmychm, 2022-05-29:
**Describe the bug**
I have been trying to install Linux on my system (UX482EG)

**To Reproduce**
I haven't been able to pinpoint what causes the sound to stop working when it initially does, but whatever it is that causes it to stop working, happens quite 

**Reproduction Rate**
How often does the issue happen ? i.e. 1/10 (once in ten attempts), 1/1000 or all the time.
Does the reproduction rate vary with any other configuration or user action, if so please describe and show the new reproduction rate.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {SHA}
    * SOF: {SHA}
2) Name of the topology file
    * Topology: {FILE}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {PLATFORM}

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

Please also include the relevant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). Kernel log is taken from *dmesg* and firmware log from *sof-logger*. See https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

jonah-chen, 2022-05-30:
I have the same issues on the Asus Zenbook UP5401EA with the LTS kernel as well.
```
$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sofhdadsp [sof-hda-dsp], device 0: HDA Analog (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 1: HDA Digital (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 3: HDMI1 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 4: HDMI2 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 5: HDMI3 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

$ inxi -A 
Audio:
  Device-1: Intel Tiger Lake-LP Smart Sound Audio
    driver: sof-audio-pci-intel-tgl
  Sound Server-1: ALSA v: k5.15.43-1-lts running: yes
  Sound Server-2: PulseAudio v: 15.0 running: yes
  Sound Server-3: PipeWire v: 0.3.51 running: yes
```

jonah-chen, 2022-06-01:
I was able to fix my issue by flashing the BIOS to an older version. However when I flashed the BIOS back to the latest version, the issue did not resurface. Hence, I believe this issue is likely caused by something else that is now resolved.

The original poster's issue is likely different. Thanks.

hereblur, 2022-06-28:
I have similar problem with my zenbook ux3402.

Sound works perfect on Bluetooth and 3.5mm but no sound on internal speakers.

I tried on fedora 36, ubuntu live-usb (22), even compiled the kernel (5.19.0-rc3).
[
[alsa-info.txt](https://github.com/thesofproject/sof/files/9000329/alsa-info.txt)
](url)

plbossart, 2022-06-28:
@hereblur Here's the reference from @ryanprescott to snoop verbs 

https://github.com/ryanprescott/realtek-verb-tools/wiki/How-to-sniff-verbs-from-a-Windows-sound-driver

aleksfadini, 2022-07-23:
I am having the same problem with a ZenBook Pro Duo UX582ZM.

The speakers do not appear in alsamixer. The headphones do, and sound is fine through headphones jack.

$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sofhdadsp [sof-hda-dsp], device 0: HDA Analog (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 3: HDMI1 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 4: HDMI2 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 5: HDMI3 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

I followed the @ryanprescott tutorial to snoop verbs, because for a previous model (UX582XT), there were a series of hda-verbs that do not work with mine. Unfortunately, I managed to compile QEMU and enable the right parameters on the kernel, but apparently I can't get windows 10 or windows 11 to get sound in QEMU, despite installing what I think are the correct drivers. Any other ideas?

lgirdwood, 2022-07-25:
@bardliao would anyone from Realtek be able to suggest the correct verbs here ?

bardliao, 2022-07-26:
> @bardliao would anyone from Realtek be able to suggest the correct verbs here ?

Kailang <kailang@realtek.com> and Pshou <pshou@realtek.com> may help.