thesofproject/sof #5877: [BUG]

dkrzysiak, 2022-06-01:
**Describe the bug**
BUG was found in:
24_00_TestBluetoothOffloadSsp16b16b2ch
24_00_TestBluetoothOffloadSsp24b32b4ch
24_00_TestBluetoothOffloadSsp32b32b8ch

Error content:
ERROR:Exception SofIpcErrorResponseException occurred at 'C:\\work\\tools\\pyexecnetcache\\cavs_scripts-py\\utilities\\ipc_driver.py':756 FW replied with error. Error code: -16 0x00000005 0x00000000 0x00004000

Last working build: SOF_FW_MASTER_01.01.00.3357

**Topology**
Description:
Simultaneous playback and capture on external loopback (same SSP) with SSP reconfiguration for different sampling rates.

```
Topology:
                        pipe_plb
                        +--------------------+
        +----------+    | +------+   +-----+ |   +----+
        | Host plb |----->|Buffer|-->| Dai |---->|SSP0|
        +----------+    | +------+   +-----+ |   +----+
                        +--------------------+    |  |
                                                  |  |
                        pipe_cap                  |  |
                        +--------------------+    |  |
        +----------+    | +------+   +-----+ |   +----+
        | Host cap |<-----|Buffer|<--| Dai |---->|SSP0|
        +----------+    | +------+   +-----+ |   +----+
                        +--------------------+
```

**To Reproduce**
Run tests with diagnostic driver
24_00_TestBluetoothOffloadSsp16b16b2ch
24_00_TestBluetoothOffloadSsp24b32b4ch
24_00_TestBluetoothOffloadSsp32b32b8ch

**Reproduction Rate**
100%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * SOF BRANCH: main HASH: https://github.com/thesofproject/sof/commit/6c4f3ea42f0d04e9a421dd8325cdbc9be31056bf
2) Name of the platform(s) on which the bug is observed.
    * Platform: TGL HDA
3) Test environment:
   BRANCH: master HASH: a145313efc604a967d057843d7c27c1aee7add7f

**Screenshots or console output**
[logs.zip](https://github.com/thesofproject/sof/files/8812638/logs.zip)

[image]

kfrydryx, 2022-06-07:
Can you confirm ssp port is enabled?
Is this a new scenario or a regression on ssp loopback?

dkrzysiak, 2022-06-13:
Confirmed, regression on ssp loopback.

wszypelt, 2023-01-02:
TGL issue does not occur
Tested FW hash: [4943aaa](https://github.com/thesofproject/sof/commit/4943aaa8907149cf6c02e9046e2ef5d40ba72d74)

closed 2023-01-02 by wszypelt