thesofproject/sof #5898: [BUG] IPC timed out for 0x60050000(TRIG_STOP) when testing playback on BDW

keqiaozhang, 2022-06-07:
**Describe the bug**
CI observed this issue on BDW_WSB_RT286 when testing playback. 
```
[  634.428450] kernel: sof-audio-acpi-intel-bdw INT3438:00: pcm: trigger stream 1 dir 0 cmd 0
[  634.428482] kernel: sof-audio-acpi-intel-bdw INT3438:00: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  634.932393] kernel: sof-audio-acpi-intel-bdw INT3438:00: ipc tx timed out for 0x60050000 (msg/reply size: 12/12)
[  634.932459] kernel: sof-audio-acpi-intel-bdw INT3438:00: preventing DSP entering D3 state to preserve context
[  634.932467] kernel: sof-audio-acpi-intel-bdw INT3438:00: ------------[ DSP dump start ]------------
[  634.932508] kernel: sof-audio-acpi-intel-bdw INT3438:00: IPC timeout
[  634.932539] kernel: sof-audio-acpi-intel-bdw INT3438:00: fw_state: SOF_FW_BOOT_COMPLETE (6)
[  634.932653] kernel: sof-audio-acpi-intel-bdw INT3438:00: unexpected fault 0x00000000 trace 0x80000000
[  634.932699] kernel: sof-audio-acpi-intel-bdw INT3438:00: error: ipc host -> DSP: pending yes complete no raw 0x80000000
[  634.932749] kernel: sof-audio-acpi-intel-bdw INT3438:00: error: mask host: pending no complete no raw 0x7fff0000
[  634.932795] kernel: sof-audio-acpi-intel-bdw INT3438:00: error: ipc DSP -> host: pending no complete no raw 0x00000000
[  634.932844] kernel: sof-audio-acpi-intel-bdw INT3438:00: error: mask DSP: pending no complete no raw 0x7f9e0000
[  634.932891] kernel: sof-audio-acpi-intel-bdw INT3438:00: ------------[ DSP dump end ]------------
[  634.932946] kernel: sof-audio-acpi-intel-bdw INT3438:00: ASoC: error at soc_component_trigger on INT3438:00: -110
[  634.932970] kernel: sof-audio-acpi-intel-bdw INT3438:00: trace IO error
[  634.933042] kernel:  Media Playback 1: ASoC: trigger FE cmd: 0 failed: -110
[  634.933334] kernel: sof-audio-acpi-intel-bdw INT3438:00: pcm: free stream 1 dir 0
[  634.933354] kernel: sof-audio-acpi-intel-bdw INT3438:00: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  635.436370] kernel: sof-audio-acpi-intel-bdw INT3438:00: ipc tx timed out for 0x60030000 (msg/reply size: 12/12)
[  635.436436] kernel: sof-audio-acpi-intel-bdw INT3438:00: preventing DSP entering D3 state to preserve context
```

**To Reproduce**
`~/sof-test/test-case/check-playback.sh -d 1 -l 100 -r 1`

**Reproduction Rate**
Only observed this issue once in CI, no reproductions by manual check.

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: [f7d6f01d0d16](https://github.com/thesofproject/linux/commit/f7d6f01d0d16)

SOF Branch: main
SOF Commit: [ebc222dab568](https://github.com/thesofproject/sof/commit/ebc222dab568)

Platform: BDW Wilson Beach notebook.

[logger.txt](https://github.com/thesofproject/sof/files/8849572/logger.txt)
[dmesg.txt](https://github.com/thesofproject/sof/files/8849574/dmesg.txt)

lenghonglin, 2022-06-13:
ll_schedule or zephyr_ll ? 

https://github.com/thesofproject/sof/pull/5912

lgirdwood, 2022-06-13:
> ll_schedule or zephyr_ll ?
> 
> see this #5912

ll_schedule, BDW platform not yet on Zephyr.

lenghonglin, 2022-06-14:
> > ll_schedule or zephyr_ll ?
> > see this #5912
> 
> ll_schedule, BDW platform not yet on Zephyr.

I met this issue before, the reason is the `edf_workq` have not be schedule, maybe the `pipe-task` not suspend by itself, So other task could not have chance to run.

mengdonglin, 2022-06-21:
Move BDW platform to maintenance mode and SOF v2.2 will be the last upstream release with support for this platform. There will be no more bug fixing effort on this platform.

closed 2022-06-21 by mengdonglin