thesofproject/sof #5904: [BUG] Failed to load intel/sof/sof-adl.ri (-2) Dummy output.

hyunlee1o, 2022-06-08:
**Describe the bug**

**What have you tried to diagnose or workaround this issue?**
I tried to install firmware-sof-signed, also also followed the installer from this repo correctly. Didnt work.

**To Reproduce**
Just boot.

**Reproduction Rate**
Works on each boot

**Expected behavior**
Sound working.

**Impact**
Sound doesn't work

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 5.17.0-1-amd64
    * SOF: 2.1.1
2) Name of the topology file
    * Topology: ???
3) Name of the platform(s) on which the bug is observed.
    * Platform: Debian bookworm

**Screenshots or console output**
```
[    7.934554] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    7.934571] sof-audio-pci-intel-tgl 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    7.934582] sof-audio-pci-intel-tgl 0000:00:1f.3: enabling device (0000 -> 0002)
[    7.934709] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    7.934798] sof-audio-pci-intel-tgl 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    7.941656] sof-audio-pci-intel-tgl 0000:00:1f.3: use msi interrupt mode
[    7.988015] typec port0: bound usb1-port2 (ops connector_ops [usbcore])
[    7.988037] typec port0: bound usb4-port1 (ops connector_ops [usbcore])
[    8.013011] sof-audio-pci-intel-tgl 0000:00:1f.3: hda codecs found, mask 5
[    8.013015] sof-audio-pci-intel-tgl 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[    8.013018] sof-audio-pci-intel-tgl 0000:00:1f.3: DMICs detected in NHLT tables: 2
[    8.013041] sof-audio-pci-intel-tgl 0000:00:1f.3: firmware: failed to load intel/sof/sof-adl.ri (-2)
[    8.013044] sof-audio-pci-intel-tgl 0000:00:1f.3: Direct firmware load for intel/sof/sof-adl.ri failed with error -2
[    8.013046] sof-audio-pci-intel-tgl 0000:00:1f.3: error: sof firmware file is missing, you might need to
[    8.013047] sof-audio-pci-intel-tgl 0000:00:1f.3:        download it from https://github.com/thesofproject/sof-bin/
[    8.013048] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to load DSP firmware -2
[    8.101445] sof-audio-pci-intel-tgl 0000:00:1f.3: error: sof_probe_work failed err: -2
```
Please also include the re
[dmesg.log](https://github.com/thesofproject/sof/files/8859855/dmesg.log)
levant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). Kernel log is taken from *dmesg* and firmware log from *sof-logger*. See https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

kv2019i, 2022-06-08:
@hyunlee1o Do you have /lib/firmware/intel/sof/sof-adl.ri on your system? It would seem something has gone wrong in the FW file install process.

hyunlee1o, 2022-06-08:
I even installed the specific debian package `firmware-sof-signed `for this and it did absolutely nothing.
Not even `isenkram-autoinstall` thing did help.

hyunlee1o, 2022-06-08:
Supposedly i am using the latest debian testing branch so this is why it should work.
Thank you for being so nice with this kinda undocumented report.
Im closing then.

closed 2022-06-08 by hyunlee1o