thesofproject/sof #5909: [BUG]APL UP2 NOCODEC playback ipc error when PCM_PARAMS at the 164th loop

XiaoyunWu6666, 2022-06-13:
**Describe the bug**
ipc error when PCM_PARAMS at the 64/100 loop on 2rd pipeline , OR at 164/200 loop on 1st pipeline when doing stress playbacks.

**To Reproduce**
```TPLG=/lib/firmware/intel/sof-tplg/sof-apl-nocodec.tplg MODEL=APL_UP2_NOCODEC ~/sof-test/test-case/check-playback.sh -d 1 -l 100 -r 1``` and ipc error when PCM_PARAMS at the 64/100 loop on 2rd pipeline
OR
```TPLG=/lib/firmware/intel/sof-tplg/sof-apl-nocodec.tplg MODEL=APL_UP2_NOCODEC ~/sof-test/test-case/check-playback.sh -d 1 -l 200 -r 1```and ipc error when PCM_PARAMS at the 164/200 loop on 1st pipeline

**Reproduction Rate**
100%

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: [4f67e3639219](https://github.com/thesofproject/linux/commit/4f67e3639219)
SOF Branch: main
SOF Commit: [2798398f5c89](https://github.com/thesofproject/sof/commit/2798398f5c89)
 Topology: sof-apl-nocodec.tplg
 Platform: APL UP2 NOCODEC

**Screenshots or console output**
[dmesg]
```
[  946.056051] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x30100000: GLB_TPLG_MSG: PIPE_NEW
[  946.056418] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: widget PIPELINE.11.SSP5.OUT setup complete
[  946.056428] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[  946.056585] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: widget PCM3P setup complete
[  946.056592] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x30200000: GLB_TPLG_MSG: BUFFER_NEW
[  946.057596] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: widget BUF11.0 setup complete
[  946.057612] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[  946.057743] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  946.057867] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: widget PGA11.0 setup complete
[  946.057874] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x30200000: GLB_TPLG_MSG: BUFFER_NEW
[  946.057982] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: widget BUF11.1 setup complete
[  946.057990] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x30100000: GLB_TPLG_MSG: PIPE_NEW
[  946.058295] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: widget PIPELINE.5.SSP5.OUT setup complete
[  946.058303] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[  946.058552] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: widget MIXER5.0 setup complete
[  946.058560] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x30200000: GLB_TPLG_MSG: BUFFER_NEW
[  946.058696] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: widget BUF5.0 setup complete
[  946.058705] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[  946.058801] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  946.058883] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: widget PGA5.0 setup complete
[  946.058889] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x30200000: GLB_TPLG_MSG: BUFFER_NEW
[  946.058965] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: widget BUF5.1 setup complete
[  946.058973] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[  946.059136] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[  946.059262] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: widget SSP5.OUT setup complete
[  946.059268] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: setting up route PCM3P -> BUF11.0
[  946.059272] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
[  946.059333] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: setting up route BUF11.0 -> PGA11.0
[  946.059337] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
[  946.059395] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: setting up route PGA11.0 -> BUF11.1
[  946.059399] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
[  946.059460] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: setting up route BUF11.1 -> MIXER5.0
[  946.059464] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
[  946.059616] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: setting up route MIXER5.0 -> BUF5.0
[  946.059622] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
[  946.059684] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: setting up route BUF5.0 -> PGA5.0
[  946.059688] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
[  946.059746] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: setting up route PGA5.0 -> BUF5.1
[  946.059750] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
[  946.059808] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: setting up route BUF5.1 -> SSP5.OUT
[  946.059812] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
[  946.059870] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: tplg: complete pipeline PIPELINE.11.SSP5.OUT id 55
[  946.059876] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x30130000: GLB_TPLG_MSG: PIPE_COMPLETE
[  946.060135] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: tplg: complete pipeline PIPELINE.5.SSP5.OUT id 27
[  946.060140] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x30130000: GLB_TPLG_MSG: PIPE_COMPLETE
[  946.060207] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: stream_tag 1
[  946.060214] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[  946.060844] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx error for 0x60010000 (msg/reply size: 108/20): -12
[  946.060850] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: HW params ipc failed for stream 1
[  946.060852] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:0e.0: -12
[  946.060860] kernel:  Port5 Deep Buffer: ASoC: __soc_pcm_hw_params() failed (-12)
[  946.060875] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[  946.061016] kernel:  Port5 Deep Buffer: ASoC: dpcm_fe_dai_hw_params failed (-12)
[  946.061026] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: pcm: free stream 3 dir 0
[  946.061084] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x30020000: GLB_TPLG_MSG: COMP_FREE
[  946.061234] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: widget PCM3P freed
[  946.061241] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x30210000: GLB_TPLG_MSG: BUFFER_FREE
[  946.061318] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: widget BUF11.0 freed
[  946.061323] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x30020000: GLB_TPLG_MSG: COMP_FREE
[  946.061398] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: widget PGA11.0 freed
[  946.061403] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x30210000: GLB_TPLG_MSG: BUFFER_FREE
[  946.061480] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x30110000: GLB_TPLG_MSG: PIPE_FREE
[  946.061789] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: widget PIPELINE.11.SSP5.OUT freed
[  946.061793] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: widget BUF11.1 freed
[  946.061797] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x30020000: GLB_TPLG_MSG: COMP_FREE
[  946.061864] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: widget MIXER5.0 freed
```

XiaoyunWu6666, 2022-06-13:
Could it be a memory issue ? The number of loops that cause such error is quite certain , the 164th loop of playback

XiaoyunWu6666, 2022-06-13:
under bisect

lgirdwood, 2022-06-13:
@ranj063 fyi - looks like a race in module API/volume somewhere.

XiaoyunWu6666, 2022-06-17:
hi @ranj063 Have you seen any suspicious fault in this commit?
I can provide a double confirm if necessary.

ranj063, 2022-06-20:
@keqiaozhang can I please have a link to the SOF logger output? dmesg shows an IPC tx error

keqiaozhang, 2022-06-20:
@ranj063 , please refer to https://sof-ci.sh.intel.com/#/result/planresultdetail/13434?model=APL_UP2_NOCODEC&testcase=check-playback-100times

lgirdwood, 2022-06-20:
Not a P1 blocker as FW still operational (i.e. no crash) and is intermittent.

plbossart, 2022-06-21:
it'd be a blocker if pm_runtime is disabled on the host side @lgirdwood 

ipc errors are fatal for now.

ranj063, 2022-06-21:
@keqiaozhang could you please check if https://github.com/thesofproject/sof/issues/5909 fixes the issue?

fredoh9, 2022-06-21:
@ranj063 you mean, https://github.com/thesofproject/sof/pull/5943?

fredoh9, 2022-06-21:
I tested with daily build(without the fix) first, failed at 64/100. I'm using this comamnd,

`TPLG=/lib/firmware/intel/sof-tplg/sof-apl-nocodec.tplg MODEL=APL_UP2_NOCODEC ~/sof-test/test-case/check-playback.sh -d 1 -l 100 -r 1`

With https://github.com/thesofproject/sof/pull/5943, I am able to run the command twice without any problem. Looks good.

XiaoyunWu6666, 2022-06-23:
Nice to see a fix merged yesterday!
The error did not happen in today's daily test , will keep monitoring for a few days to see whether we can close this issue :)

XiaoyunWu6666, 2022-06-24:
did not happen in daily tests of June 23 24

XiaoyunWu6666, 2022-06-24:
close with #5943 and a couple of good results in daily tests

closed 2022-06-24 by XiaoyunWu6666