thesofproject/sof #5911: [BUG] DSP panic when setting volume after checking kmod load-unload after playback on BDW_WSB_RT286

miRoox, 2022-06-13:
**Describe the bug**
The CI (ID: 13257) observed a DSP panic on BDW_WSB_RT286 when it tests volume after checking kmod load-unload after playback.

**To Reproduce**

```
TPLG=/lib/firmware/intel/sof-tplg/sof-bdw-rt286.tplg MODEL=BDW_WSB_RT286 ~/sof-test/test-case/check-kmod-load-unload-after-playback.sh -l 5
TPLG=/lib/firmware/intel/sof-tplg/sof-bdw-rt286.tplg MODEL=BDW_WSB_RT286 ~/sof-test/test-case/volume-basic-test.sh -l 50
```

**Reproduction Rate**
Almost 100%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: https://github.com/thesofproject/linux/commit/4f67e3639219
    * SOF: https://github.com/thesofproject/sof/commit/2798398f5c89
2) Name of the topology file
    * Topology: sof-bdw-rt286.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: BDW_WSB_RT286

**Screenshots or console output**

**dmesg**

```
[ 1874.218509] kernel: sof-audio-acpi-intel-bdw INT3438:00: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE
[ 1874.218622] kernel: sof-audio-acpi-intel-bdw INT3438:00: ------------[ DSP dump start ]------------
[ 1874.218627] kernel: sof-audio-acpi-intel-bdw INT3438:00: DSP panic!
[ 1874.218630] kernel: sof-audio-acpi-intel-bdw INT3438:00: fw_state: SOF_FW_BOOT_COMPLETE (6)
[ 1874.218699] kernel: sof-audio-acpi-intel-bdw INT3438:00: reason: runtime exception (0x6)
[ 1874.218703] kernel: sof-audio-acpi-intel-bdw INT3438:00: trace point: 0x00000800
[ 1874.218706] kernel: sof-audio-acpi-intel-bdw INT3438:00: panic at :0
[ 1874.218708] kernel: sof-audio-acpi-intel-bdw INT3438:00: error: DSP Firmware Oops
[ 1874.218710] kernel: sof-audio-acpi-intel-bdw INT3438:00: error: Exception Cause: IllegalInstructionCause, Illegal instruction
[ 1874.218714] kernel: sof-audio-acpi-intel-bdw INT3438:00: EXCCAUSE 0x00000000 EXCVADDR 0x00000000 PS       0x00060325 SAR     0x00000000
[ 1874.218718] kernel: sof-audio-acpi-intel-bdw INT3438:00: EPC1     0x00000e4b EPC2     0x00000000 EPC3     0x0001aeaa EPC4    0x00000000
[ 1874.218723] kernel: sof-audio-acpi-intel-bdw INT3438:00: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[ 1874.218726] kernel: sof-audio-acpi-intel-bdw INT3438:00: EPS2     0x00000000 EPS3     0x00060120 EPS4     0x00000000 EPS5    0x00000000
[ 1874.218730] kernel: sof-audio-acpi-intel-bdw INT3438:00: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x00002040
[ 1874.218734] kernel: sof-audio-acpi-intel-bdw INT3438:00: stack dump from 0x0041f7c0
[ 1874.218739] kernel: sof-audio-acpi-intel-bdw INT3438:00: 0x0041f7c0: 00000000 ffffffd8 0041f7e0 0041f500
[ 1874.218744] kernel: sof-audio-acpi-intel-bdw INT3438:00: 0x0041f7d0: c002071c 0041f810 0041f840 00000000
[ 1874.218748] kernel: sof-audio-acpi-intel-bdw INT3438:00: 0x0041f7e0: 0041f7c0 0041f820 0041f850 0000001c
[ 1874.218752] kernel: sof-audio-acpi-intel-bdw INT3438:00: 0x0041f7f0: 0041f810 0041a000 0041a000 00000001
[ 1874.218756] kernel: sof-audio-acpi-intel-bdw INT3438:00: 0x0041f800: 40022cbd 0041f840 00000001 00000002
[ 1874.218760] kernel: sof-audio-acpi-intel-bdw INT3438:00: 0x0041f810: 00000e4b 0041f840 00000000 00000008
[ 1874.218764] kernel: sof-audio-acpi-intel-bdw INT3438:00: 0x0041f820: 0040d4bc 00000002 0040d4ac 00417780
[ 1874.218769] kernel: sof-audio-acpi-intel-bdw INT3438:00: 0x0041f830: 8001611c 0041fa40 00000001 c0022cbd
[ 1874.218773] kernel: sof-audio-acpi-intel-bdw INT3438:00: error: ipc host -> DSP: pending no complete no raw 0x00000800
[ 1874.218777] kernel: sof-audio-acpi-intel-bdw INT3438:00: error: mask host: pending yes complete no raw 0x7fff0002
[ 1874.218781] kernel: sof-audio-acpi-intel-bdw INT3438:00: error: ipc DSP -> host: pending yes complete no raw 0x8dead006
[ 1874.218785] kernel: sof-audio-acpi-intel-bdw INT3438:00: error: mask DSP: pending yes complete no raw 0x7f9e0002
[ 1874.218788] kernel: sof-audio-acpi-intel-bdw INT3438:00: ------------[ DSP dump end ]------------
```

[dmesg.txt](https://github.com/thesofproject/sof/files/8887621/full_dmesg.txt)
[slogger.txt](https://github.com/thesofproject/sof/files/8887625/full_slogger.txt)

miRoox, 2022-06-13:
After bisecting, this regression is introduced by f7663de1f41308b80bac4618b5835fcda1ff8fac.

plbossart, 2022-06-13:
Likely that the legacy config was not added for Broadwell @ranj063  ?

miRoox, 2022-06-14:
> everything works well until the kernel module in unloaded and reloaded ?

The DSP panic happened when setting the volume with playback, all the previous steps (check playback, kernel module unload-reload) worked well.

> Can you try and reproduce on TGL or ADL ?

I can't reproduce this issue with the same steps on TGL or ADL.

> Can you also attach the map file for this build or dump the FW stack.

The map file is attached under the dmesg.

lgirdwood, 2022-06-14:
```
                0x0001ae80                platform_interrupt_init
                0x0001ae88                platform_interrupt_set
                0x0001ae94                platform_interrupt_clear   <<<<<<< Level3 handler is here (but looks OK).
                0x0001aebc                platform_interrupt_get_enabled
                0x0001aed0                interrupt_mask
                0x0001af20                interrupt_unmask
```

and it looks like it's using module adapter ?

```
[     1148482.871030] (       15532.395508) c0 pga          1.2   ......./module_adapter.c:555  INFO module_adapter_copy(): start
[     1149616.516818] (        1133.645752) c0 pga          1.3   ......./module_adapter.c:555  INFO module_adapter_copy(): start
[     1165150.578701] (       15534.061523) c0 pga          1.2   ......./module_adapter.c:555  INFO module_adapter_copy(): start
[     1166283.807823] (        1133.229126) c0 pga          1.3   ......./module_adapter.c:555  INFO module_adapter_copy(): start
[     1181816.203039] (       15532.395508) c0 pga          1.2   ......./module_adapter.c:555  INFO module_adapter_copy(): start
[     1182949.275910] (        1133.072876) c0 pga          1.3   ......./module_adapter.c:555  INFO module_adapter_copy(): start
[     1198483.910710] (       15534.634766) c0 pga          1.2   ......./module_adapter.c:555  INFO module_adapter_copy(): start
[     1199617.243998] (        1133.333252) c0 pga          1.3   ......./module_adapter.c:555  INFO module_adapter_copy(): start
[     1215149.535048] (       15532.291016) c0 pga          1.2   ......./module_adapter.c:555  INFO module_adapter_copy(): start
```

Afaict, it looks like we are doing a volume ramp? We have a crash at IRQ level 5 (which only has timer 3)

ranj063, 2022-06-15:
@miRoox can you please check if this helps: https://github.com/thesofproject/sof/pull/5927

miRoox, 2022-06-16:
Thanks. I checked it multiple times and never see it again with #5927.