thesofproject/sof #5921: [BUG] alsabat test failed on ADLP_RVP_NOCODEC_ZEPHYR

keqiaozhang, 2022-06-15:
**Describe the bug**
CI detected alsabat test failed on ADLP_RVP_NOCODEC_ZEPHYR in this week. After some debugging, this issue only happens with Zephyr firmware and the repro rate is high ~50%.  No reproductions with xtos firmware. 
It seems that it's a Zephyr regression, will try to bisect next.

**To Reproduce**
 ~/sof-test/test-case/check-alsabat.sh -p hw:sofnocodec,0 -c hw:sofnocodec,0 -C 2

**Detail Info**
Kernel Branch: topic/sof-dev
Kernel Commit: 
[4f67e3639219](https://github.com/thesofproject/linux/commit/4f67e3639219)

SOF Branch: main
SOF Commit: 
[2798398f5c89](https://github.com/thesofproject/sof/commit/2798398f5c89)

Zephyr Commit: 
[b9e833366f13](https://github.com/zephyrproject-rtos/zephyr/commit/b9e833366f13)

[image]
 
**Screenshots or console output**
```
Entering capture thread (ALSA).
Get period size: 2048  buffer size: 16384
Recording ...
Capture completed.

BAT analysis: signal has 65536 frames at 48000 Hz, 2 channels, 2 bytes per sample.

Channel 1 - Checking for target frequency 997.00 Hz
Amplitude: 28266.5; Percentage: [86]
Detected peak at 989.50 Hz of 21.36 dB
 Total 21.4 dB from 989.50 to 989.50 Hz
Detected peak at 996.83 Hz of 38.42 dB
 Total 42.1 dB from 990.97 to 1004.15 Hz
 PASS: Peak detected at target frequency
Detected at least 2 signal(s) in total

Return value is -1003
```
[bat.wav.tar.gz](https://github.com/thesofproject/sof-test/files/8895691/bat.wav.tar.gz)

marc-hb, 2022-06-15:
Transferred to the SOF repo.

> Strangely enough, this issue only happens with CI built firmware, I cannot reproduce this issue with the firmware built by myself. 

Can you make sure BOTH the SOF versions and Zephyr versions are the same? They both appear in sof-logger output.

> I also checked with beyond compare, there're a lot of differences between these 2 firmware binaries.

Starting with TGL the rimage signature is randomized with a salt; so you cannot compare two `.ri` files directly.
You cannot compare `.elf` or `.o` files either because we compile with `-g`.
This is why I added the `--erase_vars` feature to `sof/tools/sof_ri_info/sof_ri_info.py`, it fills the variable parts with zeroes. Compare `.ri` binaries only AFTER running `sof/tools/sof_ri_info/sof_ri_info.py`.

For the record linking to this other strange binary difference:
- #5917

keqiaozhang, 2022-06-16:
> @keqiaozhang what happens if you don't use the debug option and just : "./scripts/xtensa-build-zephyr.py tgl"

There is no difference w/o `-d`. This issue is much more complicated than I though. I spent more than 5 hours debugging it, but I got lost in bisecting.  The current conclusion is that this issue has nothing to do with the kernel, but it is still uncertain whether it is caused by FW or Zephyr. This issue has been existed much longer than I expected but with low reproduction rate. I will continue to debug this issue.

lgirdwood, 2022-06-16:
@keqiaozhang Zephyr native drivers ? Looks like a performance issue that we need to track as waveform is good, the FFT failure is caused by a glitch.

lgirdwood, 2022-06-16:
@marc-hb please dont set bug priorities. I or @mengdonglin will set and track Intel issues. Thanks.

marc-hb, 2022-06-16:
I won't override any bug priority set by you, @mengdonglin and others but in this case there was no priority set at all and this is making many PRs fail (when almost no other bug does anymore).

BTW it would be great to have some basic triage guidelines written down.

keqiaozhang, 2022-06-17:
I have done the bisection and the bad commit points to: 

https://github.com/zephyrproject-rtos/zephyr/commit/d660492914ec5524d94536604d0a2dbb95b29c9f

```
commit d660492914ec5524d94536604d0a2dbb95b29c9f
Author: Krzysztof Chruscinski [krzysztof.chruscinski@nordicsemi.no](mailto:krzysztof.chruscinski@nordicsemi.no)
Date:   Mon Sep 27 11:46:30 2021 +0200

    logging: Add option for prolonged backend initialization

    Extended logging backend API with log_backend_is_ready call which
    returns 0 is backend is ready. Logging core will make sure that
    all autostarted backends are ready before they are enabled.

    This option allows to handle backends which are not yet ready
    after init function is called (e.g. usb backend that is not plugged
    in). If this is the only backend in the system, logging processing
    will not start util first backend is ready.

    Function for checking readiness is optional and when backend has
    no such function it is assumed that backend is ready after
    initialization function returns which makes this feature backward
    compatible.

    Signed-off-by: Krzysztof Chruscinski [krzysztof.chruscinski@nordicsemi.no](mailto:krzysztof.chruscinski@nordicsemi.no)

 include/zephyr/logging/log_backend.h | 40 ++++++++++++++++++
 subsys/logging/log_core.c            | 78 +++++++++++++++++++++++++++++++-----
 2 files changed, 107 insertions(+), 11 deletions(-)
```
I also did below tests to make sure the result is reliable:
1.	Tested the previous commit d84989a and 1K iterations passed.
 
[image]

2.	Tested with the bad commit again, issue was still easy to reproduce. 
3.	Reverted the bad commits form tip zephyr, then issue disappeared(300 iterations passed). 
Since there’re some dependencies, so I have to revert 3 commits:
•	git revert c5f2cdef098b9aba5a67aab0127ccfb6f85f17d5
•	git revert ad12e4be32b55fb22f6ec353e67b8039319e1a2d
•	git revert d660492914ec5524d94536604d0a2dbb95b29c9f

marc-hb, 2022-06-17:
Impressive work @keqiaozhang 

@nordic-krch any insight?

lgirdwood, 2022-06-17:
@nashif any ideas why above Zephyr commits could cause some audio glitches (could be scheduling ?) as my understanding is that above bisect commit would only impact the log startup ? or would this impact every log event or log flush ?

kv2019i, 2022-06-17:
Manage to reproduce this on a TGL system now. Our ll timer logs reveal something interesting:

```
[     3102124.713542] (      986313.062500) c0 zll-schedule       ......../zephyr_domain.c:90   INFO ll timer avg 243, max 2271, overruns 0
[     3152966.848958] (          90.703125) c2 zll-schedule       ......../zephyr_domain.c:90   INFO ll timer avg 6324, max 10351, overruns 0
[     4127743.411458] (      974776.562500) c0 zll-schedule       ......../zephyr_domain.c:90   INFO ll timer avg 226, max 2061, overruns 0
[     4177964.479167] (          89.921875) c2 zll-schedule       ......../zephyr_domain.c:90   INFO ll timer avg 8185, max 9194, overruns 0
[     5154144.479167] (      148330.359375) c0 zll-schedule       ......../zephyr_domain.c:90   INFO ll timer avg 222, max 1196, overruns 0
[     5204687.994792] (          12.083333) c2 zll-schedule       ......../zephyr_domain.c:90   INFO ll timer avg 7598, max 22326, overruns 0
[     6193564.583333] (      851867.875000) c0 zll-schedule       ......../zephyr_domain.c:90   INFO ll timer avg 226, max 1090, overruns 0
[     6244427.552083] (          11.953125) c2 zll-schedule       ......../zephyr_domain.c:90   INFO ll timer avg 5209, max 6438, overruns 0
[     7232441.640625] (       28044.843750) c0 zll-schedule       ......../zephyr_domain.c:90   INFO ll timer avg 228, max 963, overruns 0
[     8276821.484375] (     1031122.500000) c0 zll-schedule       ......../zephyr_domain.c:90   INFO ll timer avg 223, max 680, overruns 0
[     9321221.588542] (     1031142.250000) c0 zll-schedule       ......../zephyr_domain.c:90   INFO ll timer avg 223, max 667, overruns 0
```

So when the test fails, we can observe two very large spikes in pipeline execution time (22326, that 3x the average). This is still within DSP capacity and we observe no overruns, so not immediately clear why we are seeing glitches.

lgirdwood, 2022-06-20:
> So when the test fails, we can observe two very large spikes in pipeline execution time (22326, that 3x the average). This is still within DSP capacity and we observe no overruns, so not immediately clear why we are seeing glitches.

@kv2019i looks like we have a time difference between glitch and previous corresponding sample of 15 samples. This aligns with FIFO depth. Do we see a corresponding extra time for the dma_copy(), in particular any delay around clearing LLI DONE bits ?

I am suspicious we may see the glitches when we have a FIFO handshake that is at the end of a DMA LLI copy. i.e. the DMA has to reload and copy (and we miss). This sould show as multiples of 1ms between glitches.

kv2019i, 2022-06-20:
Thanks @singalsu . The glitches are indeed present. Curiously, there is only one unexpected sample, surrounded by correct samples, and even that one incorrect sample has a value that is very close to expected value. Following is a dump of individual sample values in a a sequence where spectrum reveal glitches. The test tone is 997Hz sine samples at 48kHz sampling. Values in float 32bit representation:

```
0.23419	
0.11240		
-0.01129
-0.20526  # glitch, expected sample value ~ -0.13500
-0.25604
```

This is quite odd as the lookback is digital with a shared clock. The error also seems to be large enough that this cannot be a case where MSB byte is correct but LSB byte of the sample is read incorrectly (16bit samples at DAI). 

This could still be explained by the FIFO running out, so dma_copy() timing needs to be looked at.

More today's observations:
 - I've seen glitches in the FFT plot even when alsabat reports success (built with Zephyr) -- so the failure may be even more frequent
 - if I disable logging in Zephyr, I've so far been not able to reproduce the problem (confirming @keqiaozhang 's findings)
 - I have not seen the errors in any config with XTOS

kv2019i, 2022-06-21:
Quick update:
 - with a longer test sequence, errors can still be seen even with logging disabled and I can see also multi sample errors
 - I could see glitches also with older Zephyr 
 - it seems there is a connection to https://github.com/zephyrproject-rtos/zephyr/issues/46378 -- with a workaround to that bug, I see clean long test sequences with recent Zephyr and logging enabled

lgirdwood, 2022-06-21:
@lenghonglin fyi - could be similar to your issue.

lgirdwood, 2022-06-21:
> * with a longer test sequence, errors can still be seen even with logging disabled and I can see also multi sample errors

so, logger affects the reproduction rate. logger does NOT share the same DMAC, but may be a heavy uncache user ?

kv2019i, 2022-06-21:
It seems logging is one symptom and not the rootcause. One new finding is that this specifically happens with PCM0 where the playback path goes via shared mixer path. If alsabat is run on PCM2 (mixer with only one input), I have not been able to reproduce the problem. When the problem triggers and captured audio has corruption, following is seen at timer tick:

```
[     1024143.932292] (         982.031250) c3 zll-schedule       ......../zephyr_domain.c:126  INFO domain tick avg 38400, max 39073
[     2063513.203125] (         982.968750) c3 zll-schedule       ......../zephyr_domain.c:126  INFO domain tick avg 38976, max 39656
[     3101570.104167] (         961.953125) c3 zll-schedule       ......../zephyr_domain.c:126  INFO domain tick avg 38927, max 39672
[     4141570.104167] (         962.005188) c3 zll-schedule       ......../zephyr_domain.c:126  INFO domain tick avg 39000, max 39687
[     5181291.067708] (         982.083313) c3 zll-schedule       ......../zephyr_domain.c:126  INFO domain tick avg 38989, max 39682
```

I tested this with same methodology we used for https://github.com/zephyrproject-rtos/zephyr/issues/43964 .

When the recording is clean, the domain timer stats look like:

```
[    30720141.770833] (         962.187500) c3 zll-schedule       ......../zephyr_domain.c:126  INFO domain tick avg 38399, max 38521
[    31744141.666667] (         961.953125) c3 zll-schedule       ......../zephyr_domain.c:126  INFO domain tick avg 38399, max 38589
[    32768142.109375] (         979.583313) c3 zll-schedule       ......../zephyr_domain.c:126  INFO domain tick avg 38400, max 38713
[    33792141.927083] (         979.713562) c3 zll-schedule       ......../zephyr_domain.c:126  INFO domain tick avg 38399, max 38652
```

There seems to be a connection to https://github.com/zephyrproject-rtos/zephyr/issues/46378 and I submitted https://github.com/zephyrproject-rtos/zephyr/pull/46744 that should solve the issue for SOF builds for now. The rootcause should still be found (in context of above bug).

kv2019i, 2022-06-23:
@keqiaozhang With https://github.com/zephyrproject-rtos/zephyr/pull/46744 merged to upstream, I expect the problem to no longer occur in SOF CI tests. Let's monitor this next week. If the fix holds, we can close this bug and continue follow up in the Zephyr bug 46378 .

kv2019i, 2022-06-23:
First Intel daily test clean (13499), using Zephyr upstream commit 13499 (with the fix included). Will continue monitoring and keeping the bug open until more data is available.

lenghonglin, 2022-06-28:
> @lenghonglin fyi - could be similar to your issue.

nope , i will describe more detail in https://github.com/thesofproject/sof/pull/5912

keqiaozhang, 2022-07-04:
Confirmed that this issue is gone in CI. Closing it.

closed 2022-07-04 by keqiaozhang