thesofproject/sof #5929: [BUG]

WalterSmuts, 2022-06-16:
**Describe the bug**
Sound stops working when I wake from hibernation.

**To Reproduce**
Hibernate and wake.

**Reproduction Rate**
100%.

**Expected behavior**
Sound works after hibernation as before.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Environment**
Not sure about all these options, but here's my setup:
```
Linux cuddles 5.18.3-arch1-1 #1 SMP PREEMPT_DYNAMIC Thu, 09 Jun 2022 16:14:10 +0000 x86_64 GNU/Linux
```
It's using the most up to date arch sof drivers found [here](https://archlinux.org/packages/extra/x86_64/sof-firmware/).

**Screenshots or console output**
```
[24668.759477] sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[24668.759482] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware boot failure due to timeout
[24668.759484] sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state: SOF_FW_BOOT_IN_PROGRESS (2)
[24668.759655] sof-audio-pci-intel-tgl 0000:00:1f.3: invalid header size 0xffffffff. FW oops is bogus
[24668.759658] sof-audio-pci-intel-tgl 0000:00:1f.3: unexpected fault 0x00000000 trace 0x00000000
[24668.759661] sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
[24668.759662] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to boot DSP firmware after resume -5
[24668.759664] sof-audio-pci-intel-tgl 0000:00:1f.3: PM: dpm_run_callback(): pci_pm_restore+0x0/0xe0 returns -5
[24668.759683] sof-audio-pci-intel-tgl 0000:00:1f.3: PM: failed to restore async: error -5
```

Not sure about sof-logger but [here are the dmesg logs](https://pastebin.com/kX3xms5V).

plbossart, 2022-06-16:
Known issue https://github.com/thesofproject/sof/issues/5892

plbossart, 2022-06-16:
Fixed by https://github.com/thesofproject/linux/pull/3687

lgirdwood, 2022-06-16:
Closing since fix is merged.

closed 2022-06-16 by lgirdwood