thesofproject/sof #5965: [BUG] Zephyr ignores the topology setting for pipeline scheduling period

ranj063, 2022-06-29:
**Describe the bug**
Pipeline tasks are always scheduled every ms irrespective of what the topology requests.

**To Reproduce**
Use https://github.com/thesofproject/sof/pull/5958 for setting the pipeline period in topology

**Reproduction Rate**
100%

**Expected behavior**
Pipelines must be scheduled based on the period set in topology

**Impact**
We cannot enable multi-stream feature with pipelines with variable period sizes

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: [topic/sof-dev](https://github.com/thesofproject/linux/commit/ad489ae9054724c9b61e020fa8fefd84381a9490)
    * SOF: Use SOF PR: https://github.com/thesofproject/sof/pull/5958
2) Name of the topology file
    * Topology: sof-tgl-nocodec.tplg with PR: https://github.com/thesofproject/sof/pull/5958
3) Name of the platform(s) on which the bug is observed.
    * Platform: TGL, ADL

ranj063, 2022-06-29:
@lyakh @kv2019i  @lgirdwood FYI

kv2019i, 2022-06-30:
I don't think this blocks larger period multiple streams -> https://github.com/thesofproject/sof/pull/5958#discussion_r910692671 
Still a missing bit in the scheduler.

hongshui3000, 2022-06-30:
I think this PR should also be related to this issue:https://github.com/thesofproject/sof/pull/5912

lgirdwood, 2022-07-04:
@lyakh can you take care of this ?

plbossart, 2022-12-13:
Is this issue still relevant @lyakh @ranj063 ? Or do we need to wait for DP processing to be available before the deep-buffer is used?

ranj063, 2022-12-13:
This issue is only relevant with IPC3+zephyr+topology1. Not sure if this will ever be fixed though

lyakh, 2022-12-13:
> This issue is only relevant with IPC3+zephyr+topology1. Not sure if this will ever be fixed though

@ranj063 why is this irrelevant for IPC4?

ranj063, 2022-12-13:
> @ranj063 why is this irrelevant for IPC4? IPC4 doesn't provide a scheduling period?

@lyakh the scheduling period is relevant of course, its just not set in topology as we do with IPC3+topology1. Im not sure how this should be done for IPC4

lgirdwood, 2022-12-14:
@mwasko how is LL scheduling period configured via IPC4 ?

pblaszko, 2022-12-16:
> @mwasko how is LL scheduling period configured via IPC4 ?

In closed source FW (IPC4), scheduler analyze DMA buffer sizes on stream endpoints, and calculates global scheduling period itself, based on lowest active buffer size. When scheduling period > 1ms, scheduler use flexible scheduling to aggregate 1ms processing cycles together and go idle for the rest of scheduling period.
When enabling multiple streams, FW use scheduling period required to service stream with lowest buffers.
So driver does not have option to directly configure scheduling period, but may do this indirectly, by configuring appropriate DMA buffer sizes.
As far as I know Converged FW does not support flexible scheduling. So it will schedule with 1ms period anyway.

There is also an option for local scheduling cycles for DP tasks. Then pipeline is scheduled with global scheduling period, but DP modules are executed only when enough data is available in input buffer. It is configurable by size of DP module IBS/OBS.
This option is available only for DP modules. LL modules must have buffers maximum 1ms.

lgirdwood, 2022-12-19:
ok, IIUC - for IPC4 the LL period is determine by the DMA buffer size in conjunction with the rate and frame size. I think we need to add an overide value in the topology for this (i.e. if 0 ignore) so that DMA configuration can be more loosely coupled to the scheduling period e.g. in the case where we need larger DMA buffers to deal with HW issues.

kv2019i, 2023-03-06:
@lyakh Still on track for V2.5 -- we aim to branch this week?

kv2019i, 2023-03-09:
No response, moving to v2.6.

lgirdwood, 2023-05-10:
Should be addressed by DP. Move to v2.7

closed 2025-07-25 by github-actions[bot]