thesofproject/sof #5989: [BUG] microphone detected but not working on 2022 Razer Blade 15

ewfuentes, 2022-07-06:
**Describe the bug**
I've installed Ubuntu 22.04 on a 2022 Razer Blade 15. Audio playback appears to be working (apart from L/R audio channels being flipped), but the built in microphone, while detected, doesn't seem to be recording any sound. This was tested by going to the ubuntu sound settings, making some noise and not seeing any activity from the meter. 

[image]

I'm unfamiliar with linux sound, but I'm willing to do some digging in. Thank you for your time!

**To Reproduce**
Clean install of Ubuntu 22.04. I have updated the kernel to 5.17.3 to get wifi support.

**Reproduction Rate**
100%

**Expected behavior**
I expect the microphone to record when enabled.

**Impact**
I can't use the built in microphone for video conferencing. I haven't tried wired/bluetooth or USB microphones as a workaround just yet.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: [f6a218ea3564d2c2e53199577929e02e173a499c](https://kernel.ubuntu.com/~kernel-ppa/mainline/v5.17.3/)
    * SOF: I'm not quite sure. I've installed [firmware-sof-signed 2.0-1ubuntu3](https://packages.ubuntu.com/jammy-updates/firmware-sof-signed)
2) Name of the topology file
    * Topology: I'm not sure how to determine this... I've included a link to the alsa-info artifact, but nothing sticks out to me as having this information
3) Name of the platform(s) on which the bug is observed.
    * Platform: x86_64

**Screenshots or console output**

http://alsa-project.org/db/?f=625e3480bfdd42663c730e65229c8619950e749c

plbossart, 2022-07-06:
There's no smoking gun @ewfuentes, the only part where we have a known issue is the case of 2 DMICs connected to PDM1, instead of the usual PDM0.

Can you please add this to /etc/modprobe.d/alsa-base.conf

````
options snd_sof_intel_hda_common dmic_num=4
````
reboot and run this command

````
arecord -l
````
check that the DMIC device is  card 0, device 6, then

````
arecord -Dhw:0,6 -c4 -r48000 -sS32_LE -d 10 test.wav
````

Try to record your voice during this 10s test, and then look at the test.wav file with audacity.

If you see any signs of life on 2 channels, we have a mic configuration issue. If you see nothing, then something else is at play.

Hope this helps.

ewfuentes, 2022-07-06:
After adding the option to alsa-base.conf, `arecord -l` shows the microphone as being on card 1 (instead of 0) device 6.
```
**** List of CAPTURE Hardware Devices ****
card 1: sofhdadsp [sof-hda-dsp], device 0: HDA Analog (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: sofhdadsp [sof-hda-dsp], device 1: HDA Digital (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: sofhdadsp [sof-hda-dsp], device 6: DMIC (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: sofhdadsp [sof-hda-dsp], device 7: DMIC16kHz (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
  ```
  
I used the following command to record myself (note the device and `-f` instead of `-s`):
```
arecord -Dhw:1,6 -c4 -r48000 -fS32_LE -d 10 test.wav
```

Looking at the wav file in audacity shows some promise! There are 4 channels and channels 3 and 4 contain me speaking. Channels 1 and 2 have some audio artifact , but are otherwise silent.

[image]

For what it's worth, the audio meter in the sound settings and google meet don't appear to recognize any audio.

plbossart, 2022-07-06:
excellent @ewfuentes 

Can you do me a favor and retest with this file
[sof-hda-generic-2ch-pdm1.tplg.gz](https://github.com/thesofproject/sof/files/9056487/sof-hda-generic-2ch-pdm1.tplg.gz)

it should be the same functionality as before, just trying to make sure the updated file is correct.

plbossart, 2022-07-06:
copy it over as sof-hda-generic-2ch.tplg

plbossart, 2022-07-06:
Can you try with this update 
[sof-hda-generic-2ch.tplg.gz](https://github.com/thesofproject/sof/files/9056962/sof-hda-generic-2ch.tplg.gz)

I only hard-coded a PDM0 to PDM1 change, if that doesn't work we'll have to ask @singalsu for help.

````diff
diff --git a/tools/topology/topology1/platform/intel/intel-generic-dmic.m4 b/tools/topology/topology1/platform/intel/intel-generic-dmic.m4
index c9808c3d9..5dfef3a5b 100644
--- a/tools/topology/topology1/platform/intel/intel-generic-dmic.m4
+++ b/tools/topology/topology1/platform/intel/intel-generic-dmic.m4
@@ -157,7 +157,7 @@ ifelse(DMIC_DAI_CHANNELS, 4,
 `DAI_CONFIG(DMIC, 0, DMIC_DAI_LINK_48k_ID, DMIC_DAI_LINK_48k_NAME,
            DMIC_CONFIG(1, 2400000, 4800000, 40, 60, 48000,
                 DMIC_WORD_LENGTH(s32le), 200, DMIC, 0,
-                PDM_CONFIG(DMIC, 0, STEREO_PDM0)))')
+                PDM_CONFIG(DMIC, 0, STEREO_PDM1)))')
 
 ifdef(NO16KDMIC, `',
 `ifelse(DMIC16K_DAI_CHANNELS, 4,
@@ -168,4 +168,4 @@ ifdef(NO16KDMIC, `',
 `DAI_CONFIG(DMIC, 1, DMIC_DAI_LINK_16k_ID, DMIC_DAI_LINK_16k_NAME,
            DMIC_CONFIG(1, 2400000, 4800000, 40, 60, 16000,
                 DMIC_WORD_LENGTH(s32le), 400, DMIC, 1,
-                PDM_CONFIG(DMIC, 1, STEREO_PDM0)))')')
+                PDM_CONFIG(DMIC, 1, STEREO_PDM1)))')')

````

There are additional changes in the config that I don't quite get, the old working file has this
DMIC_CONFIG(1, 500000, 4800000, 40, 60, 48000,
and the newer files:
DMIC_CONFIG(1, 2400000, 4800000, 40, 60, 48000,

ewfuentes, 2022-07-06:
The one you just sent me (`dd17ba1a`) does work.

```
erick@neptune:~/Downloads$ sudo md5sum /lib/firmware/intel/sof-tplg/sof-hda-generic-2ch.tplg 
dd17ba1ab213a6b1827334ca83a382a4  /lib/firmware/intel/sof-tplg/sof-hda-generic-2ch.tplg
erick@neptune:~/Downloads$ md5sum *.tplg*
3a0f76a6e276b9d0cb10488566fdcf4b  sof-hda-generic-2ch.tplg.bak <--- originally installed
d3d7af694582142e087452e845817ed2  sof-hda-generic-2ch-pdm1-orig.tplg <--- from lenovo thread
df3fab3ec07c925e391deae8eeb4363f  sof-hda-generic-2ch-pdm1.tplg <--- First one you sent me that didn't work
dd17ba1ab213a6b1827334ca83a382a4  sof-hda-generic-2ch.tplg <--- second one you sent me
```

plbossart, 2022-07-06:
ok, that's good @ewfuentes. I had a mistake in my M4 file, fixed now.

So the short story is that you have a working solution. If you want a long-term solution, we will have this topology file included in official releases (2.3+) and you will be able to use a kernel parameter to select the topology (no file override required). Unfortunately, we have a number of bugs in the kernel override so you will need additional patches from https://github.com/thesofproject/linux/pull/3740.

ewfuentes, 2022-07-06:
Amazing! Thank you so much for your help!

To recap, the steps are:
1. Download [this updated topology](https://github.com/thesofproject/sof/files/9056962/sof-hda-generic-2ch.tplg.gz).
2. unzip it using `gunzip sof-hda-generic-2ch.tplg.gz`. This result in `sof-hda-generic-2ch.tplg` being extracted from the archive.
3. Make a back up copy of the existing topology by running `sudo cp /lib/firmware/intel/sof-tplg/sof-hda-generic-2ch.tplg sof-hda-generic-2ch.tplg.bak`
4. Copy the updated topology in place by running `sudo cp sof-hda-generic-2ch.tplg /lib/firmware/intel/sof-tplg/sof-hda-generic-2ch.tplg`
5. compute md5 checksum of the new file by running `sudo md5sum /lib/firmware/intel/sof-tplg/sof-hda-generic-2ch.tplg` and verify it matches `dd17ba1ab213a6b1827334ca83a382a4`.
6. Reboot.
7. Try to record using audacity to verify that the fix works.

lgirdwood, 2022-07-06:
@plbossart do you want to keep this open until https://github.com/thesofproject/linux/pull/3740 is merged ?

gregory112, 2022-10-05:
@plbossart Apologize to disturb this already closed issue, just want to confirm with Linux 6 (5.20+ that you probably meant), do we still need to specify the topology file as module parameter? Or will everything work with default settings in Linux 6 or sof-firmware or sof-bin 2.3?

plbossart, 2022-10-05:
@gregory112 there is no guarantee that things will work. I suggested an improvement that does work for some people, you'll have to try and report an issue if the result is different than expected.

gregory112, 2022-10-17:
@plbossart I did try with
```
options snd-sof-pci tplg_filename=sof-hda-generic-2ch-pdm1.tplg
```
with the recently released Linux 6.0.1 and I want to confirm that the microphone indeed works. Thank you for the fix. The outgoing sound channel is still reversed though (left channel output in right speaker and vice versa), do you think it is related with this topology file?

wsxarcher, 2022-10-21:
With headphones the channels are not inverted, so I guess it's an amplifier configuration issue as you suggested

plbossart, 2022-10-22:
Please try also with this added to /etc/modprobe.d/alsa-base.conf:

options snd-intel-dspcfg dsp_driver=1

this will force the use of the legacy HDaudio driver. You will lose the ability to use the microphone, but that should confirm the left-right inversion at the amplifier level. Thanks.

wsxarcher, 2022-10-22:
With `options snd-intel-dspcfg dsp_driver=1` I confirm that the speakers are still reversed (the headphones remain correct). I also hear a background noise while playing audio but I guess it's a problem with the legacy driver.

wsxarcher, 2022-10-22:
@gregory112 Never happened but I mostly use external audio

tiwai, 2022-10-24:
Hrm, channel swaps have almost never happened on HD-audio (since HD-audio channels are always stereo, hence most of hardware implementations are in stereo, too).  There is no such quirk, so far.

Does the sound work as expected on Windows?

gregory112, 2022-10-25:
@tiwai Yes it does work as intended in Windows. In Windows it is detected as Realtek Audio Codec with THX Spatial Audio. It has THX Spatial Audio, not sure what this means in Linux though.

@wsxarcher Just FYI, perhaps you never use Spotify or Discord like I do but it does seem that it is outputting high-pitched sound when I open Discord or Spotify or both and the sound vanish when I close both (or one of them? I'm kinda not sure).

tiwai, 2022-10-25:
And the channel swap happens on both the speaker and the headphone outputs?  Or it's only on speakers?

wsxarcher, 2022-10-25:
Only on the speakers

tiwai, 2022-10-25:
So there is something below HD-audio that requires some extra setup to correct the connection.

themarpe, 2022-10-26:
@plbossart 

Thanks for this fix! I've initially tried with Kernel 6.0.3 but the audio didn't work at all (neither speakers nor microphone was detected anymore). Updated to 6.0.5 and applied the `alsa.conf` option, all works well!

mbana, 2022-12-29:
Firstly thanks for the fix.

I'm running a Razer Blade 2022 with 12th Gen Intel.

## Speakers

The speakers work out of the box but they are reversed.

### Microphone

The microphone does not work out of the box. I can confirm the fix detailed on <https://abarry.org/ubuntu-on-razer-blade-15-2022-advanced/> to get the microphone working on the Razer Blade 2022 works.

### Details

I am using Fedora 37.

```sh
$ inxi -A
Audio:
  Device-1: Intel Alder Lake PCH-P High Definition Audio
    driver: sof-audio-pci-intel-tgl
  Device-2: NVIDIA GA106 High Definition Audio driver: snd_hda_intel
  Sound API: ALSA v: k6.0.15-300.fc37.x86_64 running: yes
  Sound Server-1: PipeWire v: 0.3.63 running: yes
```

```sh
$ cat /etc/os-release 
NAME="Fedora Linux"
VERSION="37 (Workstation Edition)"
ID=fedora
VERSION_ID=37
VERSION_CODENAME=""
PLATFORM_ID="platform:f37"
PRETTY_NAME="Fedora Linux 37 (Workstation Edition)"
ANSI_COLOR="0;38;2;60;110;180"
LOGO=fedora-logo-icon
CPE_NAME="cpe:/o:fedoraproject:fedora:37"
DEFAULT_HOSTNAME="fedora"
HOME_URL="https://fedoraproject.org/"
DOCUMENTATION_URL="https://docs.fedoraproject.org/en-US/fedora/f37/system-administrators-guide/"
SUPPORT_URL="https://ask.fedoraproject.org/"
BUG_REPORT_URL="https://bugzilla.redhat.com/"
REDHAT_BUGZILLA_PRODUCT="Fedora"
REDHAT_BUGZILLA_PRODUCT_VERSION=37
REDHAT_SUPPORT_PRODUCT="Fedora"
REDHAT_SUPPORT_PRODUCT_VERSION=37
SUPPORT_END=2023-11-14
VARIANT="Workstation Edition"
VARIANT_ID=workstation
```

```sh
$ dmesg | grep 'sof-audio-pci-intel-tgl'
[    7.760590] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    7.760690] sof-audio-pci-intel-tgl 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    7.760704] sof-audio-pci-intel-tgl 0000:00:1f.3: enabling device (0000 -> 0002)
[    7.760859] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    7.760928] sof-audio-pci-intel-tgl 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    7.767464] sof-audio-pci-intel-tgl 0000:00:1f.3: use msi interrupt mode
[    7.817734] sof-audio-pci-intel-tgl 0000:00:1f.3: hda codecs found, mask 5
[    7.817737] sof-audio-pci-intel-tgl 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[    7.817741] sof-audio-pci-intel-tgl 0000:00:1f.3: DMICs detected in NHLT tables: 2
[    7.836796] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 2:2:0-57864
[    7.836800] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:22:1 Kernel ABI 3:23:0
[    7.836829] sof-audio-pci-intel-tgl 0000:00:1f.3: unknown sof_ext_man header type 3 size 0x30
[    7.929564] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 2:2:0-57864
[    7.929568] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:22:1 Kernel ABI 3:23:0
[    9.691378] sof-audio-pci-intel-tgl 0000:00:1f.3: Topology: ABI 3:22:1 Kernel ABI 3:23:0
```

The line `[    7.836829] sof-audio-pci-intel-tgl 0000:00:1f.3: unknown sof_ext_man header type 3 size 0x30` looks suspicious.

### Misc

If you need further information in fixing this so this just works, please ping me.

mbana, 2023-02-25:
> @tiwai Yes it does work as intended in Windows. In Windows it is detected as Realtek Audio Codec with THX Spatial Audio. It has THX Spatial Audio, not sure what this means in Linux though.

I have the same or similar laptop and on Windows it's reported as "Realtek Audio Codec with THX Spatial Audio" like @gregory112 mentioned.

@tiwai, is t possible to get THX Spatial Audio on Linux somehow?

jheanley, 2023-12-08:
Adding this in case others wonder if their issue may be fixed by this.
Razer Blade 17 12th gen i7 (2022)
I was having exactly same issues
Left/Right output channels work, but are switched. No Mic input (device appears to be registered
New topology file fixed mic issue
`
Audio:
  Device-1: Intel Alder Lake PCH-P High Definition Audio
    driver: sof-audio-pci-intel-tgl
  Device-2: NVIDIA GA104 High Definition Audio driver: snd_hda_intel
  Device-3: DisplayLink Dell Universal Dock D6000 type: USB
    driver: cdc_ncm,snd-usb-audio,usbfs
  Sound Server-1: ALSA v: k6.2.0-37-generic running: yes
  Sound Server-2: PulseAudio v: 15.99.1 running: yes
  Sound Server-3: PipeWire v: 0.3.48 running: yes
`
`
PRETTY_NAME="Ubuntu 22.04.3 LTS"
NAME="Ubuntu"
VERSION_ID="22.04"
VERSION="22.04.3 LTS (Jammy Jellyfish)"
VERSION_CODENAME=jammy
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=jammy
`

gregory112, 2024-10-26:
Until today I still have this issue. Probably many people still do. Not sure if this is a hardware problem, but I think it is probably best to open a separate issue about this.