thesofproject/sof #6007: [BUG]Build topology2 with combination of SSP DAI duplex and playback triggered coredump

yongzhi1, 2022-07-13:
**Describe the bug**

Modified cavs-nocodec to use two SSP port:

SSP0 - duplex
SSP1 - playback

Sample changes https://github.com/yongzhi1/sof/tree/test-ssp1-playback

**To Reproduce**
make topology2_sof-mtl-nocodec
`Generating sof-mtl-nocodec.tplg
find_modes(): fs not set
saving nhlt as binary in nhlt-ace-mtl-nocodec.bin
alsatplg: malloc.c:2379: sysmalloc: Assertion `(old_top == initial_top (av) && old_size == 0) || ((unsigned long) (old_size) >= MINSIZE && prev_inuse (old_top) && ((unsigned long) old_end & (pagesize - 1)) == 0)' failed.
Aborted (core dumped)
make[3]: *** [topology/topology2/cavs/CMakeFiles/topology2_sof-mtl-nocodec.dir/build.make:73: topology/topology2/cavs/sof-mtl-nocodec.tplg] Error 134`

**Expected behavior**
Build return without error.

bardliao, 2022-07-14:
The issue happens if SSP2 is removed.
I can build with [test.diff.txt](https://github.com/thesofproject/sof/files/9109079/test.diff.txt), but I will see build error if I remove SSP2. @ranj063 @juimonen Can you take a look?

lgirdwood, 2022-07-15:
@yongzhi1 are you able to manually invoke the tplg compiler with this source file under valgrind ? This will throw up the errors and show which lines are at fault  (may be easy to fix too).

yongzhi1, 2022-07-20:
> @yongzhi1 are you able to manually invoke the tplg compiler with this source file under valgrind ? This will throw up the errors and show which lines are at fault (may be easy to fix too).

I tried the following cmd line but it returned error, will spend a little more time later:

alsatplg -I /home/cowboy/sof/tools/topology/topology2/cavs -D PLATFORM=mtl,NUM_DMICS=2,PREPROCESS_PLUGINS=nhlt,NHLT_BIN=nhlt-ace-mtl-nocodec.bin -p -c sof-mtl-nocodec.conf -o sof-mtl-nocodec.tplg
ALSA lib conf.c:810:(get_char_skip_comments) Invalid search dir /home/cowboy/local/share/alsa/cavs
ALSA lib conf.c:2007:(_snd_config_load_with_include) _toplevel_:6:16:Invalid argument
Unable not load configuration

yongzhi1, 2022-07-20:
>

juimonen, 2022-07-25:
@yongzhi1 I'm able to build your branch... let's sync offline our alsa-utils versions.

yongzhi1, 2022-07-30:
> @yongzhi1 I'm able to build your branch... let's sync offline our alsa-utils versions.

Thank you for the info, not sure if this made sense https://github.com/thesofproject/alsa-utils/commit/e10ce54c80b441eca11ecb8265d7f42fc0c4e446

lgirdwood, 2022-08-01:
@juimonen any update ?

juimonen, 2022-08-01:
@lgirdwood I'll check still with Yong the issue as I can't reproduce... but we might have different versions of alsa-utils

closed 2022-08-02 by yongzhi1