thesofproject/sof #6013: No Sound - Arch Linux 5.18.12-arch1-1  - hanging shutdown [BUG]

changedsoul, 2022-07-17:
This on a Dell XPS 17 9720 Laptop with Intel Corporation Alder Lake PCH-P High Definition Audio Controller (rev 01).
I have gone thru the wiki and installed sof-firmware as instructed.

sof version: sof-firmware 2.2-1

with sof-installed:

> 0000:00:1f.3 Multimedia audio controller: Intel Corporation Alder Lake PCH-P High Definition Audio Controller (rev 01)
        Subsystem: Dell Device 0afe
        Kernel driver in use: sof-audio-pci-intel-tgl
        Kernel modules: snd_hda_intel, snd_sof_pci_intel_tgl

when the sof-firmware is installed the laptop hangs on shutdown indefinitely and requires holding the power button down.
looking at the next boot 'journalctl -b' i see these listed:

> Jul 16 23:15:36 olderman kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
Jul 16 23:15:36 olderman kernel: fbcon: i915drmfb (fb0) is primary device
Jul 16 23:15:36 olderman kernel: Console: switching to colour frame buffer device 240x75
Jul 16 23:15:36 olderman kernel: i915 0000:00:02.0: [drm] fb0: i915drmfb frame buffer device
Jul 16 23:15:36 olderman kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: use msi interrupt mode
Jul 16 23:15:36 olderman kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: hda codecs found, mask 4
Jul 16 23:15:36 olderman kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 2:2:0-57864
Jul 16 23:15:36 olderman kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:22:1 Kernel ABI 3:19:1
Jul 16 23:15:36 olderman kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: warn: FW ABI is more recent than kernel
Jul 16 23:15:36 olderman kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: unknown sof_ext_man header type 3 size 0x30
Jul 16 23:15:36 olderman kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 2:2:0-57864
Jul 16 23:15:36 olderman kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:22:1 Kernel ABI 3:19:1
Jul 16 23:15:36 olderman kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: warn: FW ABI is more recent than kernel
...
Jul 16 23:15:38 olderman kernel: soundwire sdw:3:025d:0714:01: Probe not complete, timed out
Jul 16 23:15:38 olderman kernel: soundwire sdw:3:025d:0714:01: Update Slave status failed:-110
Jul 16 23:15:38 olderman kernel: soundwire sdw:2:025d:1316:01: Probe not complete, timed out
Jul 16 23:15:38 olderman kernel: soundwire sdw:2:025d:1316:01: Update Slave status failed:-110
Jul 16 23:15:38 olderman kernel: soundwire sdw:1:025d:1316:01: Probe not complete, timed out
Jul 16 23:15:38 olderman kernel: soundwire sdw:1:025d:1316:01: Update Slave status failed:-110
Jul 16 23:15:38 olderman kernel: soundwire sdw:0:025d:0711:01: Probe not complete, timed out
Jul 16 23:15:38 olderman kernel: soundwire sdw:0:025d:0711:01: Update Slave status failed:-110
...
Jul 16 23:15:39 olderman kernel: sof_sdw sof_sdw: snd_soc_register_card failed -517
Jul 16 23:15:40 olderman kernel: sof_sdw sof_sdw: snd_soc_register_card failed -517
Jul 16 23:15:40 olderman kernel: sof_sdw sof_sdw: snd_soc_register_card failed -517
Jul 16 23:15:40 olderman kernel: sof_sdw sof_sdw: snd_soc_register_card failed -517
Jul 16 23:15:40 olderman kernel: sof_sdw sof_sdw: snd_soc_register_card failed -517
...
ul 16 23:16:39 olderman systemd-udevd[368]: sdw:0:025d:0711:01: Worker [374] processing SEQNUM=5150 is taking a long time
Jul 16 23:16:39 olderman systemd-udevd[368]: sdw:3:025d:0714:01: Worker [422] processing SEQNUM=5162 is taking a long time
Jul 16 23:18:39 olderman systemd-udevd[368]: sdw:0:025d:0711:01: Worker [374] processing SEQNUM=5150 killed
Jul 16 23:18:39 olderman systemd-udevd[368]: sdw:3:025d:0714:01: Worker [422] processing SEQNUM=5162 killed
...
ul 16 23:19:36 olderman kernel: INFO: task systemd-udevd:374 blocked for more than 122 seconds.
Jul 16 23:19:36 olderman kernel:       Tainted: P           OE     5.18.12-arch1-1 #1
Jul 16 23:19:36 olderman kernel: "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
Jul 16 23:19:36 olderman kernel: task:systemd-udevd   state:D stack:    0 pid:  374 ppid:   368 flags:0x00000006
Jul 16 23:19:36 olderman kernel: Call Trace:
Jul 16 23:19:36 olderman kernel:  <TASK>

If I remove sof-firmware the system will again shutdown fine.

direc85, 2022-07-18:
I noticed you have the same issue with the same lines in `dmesg` than I did:

> sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 2:2:0-57864
> sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:22:1 Kernel ABI 3:19:1
> sof-audio-pci-intel-tgl 0000:00:1f.3: warn: FW ABI is more recent than kernel

Same here with a Tiger Lake laptop (Lenovo X1 Carbon) and Manjaro Linux. I made a [Manjaro forum post](https://forum.manjaro.org/t/no-sound-with-sof-firmware-2-2-pipewire-alsa-1-2-7-1-sequence-command-devdisall-is-ignored/116874) about it (pardon the URL and original title, I was sidetracked at first).

I managed to restore sound (and general system responsiveness) by downgrading `sof-firmware` from `2.2-1` to `2.1.1.a-1`. I still get a different ABI version mismatch message, though:

> sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 2:1:1-3964a
> sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:21:0 Kernel ABI 3:19:1
> sof-audio-pci-intel-tgl 0000:00:1f.3: warn: FW ABI is more recent than kernel

plbossart, 2022-07-18:
@changedsoul the shutdown issues are known, they fixes were contributed upstream last week.

The bigger issue is why the probe fails. Can you please add this file 
[sof-dyndbg.conf.txt](https://github.com/thesofproject/sof/files/9133208/sof-dyndbg.conf.txt) as /etc/modprobe.d/sof-dyndbg.conf (the extension matters), reboot and attach the dmesg log.

Also please attach the 'alsa-info' log.

plbossart, 2022-07-18:
> I managed to restore sound (and general system responsiveness) by downgrading `sof-firmware` from `2.2-1` to `2.1.1.a-1`. I still get a different ABI version mismatch message, though:

That's not normal and should be filed as a separate bug @direc85. I have no idea why downgrading the firmware would help.
@lgirdwood FYI.

lgirdwood, 2022-07-18:
@changedsoul we had a similar issue on a Chromebook that was resolved with a slightly different shutdown flow. Adding @kv2019i who is back next week and implemented the Chromebook fix. It may additionally need to be used here. Please keep using v2.1.1 for the moment.

plbossart, 2022-07-18:
@lgirdwood the shutdown issue is only on SoundWire device where we tried to remove a headset device property in the machine driver that was not set on deferred probe. Nothing to do with the SOF PCI shutdown problems.

direc85, 2022-07-19:
> That's not normal and should be filed as a separate bug

Thanks, I'll do that,.

plbossart, 2022-07-19:
@changedsoul the Dell SKU is not handled in the kernel for now, this is a device Dell didn't mention to us....

````
System SKU:        0AFE
````

The weird part is that we end-up selecting an existing configuration that fails at run-time with all kinds of SoundWire issues.

I think what's happening is that we detect a 2-amp machine, but since we are missing the SKU support we use the 1-amp version in the machine driver.