thesofproject/sof #6014: Why do topologies have generic names?

psykana, 2022-07-17:
Take a look at this commit, for example: https://github.com/thesofproject/sof/pull/5974/files
The only thing that indicates that this tolopogy is for an HP Omen 16 device is the commit message. This goes for pretty much for the entire library of SOF topologies.
As someone who's just getting into SOF, I am really surprised that finding what runs on your machine appears to require some sacred knowledge.
IMO, OEMs or whoever submits PR's with device-specific topologies, should name them accordingly.
Please tell me what's the reasoning behind this and if I'm missing something here.
Thanks.

ranj063, 2022-07-17:
@psykana the main reason it to reduce maintenaning multiple topology files and promote reuse. The topology file associate with a machine is associated with the platform, the DAI's on the platform and the codecs on a device rather than the device name itself. There could be several devices out there with the same set of DAI's/codecs for the same platform. It is unrealistic to maintain a separate topology file for each of those devices out there. Hope this helps

plbossart, 2022-07-18:
@psykana there's nothing 'sacred' here, what we try to do in the Linux driver is to at boot time from the hardware, PCI and ACPI subsystem what the platform is. It's not different from what is done in other part of the Linux sound subsystem. We typically don't use commercial names unless we have no alternative, the DMI quirks that were introduced for the HP Omen 16 are the last-resort to fix a broken BIOS, not the baseline.

closed 2022-10-25 by psykana