thesofproject/sof #6015: [BUG] ES8336: firmware complains about missing topology file sof-apl-es8336-ssp5.tplg

akssri, 2022-07-18:
**Describe the bug**
Laptop (obscure JP model: Donki Nanote Next UMPC03-SR) tries to load es8336 module but fails due to missing topology file. Searching github for "sof-apl-es8336-ssp5.tplg" yielded no results.

**To Reproduce**
Start system; check dmesg logs.

**Reproduction Rate**
Every boot.

**Expected behavior**
Sound driver to load.

**Impact**
No audio device shown with ALSA etc.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 2b84fa749610e899b13f9b6e367fd34c2dc47042 (5.18.12-arch1-1; archlinux package; arch-commits on top don't change much)
    * SOF: 57864296b07c5ffd673214dd5b1e3d07ee274a22 (via sof-bin v2.2; archlinux package)
2) Name of the topology file
    * Topology: firmware tries to load sof-apl-es8336-ssp5.tplg but fails
3) Name of the platform(s) on which the bug is observed.
    * Platform: archlinux (latest)

**Screenshots or console output**
lspci output 
``` 
~ > lspci | grep -i audio
00:0e.0 Multimedia audio controller: Intel Corporation Celeron N3350/Pentium N4200/Atom E3900 Series Audio Cluster (rev 0b)
```
dmesg output
```
~ > dmesg | grep -i audio
[   18.432174] sof-audio-pci-intel-apl 0000:00:0e.0: DSP detected with PCI class/subclass/prog-if info 0x040100
[   18.432619] sof-audio-pci-intel-apl 0000:00:0e.0: DSP detected with PCI class/subclass/prog-if 0x040100
[   18.433008] sof-audio-pci-intel-apl 0000:00:0e.0: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[   18.439050] sof-audio-pci-intel-apl 0000:00:0e.0: use msi interrupt mode
[   18.479836] sof-audio-pci-intel-apl 0000:00:0e.0: NHLT_DEVICE_I2S detected, ssp_mask 0x21
[   18.479854] sof-audio-pci-intel-apl 0000:00:0e.0: hda codecs found, mask 4
[   18.515539] sof-audio-pci-intel-apl 0000:00:0e.0: Firmware info: version 2:2:0-57864
[   18.515554] sof-audio-pci-intel-apl 0000:00:0e.0: Firmware: ABI 3:22:1 Kernel ABI 3:19:1
[   18.515558] sof-audio-pci-intel-apl 0000:00:0e.0: warn: FW ABI is more recent than kernel
[   18.515580] sof-audio-pci-intel-apl 0000:00:0e.0: unknown sof_ext_man header type 3 size 0x30
[   18.566367] sof-audio-pci-intel-apl 0000:00:0e.0: Firmware info: version 2:2:0-57864
[   18.566381] sof-audio-pci-intel-apl 0000:00:0e.0: Firmware: ABI 3:22:1 Kernel ABI 3:19:1
[   18.566385] sof-audio-pci-intel-apl 0000:00:0e.0: warn: FW ABI is more recent than kernel
[   18.651042] sof-audio-pci-intel-apl 0000:00:0e.0: Direct firmware load for intel/sof-tplg/sof-apl-es8336-ssp5.tplg failed with error -2
[   18.651063] sof-audio-pci-intel-apl 0000:00:0e.0: error: tplg request firmware intel/sof-tplg/sof-apl-es8336-ssp5.tplg failed err: -2
[   18.651066] sof-audio-pci-intel-apl 0000:00:0e.0: you may need to download the firmware from https://github.com/thesofproject/sof-bin/
[   18.651069] sof-audio-pci-intel-apl 0000:00:0e.0: error: failed to load DSP topology -2
[   18.651072] sof-audio-pci-intel-apl 0000:00:0e.0: ASoC: error at snd_soc_component_probe on 0000:00:0e.0: -2
```

akssri, 2022-07-18:
@lgirdwood can confirm. couldn't find the file in either in the installed package or in sof-bin repo.

plbossart, 2022-07-18:
Indeed we've only seen SSP0 or SSP2 used, we've never seen a case where SSP5 was used. It could be a mistake in the NHLT table, we currently use the SSP with the highest index declared in NHLT, maybe this is not correct.

@akssri can you dump the ACPI tables and/or look at the NHTL with the latest ACPICA tools?

akssri, 2022-07-19:
PFA the acpidump
[acpidump.txt](https://github.com/thesofproject/sof/files/9142032/acpidump.txt)

plbossart, 2022-07-19:
That's a tough one. The NHLT indicates support for SSP0 and SSP5, but the SSP5 is referenced multiple times on render. That looks like a bug.

[nhlt.dsl.txt](https://github.com/thesofproject/sof/files/9142675/nhlt.dsl.txt)

Please try with this file, unzip it and copy it to /lib/firmware/intel/sof-tplg
[sof-apl-es8336-ssp5.tplg.gz](https://github.com/thesofproject/sof/files/9142753/sof-apl-es8336-ssp5.tplg.gz)

If this doesn't work, we may have to manually edit the files and force the use of SSP0.

akssri, 2022-07-21:
Thanks! I tried using the new topology file but it looks there are some ABI issues,
```
vamana ~ > sudo dmesg | grep -i audio 
[sudo] password for akssri: 
[    0.249700] ACPI: Added _OSI(Linux-Lenovo-NV-HDMI-Audio)
[   22.277476] sof-audio-pci-intel-apl 0000:00:0e.0: DSP detected with PCI class/subclass/prog-if info 0x040100
[   22.277918] sof-audio-pci-intel-apl 0000:00:0e.0: DSP detected with PCI class/subclass/prog-if 0x040100
[   22.278162] sof-audio-pci-intel-apl 0000:00:0e.0: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[   22.284062] sof-audio-pci-intel-apl 0000:00:0e.0: use msi interrupt mode
[   22.349744] sof-audio-pci-intel-apl 0000:00:0e.0: NHLT_DEVICE_I2S detected, ssp_mask 0x21
[   22.349763] sof-audio-pci-intel-apl 0000:00:0e.0: hda codecs found, mask 4
[   22.403500] sof-audio-pci-intel-apl 0000:00:0e.0: Firmware info: version 2:2:0-57864
[   22.403513] sof-audio-pci-intel-apl 0000:00:0e.0: Firmware: ABI 3:22:1 Kernel ABI 3:19:1
[   22.403517] sof-audio-pci-intel-apl 0000:00:0e.0: warn: FW ABI is more recent than kernel
[   22.403541] sof-audio-pci-intel-apl 0000:00:0e.0: unknown sof_ext_man header type 3 size 0x30
[   22.454849] sof-audio-pci-intel-apl 0000:00:0e.0: Firmware info: version 2:2:0-57864
[   22.454863] sof-audio-pci-intel-apl 0000:00:0e.0: Firmware: ABI 3:22:1 Kernel ABI 3:19:1
[   22.454866] sof-audio-pci-intel-apl 0000:00:0e.0: warn: FW ABI is more recent than kernel
[   22.513479] sof-audio-pci-intel-apl 0000:00:0e.0: Direct firmware load for intel/sof-tplg/sof-apl-es8336-ssp5.tplg failed with error -2
[   22.513498] sof-audio-pci-intel-apl 0000:00:0e.0: error: tplg request firmware intel/sof-tplg/sof-apl-es8336-ssp5.tplg failed err: -2
[   22.513500] sof-audio-pci-intel-apl 0000:00:0e.0: you may need to download the firmware from https://github.com/thesofproject/sof-bin/
[   22.513503] sof-audio-pci-intel-apl 0000:00:0e.0: error: failed to load DSP topology -2
[   22.513506] sof-audio-pci-intel-apl 0000:00:0e.0: ASoC: error at snd_soc_component_probe on 0000:00:0e.0: -2

vamana ~ > uname -a   
Linux vamana 5.18.12-arch1-1 #1 SMP PREEMPT_DYNAMIC Fri, 15 Jul 2022 15:33:02 +0000 x86_64 GNU/Linux

```

plbossart, 2022-07-21:
Ah bummer. I just realized I hard-coded support for SSP0..2 in sound/soc/intel/boards/sof_es8336.c

````c	
                if (mach->mach_params.i2s_link_mask & BIT(2))
			ssp = SOF_ES8336_SSP_CODEC(2);
		else if (mach->mach_params.i2s_link_mask & BIT(1))
			ssp = SOF_ES8336_SSP_CODEC(1);
		else  if (mach->mach_params.i2s_link_mask & BIT(0))
			ssp = SOF_ES8336_SSP_CODEC(0);
````

That was stupid of me but hey I can't test any of this myself....

if you know how to recompile your own kernel, you can fix this by replacing all this code by 

ssp = SOF_ES8336_SSP_CODEC(5);

I'll provide a better solution when I get a bit of time.

plbossart, 2022-07-21:
Rather use what I had in the SOF driver

````c
			/* fls returns 1-based results, SSPs indices are 0-based */
			ssp_num = fls(mach->mach_params.i2s_link_mask) - 1;
````

plbossart, 2022-07-21:
@akssri please retest with the hack above or with the patches in PR https://github.com/thesofproject/linux/pull/3776

akssri, 2022-07-28:
(Apologies for the delay in testing this.)
 
I compiled the kernel using the config from Arch's package
https://github.com/archlinux/svntogit-packages/tree/packages/linux/trunk

Still no luck.

```
vamana ~ > sudo dmesg | grep -i audio
[    0.252547] ACPI: Added _OSI(Linux-Lenovo-NV-HDMI-Audio)
[   17.528656] sof-audio-pci-intel-apl 0000:00:0e.0: DSP detected with PCI class/subclass/prog-if info 0x040100
[   17.529094] sof-audio-pci-intel-apl 0000:00:0e.0: DSP detected with PCI class/subclass/prog-if 0x040100
[   17.529323] sof-audio-pci-intel-apl 0000:00:0e.0: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[   17.534994] sof-audio-pci-intel-apl 0000:00:0e.0: use msi interrupt mode
[   17.570291] sof-audio-pci-intel-apl 0000:00:0e.0: NHLT_DEVICE_I2S detected, ssp_mask 0x21
[   17.570305] sof-audio-pci-intel-apl 0000:00:0e.0: hda codecs found, mask 4
[   17.692044] sof-audio-pci-intel-apl 0000:00:0e.0: Firmware info: version 2:2:0-57864
[   17.692055] sof-audio-pci-intel-apl 0000:00:0e.0: Firmware: ABI 3:22:1 Kernel ABI 3:23:0
[   17.692083] sof-audio-pci-intel-apl 0000:00:0e.0: unknown sof_ext_man header type 3 size 0x30
[   17.744008] sof-audio-pci-intel-apl 0000:00:0e.0: Firmware info: version 2:2:0-57864
[   17.744018] sof-audio-pci-intel-apl 0000:00:0e.0: Firmware: ABI 3:22:1 Kernel ABI 3:23:0
```

the module does seem to be getting loaded,
```
vamana ~ > lsmod | grep -i snd
snd_soc_sof_es8336     20480  0
snd_soc_intel_hda_dsp_common    20480  1 snd_soc_sof_es8336
snd_sof_probes         24576  0
snd_hda_codec_hdmi     86016  0
snd_soc_dmic           16384  0
snd_sof_pci_intel_apl    16384  0
snd_sof_intel_hda_common   217088  1 snd_sof_pci_intel_apl
soundwire_intel        53248  1 snd_sof_intel_hda_common
snd_sof_intel_hda      20480  1 snd_sof_intel_hda_common
snd_sof_pci            24576  2 snd_sof_intel_hda_common,snd_sof_pci_intel_apl
snd_sof_xtensa_dsp     20480  1 snd_sof_intel_hda_common
snd_sof               331776  3 snd_sof_pci,snd_sof_intel_hda_common,snd_sof_probes
snd_sof_utils          20480  1 snd_sof
snd_soc_avs           172032  0
snd_soc_hda_codec      28672  1 snd_soc_avs
snd_soc_skl           217088  0
snd_soc_hdac_hda       28672  2 snd_sof_intel_hda_common,snd_soc_skl
snd_hda_ext_core       36864  6 snd_soc_avs,snd_soc_hda_codec,snd_sof_intel_hda_common,snd_soc_hdac_hda,snd_soc_skl,snd_sof_intel_hda
snd_soc_sst_ipc        20480  1 snd_soc_skl
snd_soc_sst_dsp        40960  1 snd_soc_skl
snd_soc_acpi_intel_match    69632  3 snd_sof_intel_hda_common,snd_soc_skl,snd_sof_pci_intel_apl
snd_soc_acpi           16384  3 snd_soc_acpi_intel_match,snd_sof_intel_hda_common,snd_soc_skl
snd_hda_intel          61440  0
snd_soc_es8316         53248  0
snd_intel_dspcfg       36864  5 snd_soc_avs,snd_hda_intel,snd_sof,snd_sof_intel_hda_common,snd_soc_skl
snd_soc_core          393216  11 snd_soc_avs,snd_soc_hda_codec,soundwire_intel,snd_sof,snd_sof_intel_hda_common,snd_soc_hdac_hda,snd_soc_sof_es8336,snd_soc_skl,snd_sof_probes,snd_soc_es8316,snd_soc_dmic
snd_intel_sdw_acpi     20480  2 snd_sof_intel_hda_common,snd_intel_dspcfg
snd_compress           28672  2 snd_soc_core,snd_sof_probes
snd_hda_codec         184320  6 snd_soc_avs,snd_hda_codec_hdmi,snd_soc_hda_codec,snd_hda_intel,snd_soc_intel_hda_dsp_common,snd_soc_hdac_hda
ac97_bus               16384  1 snd_soc_core
snd_hda_core          118784  11 snd_soc_avs,snd_hda_codec_hdmi,snd_soc_hda_codec,snd_hda_intel,snd_hda_ext_core,snd_hda_codec,snd_soc_intel_hda_dsp_common,snd_sof_intel_hda_common,snd_soc_hdac_hda,snd_soc_skl,snd_sof_intel_hda
snd_hwdep              16384  1 snd_hda_codec
snd_pcm_dmaengine      16384  1 snd_soc_core
snd_pcm               172032  14 snd_soc_avs,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,soundwire_intel,snd_sof,snd_sof_intel_hda_common,snd_compress,snd_soc_core,snd_sof_utils,snd_soc_skl,snd_soc_es8316,snd_hda_core,snd_pcm_dmaengine
snd_timer              49152  1 snd_pcm
snd                   131072  10 snd_hda_codec_hdmi,snd_hwdep,snd_hda_intel,snd_hda_codec,snd_sof,snd_timer,snd_compress,snd_soc_core,snd_soc_sof_es8336,snd_pcm
soundcore              16384  1 snd
```

but no sound card,
```
vamana ~ > aplay -l
aplay: device_list:275: no soundcards found...
```

plbossart, 2022-07-28:
@akssri can you please add this file 
[sof-dyndbg.conf.txt](https://github.com/thesofproject/sof/files/9210975/sof-dyndbg.conf.txt)
as /etc/modprobe.d/sof-dyndbg.conf, reboot and attach the full dmesg log. Thanks!

akssri, 2022-07-30:
PFA the full dmesg log; please lmk if you need anything else. Thanks!
[dmesg-220730.txt](https://github.com/thesofproject/sof/files/9224795/dmesg-220730.txt)

plbossart, 2022-08-01:
@akssri can you check the configuration, there's something very odd here

````
[   20.125569] snd_sof:sof_machine_register: sof-audio-pci-intel-apl 0000:00:0e.0: created machine sof-essx8336
[   20.125610] snd_sof_pci:sof_pci_probe_complete: sof-audio-pci-intel-apl 0000:00:0e.0: Completing SOF PCI probe
````

These two lines are the last step, and there's no card created. Please make sure you don't have any blacklists for sof_es8336 or try to do a manual 'modprobe snd_soc_sof_es8336'

Or maybe the log was truncated?

plbossart, 2022-08-01:
also use this file, we forgot sof_es8336... 
[sof-dyndbg.conf.txt](https://github.com/thesofproject/sof/files/9235362/sof-dyndbg.conf.txt)

akssri, 2022-08-02:
dmesg log with the new modprobe conf,
[20220802_dmesg_vamana.txt](https://github.com/thesofproject/sof/files/9238756/20220802_dmesg_vamana.txt)

Checked that there was nothing else in modprobe.d or kernel-arg being passed elsewhere, and that the module was being loaded (this was the case at bootup, but the log was dumped after manually running modprobe).
```
vamana ~ > cat /etc/modprobe.d/sof-dyndbg.conf | grep -i es833
options snd_soc_sof_es8336 dyndbg=+pmf

vamana ~ > lsmod | grep -i es8336
snd_soc_sof_es8336     20480  0
snd_soc_intel_hda_dsp_common    20480  1 snd_soc_sof_es8336
snd_soc_core          393216  11 snd_soc_avs,snd_soc_hda_codec,soundwire_intel,snd_sof,snd_sof_intel_hda_common,snd_soc_hdac_hda,snd_soc_sof_es8336,snd_soc_skl,snd_sof_probes,snd_soc_es8316,snd_soc_dmic
snd                   131072  10 snd_hda_codec_hdmi,snd_hwdep,snd_hda_intel,snd_hda_codec,snd_sof,snd_timer,snd_compress,snd_soc_core,snd_soc_sof_es8336,snd_pcm

vamana ~ > sudo aplay -l
aplay: device_list:275: no soundcards found...
```

plbossart, 2022-08-02:
it's likely the same issue reported in https://github.com/thesofproject/linux/issues/3336#issuecomment-1202616039 by @ejmarkow 

Somehow the introduction of HDMI inputs broke everything. Please revert that patch and retry.

What a mess.

plbossart, 2022-08-08:
Fixed by https://github.com/thesofproject/sof/pull/6078

See https://github.com/thesofproject/linux/wiki/ES8336-support for recommended setup, and file a new issue under the Linux project if this doesnt' work.

closed 2022-08-08 by plbossart